espressif/esp-idf #2843: Console demo does not read serial data when pin numbers are changed.

tommessick, 2018-12-17:
## Environment
- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version: v3.3-dev-424-g8b885fb93
- Build System:         Make
- Compiler version: crosstool-ng-1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:        external 5V

I am running the console example from the IDF.  If I build it with no changes it uses UART0 over the default pins and works fine.  

The ESP32-WROOM-32 is on a board with an RS232 port through a MAX3232.  Pin 32 is TX and pin 33 is RX.  I have code that sends and receives data over these pins so I am sure they are wired up correctly.

If I use make menuconfig to configure the console to use pins 32 and 33 the console demo prints to the serial port but cannot receive any data.   

I don't have a JTAG debugger.   In uart.c uart_rx_intr_handler_default() there is code to check for these bits:
        UART_RXFIFO_TOUT_INT_ST_M
        UART_RXFIFO_FULL_INT_ST_M
        UART_AT_CMD_CHAR_DET_INT_ST_M
I have modified uart.c to cause an exception when a character is received.  I get the exception when i use the original UART setup but not when I switch to the other pins.

I have used make menuconfig to select UART0 or UART1 with the same results.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2684053/sdkconfig.txt)
[uart.c.txt](https://github.com/espressif/esp-idf/files/2684059/uart.c.txt)

koobest, 2018-12-17:
Can you provide you UART pin configuration?

tommessick, 2018-12-17:
This is the unaltered example from the IDF in examples/system/console.

koobest, 2018-12-17:
@tommessick 
You need to  configure the pins, because the default log output uses UART0 and the pins are already configured during the boot stage.

tommessick, 2018-12-19:
Adding this line:

```gpio_set_direction(GPIO_NUM_33, `GPIO_MODE_DEF_INPUT);```

before the call to initialize_console() fixed the problem.
Thanks.

igrr, 2018-12-19:
Thanks for the feedback, will keep this open until this is fixed in IDF.

KonstantinKondrashov, 2019-05-16:
Fix for this issue is in our internal repo, it will be merged as soon as possible. Thanks.

closed 2019-05-21 by igrr