espressif/esp-idf #2845: GPIO interrupts service and disabling interrupts

nameofuser1, 2018-12-17:
Hi there!

I have spent last few days looking for bug in my code and found out an interesting thing: GPIO interrupt service checks only for interrupt status. At the same time clearing `GPIO.pin[gpio_num].int_ena` flag, which is done by `gpio_intr_disable()` function, does not seem to mask gpio interrupt as a consequence interrupt service will still call disabled interrupt handler. 

https://github.com/espressif/esp-idf/blob/da2116f55c555ab6a4aa9d3a982e9bee990dd7b4/components/driver/gpio.c#L120-L126

https://github.com/espressif/esp-idf/blob/da2116f55c555ab6a4aa9d3a982e9bee990dd7b4/components/driver/gpio.c#L330-L361

Certainly calling `gpio_isr_handler_remove` or clearing `GPIO.pin[gpio_num].int_type` field can be used in order to stop gpio interrupt service to call disabled interrupts, but, naturally, if I use function named `whatever_disable_interrupt()`, I expect that corresponding irq handler will never be executed, which is not true in this case and requires me to look inside function and compare it with reference manual.

negativekelvin, 2018-12-17:
If the interrupt is disabled and the flag is cleared by gpio_intr_disable then it should be impossible for the flag to be set.  So the flag check in gpio_intr_service would never succeed and the handler for that gpio would not be called.

nameofuser1, 2018-12-17:
> If the interrupt is disabled and the flag is cleared by gpio_intr_disable then it should be impossible for the flag to be set. So the flag check in gpio_intr_service would never succeed and the handler for that gpio would not be called.

That's strange because until I clear `GPIO.pin[MY_GPIO_1].int_type`, interrupt handler for MY_GPIO_1 is executed whenever edge on MY_GPIO_1 line was found and any other GPIO interrupt is triggered. I can not attach my code and screenshots from logic analyzer right now but certainly will do it tomorrow.

X-Ryl669, 2018-12-18:
Is it called once only ? 
It's possible the interrupt is currently running (on the other core) and the test for interrupt status saying it's triggered already passed when you are disabling the interrupt. In that case, it'll be triggered once for all.

nameofuser1, 2018-12-18:
Updates :)

I am a bit late but nevertheless here is my setup. I have magnetic lock state sensor(indicates whether lock is closed or opened) connected to GPIO35. Also I have periodic interrupt on GPIO22. When lock's coil is activated, the signal on GPIO35 line begin to jitter, therefore I disable interrupt. However it is still called once more when interrupt on GPIO22 occurs.

Here is the gist with code I use to test:
https://gist.github.com/nameofuser1/67aaf27605809729aca27bdad47b2f73

And here is a logic analyzer result:
[image]

Same result but smaller resolution:
[image]

For the image explanation: T0 and T1 lines - TEST_GPIO_0 and TEST_GPIO_1 from the gist, LOCK - GPIO35 signal, DT - GPIO22, SCK - not used. 

If I clear `GPIO.pin[GPIO_NUM_35].int_type` inside an IRQ then everything works as I expect.

nameofuser1, 2018-12-18:
> Is it called once only ?
> It's possible the interrupt is currently running (on the other core) and the test for interrupt status saying it's triggered already passed when you are disabling the interrupt. In that case, it'll be triggered once for all.
> 
> Might be related to #2808

Thanks for your reply! 

Nah, there is no multicore IRQ processing. I have posted more information, please take a look.

negativekelvin, 2018-12-19:
Strange, have you tried 
```
while(GPIO.status1.intr_st & BIT(LOCK_SENSE_GPIO - 32))
    GPIO.status1_w1tc.intr_st = BIT(LOCK_SENSE_GPIO - 32);
```

nameofuser1, 2018-12-19:
> Strange, have you tried
> 
> ```
> while(GPIO.status1.intr_st & BIT(LOCK_SENSE_GPIO - 32))
>     GPIO.status1_w1tc.intr_st = BIT(LOCK_SENSE_GPIO - 32);
> ```

Just tried, it did not help.

koobest, 2018-12-19:
@nameofuser1, 
Can you try to enable the pull-up  parameters to do the test again? I have test your code, but not reproduce your issue.
thanks!!

nameofuser1, 2018-12-20:
> @nameofuser1,
> Can you try to enable the pull-up parameters(out put io) to do the test again? I have test your code, but not reproduce your issue.
> thanks!!

Thanks for your reply! Got a cold, going to check in a few days.

wingunder, 2018-12-24:
@nameofuser1, Please have a look if 84bf106 does not solve your problem, once you're up and running again.

You'll _have_ to use gpio_set_intr_type(), if you want to 'disable' the interrupt, from _within_ your ISR. If you are outside of your ISR, you may use gpio_intr_enable() and/or gpio_intr_disable() for this, at any time. Just make sure you use this patch, 84bf106, else it might not work.

BTW, don't manipulate any status flags or attributes of the GPIO structure directly. Always _only_ use the API, which is what gpio.h supplies.
eg. Use gpio_set_intr_type(x) and _not_ GPIO.pin[GPIO_NUM_35].int_type = x;

nameofuser1, 2018-12-24:
> TW, don't manipulate any status flags or attributes of the GPIO structure directly. Always _only_ use the API, which is what gpio.h supplies.
> eg. Use gpio_set_intr_type(x) and _not_ GPIO.pin[GPIO_NUM_35].int_type = x;

Thanks for your reply! Will test it soon. However I have several questions based on your answer:
1. Why does API differ when disabling from inside an ISR and in task context?
2. Why should not I directly access GPIO structure? I had eventual failures because of cache misses when using functions from gpio.h inside in ISR therefore I started to use raw GPIO structure.

wingunder, 2018-12-24:
> Thanks for your reply! Will test it soon.

No problem. Thanks promising to test.

>     1. Why does API differ when disabling from inside an ISR and from inside a task context?

I'm probably not the right person to answer this, but technically, you should pass a message, using FreeRTOS queues, to somewhere outside of your ISR and then you can call  gpio_intr_enable() and/or gpio_intr_disable(). I agree that this is not really cool, but it's not to say that my patch is correct. I just wanted you to try it out, as you are able to reproduce it. Once we know that this solves it, we can hopefully work on getting gpio_intr_enable() and gpio_intr_disable() callable from anywhere, if needed.

>     2. Why should not I directly access GPIO structure? I had eventual failures because of cache misses when using functions from gpio.h inside an ISR therefore I started to use raw GPIO structure.

You can do it, but you have to be aware that if you manipulate _anything_ that gpio_intr_service() relies on, things will be unpredictable. In your case, you were setting the status, which gpio_intr_service() reads and uses the result for clearing the status register. You can't use a spin_lock to protect the status, as is done in gpio.h, as you're in the ISR. AFAICS, there's no way to manipulate the status within your ISR, without causing things go wrong sooner or later.

OTOH, setting GPIO.pin[GPIO_NUM_35].int_type directly is still OK, as this variable is not relied on in gpio_intr_service(), but this is not good practice, in my opinion.

wingunder, 2018-12-29:
> Certainly calling gpio_isr_handler_remove or clearing GPIO.pin[gpio_num].int_type field can be used in order to stop gpio interrupt service to call disabled interrupts, but, naturally, if I use [...]

@nameofuser1, I have extended PR https://github.com/espressif/esp-idf/pull/2873, so you can now:
   - call  gpio_intr_disable from anywhere and expect the interrupt _not_ to occur again.
   - rely on gpio_intr_disable to disable an interrupt consistently.
   - call gpio_intr_enable from any task, regardless on which core it runs.

koobest, 2019-01-08:
Hi, @nameofuser1, did you solve your problem with wingunder extended PR?

@wingunder, thanks for you  PR, we will create an MR for this.

Alvin1Zhang, 2019-03-14:
@nameofuser1 Hi, have you tried wingunder's extended PR? Thanks

@koobest how about the MR? Thanks

koobest, 2019-07-20:
Hi, For the interrupt status register, there are two concepts:
* Interrupt enable register: When set to 1, response to the interrupt trigger signal, When set to 0, do not response to the interrupt trigger signal.
* Interrupt Status Register: when interrupt is disabled, the interrupt status register will not be set even if the interrupt trigger signal arrives.
* Interrupt Raw Status Register: When the interrupt is disabled, the interrupt raw status register will still be set when the interrupt trigger signal arrives.

GPIO only have the interrupt raw register. So, When the interrupt is disabled, the interrupt raw status register will still be set when the interrupt trigger signal arrives. Our driver has a bug in the handling of this, we will fix it. By the way, when the int_type register is set to 0, the interrupt raw status register will not be set.

negativekelvin, 2019-07-20:
@koobest 
>GPIO only have the interrupt raw register

Is it?
GPIO_STATUS_REG - raw, but called status?
GPIO_ACPU_INT_REG - status?
GPIO_PCPU_INT_REG - status?

koobest, 2019-07-22:
@negativekelvin 
Yes,  `GPIO_ACPU_INT_REG` and `GPIO_PCPU_INT_REG` are the status register.

closed 2019-08-05 by igrr