espressif/esp-idf #2846: task switch context issue

chegewara, 2018-12-18:
This crash i dont even know how to bite. I dont have memory leak for sure and crash happened after about 20 minutes, but app usually can run over 10 hours. Here is log:
```
Heap summary for capabilities 0x00000800:
  At 0x3ffc42ac len 32767 free 3644 allocated 28968 min_free 0
    largest_free_block 2104 alloc_blocks 27 free_blocks 3 total_blocks 30
  At 0x3ffae6e0 len 6432 free 0 allocated 6300 min_free 0
    largest_free_block 0 alloc_blocks 25 free_blocks 0 total_blocks 25
  At 0x3ffc0978 len 128648 free 24 allocated 127704 min_free 0
    largest_free_block 12 alloc_blocks 220 free_blocks 2 total_blocks 222
  At 0x3ffe0440 len 15072 free 4228 allocated 10660 min_free 0
    largest_free_block 1536 alloc_blocks 27 free_blocks 11 total_blocks 38
  At 0x3ffe4350 len 113840 free 93688 allocated 20104 min_free 88600
    largest_free_block 93688 alloc_blocks 3 free_blocks 1 total_blocks 4
  At 0x400988e0 len 30496 free 30460 allocated 0 min_free 30452
    largest_free_block 30460 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 132044 allocated 193736 min_free 119052 largest_free_block 93688
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400929c5  PS      : 0x00060033  A0      : 0x400950f7  A1      : 0x3ffc2d90  
A2      : 0x00050023  A3      : 0x00000001  A4      : 0x3ffb3f60  A5      : 0x3ffb3f60  
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x0000000a  A9      : 0x00000002  
A10     : 0x1b468529  A11     : 0xe67ec000  A12     : 0x3ffb4030  A13     : 0x3620464e  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xe67ec048  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400929c5:0x3ffc2d90 0x400950f4:0x3ffc2db0 0x400950aa:0x00000003
```
```
0x400929c5 is in vTaskSwitchContext (D:/msys32/home/imper/esp/esp-idf/components/freertos/tasks.c:2870).
2865                                            if (foundNonExecutingWaiter == pdTRUE) {
2866                                                    /* If the task is not being executed
2867                                                    by another core and its affinity is
2868                                                    compatible with the current one,
2869                                                    prepare it to be swapped in */
2870                                                    if (pxTCB->xCoreID == tskNO_AFFINITY) {
2871                                                            pxCurrentTCB[xPortGetCoreID()] = pxTCB;
2872                                                            ableToSchedule = pdTRUE;
2873                                                    } else if (pxTCB->xCoreID == xPortGetCoreID()) {
2874                                                            pxCurrentTCB[xPortGetCoreID()] = pxTCB;
```
```
0x400950aa is at D:/msys32/home/imper/esp/esp-idf/components/freertos\portasm.S:206.
201
202         #ifdef __XTENSA_CALL0_ABI__
203         call0   vPortYieldFromInt       /* call dispatch inside the function; never returns */
204         #else
205         call4   vPortYieldFromInt       /* this one returns */
206         call0   _frxt_dispatch          /* tail-call dispatcher */
207         /* Never returns here. */
208         #endif
209
210     .Lnoswitch:
```
```
0x400950f4 is at D:/msys32/home/imper/esp/esp-idf/components/freertos\portasm.S:406.
401         call0   vTaskSwitchContext  // Get next TCB to resume
402         movi    a2, pxCurrentTCB
403             getcoreid a3
404             addx4   a2,  a3, a2
405         #else
406         call4   vTaskSwitchContext  // Get next TCB to resume
407         movi    a2, pxCurrentTCB
408             getcoreid a3
409             addx4   a2,  a3, a2
410         #endif
```
I mean, i know how i can try to fix it, by use tasks pinned to core or only one core, but how it can crash in this line? Only idea i have is some serious memory issues i have(in app).

igrr, 2018-12-18:
It would help if you could attach the elf file or run `disass /m 0x400929c5` in GDB for a bit more context.
Here, EXCVADDR=0xe67ec048, so load or store happened from that address, which is obviously some garbage value. You also can notice that A11=0xe67ec000, and was probably used to derive the load/store address. Where did A11 come from? It's hard to say without disassembly. But that value was likely corrupted, most likely in static memory or on the stack.

chegewara, 2018-12-18:
Hi @igrr , thanks for instructions. Here is disass output:
https://gist.github.com/chegewara/14c64c7fd1e6f6dd1b35c348ad55c363

igrr, 2018-12-18:
a11 comes from `listGET_OWNER_OF_NEXT_ENTRY( pxTCB, &( pxReadyTasksLists[ uxDynamicTopReady ] ) );`, which means that `owner` is corrupted in one of the task lists.

> some serious memory issues i have(in app).

What kind of issues do you have, not enough memory?

chegewara, 2018-12-18:
The thing is i dont see any problem with memory in my code. As you can see i have a lot free heap, and it is steady for hours:
`free 132044 allocated 193736 min_free 119052 largest_free_block 93688`

What i meant is i can have some heap corruption that i dont know yet, but after few days when i finally fixed memory leak in json parsing, i dont have any known memory problems.

nameofuser1, 2018-12-18:
You sure your tasks have enough stack? Stack overflow is tricky to detect so having no memory leaks does not mean that you memory is not corrupted. You can try to enable heap corruption detection in comprehensive as it is described [here](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/heap_debug.html). This will also be able to detect stack overflows. Once it helped me a lot.

closed 2019-05-17 by chegewara