espressif/esp-idf #2849: cannot flash automatically

tatsutaigu, 2018-12-19:
Hi,

I use DevkitC v4.
I cannot flash automatically.
I must set "menuconfig -> Serial flasherconfig -> Before flashing -> No reset" and control EN & Boot  buttons by manual.

How can I flash automatically?

> $ python /home/foo/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port COM21 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /home/foo/esp/blink/build/bootloader/bootloader.bin 0x10000 /home/foo/esp/blink/build/blink.bin 0x8000 /home/foo/esp/blink/build/partitions_singleapp.bin
> esptool.py v2.6-beta1
> Serial port COM21
> Connecting........_____....._____....._____....._____....._____....._____....._____
> 
> A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
>

eagi223, 2018-12-20:
There should either be a button labeled either 0 or boot. Holding it while the "Connecting........." dialog is happening should start the flashing process. I've had issues with the auto-flash abilities of some dev boards, so you just have to manually hold the button while you are flashing to start it. The SparkFun ESP32 Thing is one that I can consistently get to auto-flash, but it's kind of expensive compared to other ESP32 dev boards. I believe it has to do with the reset/boot circuit on the board.

tatsutaigu, 2019-01-22:
I solder 1uF capacitor on EN pin-GND.
It could flash.
I guess EN/IO0 timing is different.
I use windows 7.

Alvin1Zhang, 2019-03-21:
@tatsutaigu Hi, does your issue still exist? Thanks.

chuanqi110, 2019-03-21:
Thank you very much for your feedback. Please try to remove the capacitance C15, which is connected in parallel on IO0. Thank you!

tatsutaigu, 2019-03-26:
Hi,
C15 was already removed when i bought.
did you update flash sequence?

tatsutaigu, 2019-07-29:
Hi,
I tried latest setup tool with `idf.py -p COM20 flash`.
[https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html#esp-idf-tools-installer](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html#esp-idf-tools-installer)

However, it cannot flash.
I guess DevKitC v4 has hardware bug.
C9 must be 1uF.
I replace C9 0.1uF to 1uF, it could flash automatically.

And, esptool.py may have bug.
`idf.py menuconfig` -> `serial flasher config` -> `Before flashing (No reset)`
`idf.py -p COM20 flash`, then it flash DevkitC v4 with C9 1uF without pressing control SW1 and SW2.

mill1000, 2020-12-28:
Also seeing this on a DevKitC V4 purchased from Digikey this month. C15 is not loaded which I believe is causing the issue. 

Here's a zoomed out capture of the boot sequence.
[image]

with more detail on the rising edge of EN. Without C15, IO0 rises before EN and therefore program mode is not entered.
[image]

Adding a small cap to IO0 (.01 uF) slows it down enough to be captured properly.
[image]

projectgus, 2021-01-13:
Hi @mill1000,

Thanks for providing the traces. C15 was removed from the DevKitC in 2018 in order to keep the output signal from IO0 clean, in particular this pin can be used a clock output for an ethernet MAC.

There is a limitation with the auto reset circuit because there is a transition period where RTS & DTR are both asserted, and the "auto reset" logic will drive both EN & IO0 high, before RTS is de-asserted and IO0 goes low. The first event corresponds with "AX" (roughly) and the second event with "BX" (roughly). If the chip starts up during this period as EN passes the high threshold, it straps into the wrong boot mode.

Are you using Windows? The traces seem to show 15ms between these two transitions, normally even on Windows this period is normally <1ms (meaning that IO0's second falling edge comes before EN has finished rising.) This event is determined by the OS USB stack and driver timing.

mill1000, 2021-01-13:
Hi @projectgus 

Thanks for that information. I am using WSL to flash these devkits and I am beginning to suspect this may be part of the problem. I have occasionally encountered failures with esptool as well due to serial timeouts. So I was beginning to think whatever "mapping" WSL has to use to access the USB devices could be part of the problem.

projectgus, 2021-01-13:
Hi @mill1000, Thansk for getting back to me. That seems likely that maybe WSL is adding latency between the serial port operations. I'm glad you found the workaround by adding C15 back, although note that you may find the board now occasionally boots into Download Mode when you don't want it to. I am reminded that this is [the other reason C15 is no longer populated](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-devkitc.html#note-on-c15).

mill1000, 2021-01-13:
Ha well I've actually opt'd to forgo my own workaround and instead just rely on pressing the flash button.

danak6jq, 2021-01-24:
Something possibly related; I have been flashing an ESP32-WROVER-based device successfully with an older HP i7-4770 system, but a new Lenovo Ryzen 7 PRO 4750G system was not successfully flashing it. Both with Win 10 Pro, both with same SiLabs driver. After a lot of fiddling-around, I realized that the ports on the Lenovo are USB3.x and the HP ports are USB2. I have a USB2 external hub. Plugged the ESP32 into the **external USB2 hub** on the Lenovo and am again successfully flashing the ESP32.

projectgus, 2021-01-26:
Hi @danak6jq,

Thanks for noting this! We had a [similar report regarding esptool on some Ryzen chipsets](https://github.com/espressif/esptool/issues/516), it seems like the USB host driver is different. I'm not sure if this is the same root cause, but if it is then it's good to hear you found an effective workaround.

projectgus, 2021-02-08:
Closing this as there's nothing we can generally do to fix these issues in ESP-IDF.

If you think you've identified a flashing issue that can befixed in esptool then please [report it there](https://github.com/espressif/esptool/issues), however note that a lot of flashing issues are hardware issues (either with the development board reset circuit, or with USB host behaviour) that can't realistically be fixed in ESP-IDF.

closed 2021-02-08 by projectgus