espressif/esp-idf #2851: BLE OOB pairing support

copercini, 2018-12-19:
It's more a documentation issue:

We are planing to pair some BLE servers with ESP32 (as secure BLE client) using NFC OOB... 
looking at examples it seems supported [here as client](https://github.com/espressif/esp-idf/blob/960c240578d359a1c5cf8cb3cbd9ce3150d5a370/examples/bluetooth/gatt_security_client/main/example_ble_sec_gattc_demo.c#L393) and [here as server](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c#L315)

```cpp
    case ESP_GAP_BLE_OOB_REQ_EVT: {
        ESP_LOGI(GATTC_TAG, "ESP_GAP_BLE_OOB_REQ_EVT");
        uint8_t tk[16] = {1}; //If you paired with OOB, both devices need to use the same tk
        esp_ble_oob_req_reply(param->ble_security.ble_req.bd_addr, tk, sizeof(tk));
        break;
}
```

But on [GATT secure server tutorial](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_security_server/tutorial/Gatt_Security_Server_Example_Walkthrough.md#setting-security-parameters) it's clearing saying: `OOB Flag: describes if the device supports Out of Band passkey exchange, for example using NFC or Wi-Fi to exchange keys as TKs. Currently, the ESP32 does not support this feature.`

So, is it supported or not?

chegewara, 2018-12-19:
Good question @Weijian-Espressif

Weijian-Espressif, 2018-12-20:
@copercini @chegewara, Sorry, IDF already supports OOB, the documentation has not been updated yet.

closed 2019-03-13 by Alvin1Zhang