espressif/esp-idf #2852: UDP receiving delayed often by 100ms

cranphin, 2018-12-19:
## Environment

- Development Kit:      Bang good 'Geekcreit® ESP32 Development Board'
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v3.3-beta1-91-g14d7a6345
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB or external 5V

## Problem Description

I'm trying to handle small UDP packets sent with a 50ms interval, the timing is important (it's supposed to animate RGB leds, I've done the same in the past with esp8266 with great succes.
But quite often (several times each second) one packet gets delayed around 100ms, and then the following packets quickly catch up.

### Expected Behavior

UDP packets sent with 50ms interval are received with a mostly constant interval

### Actual Behavior

Sometimes a packet is delayed ~100ms, and then the rest quickly catches up.

**If I enable debug logging, before a packet is delayed I see:
D (9060) phy_init: wifi mac time delta: 7819**

### Steps to repropduce

Have any kind of app send small (100 byte) UDP packets to the ESP32.
Handle the UDP packets on the ESP32, and print their timing.

### Code to reproduce this issue

```cpp

#define WIFI_SSID "BlaBla"
#define WIFI_PASS "BlaBlaBla"

static EventGroupHandle_t app_event_group;
const int WIFI_CONNECTED_BIT = BIT0;
static const char *TAG = "Colourful";

static esp_err_t event_handler(void *ctx, system_event_t *event) {
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        ESP_LOGI(TAG, "got ip:%s",
                 ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
        xEventGroupSetBits(app_event_group, WIFI_CONNECTED_BIT);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        esp_wifi_connect();
        xEventGroupClearBits(app_event_group, WIFI_CONNECTED_BIT);
        break;
    default:
        break;
    }
    return ESP_OK;
}

void wifi_init_sta() {
    app_event_group = xEventGroupCreate();

    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL) );

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));

    wifi_config_t wifi_config = {};
    strcpy((char*)wifi_config.sta.ssid, WIFI_SSID);
    strcpy((char*)wifi_config.sta.password, WIFI_PASS);

    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
    ESP_ERROR_CHECK(esp_wifi_start() );

    ESP_LOGI(TAG, "wifi_init_sta finished.");
    ESP_LOGI(TAG, "connect to ap SSID:%s password:%s", WIFI_SSID, WIFI_PASS);
}

static void udp_server_task(void *pvParameters) {

    const size_t buffer_size = 1024;
    const uint16_t port = 65506;

    char rx_buffer[buffer_size];
    char addr_str[128];
    int addr_family;
    int ip_protocol;

    while (1) {
        struct sockaddr_in destAddr;
        destAddr.sin_addr.s_addr = htonl(INADDR_ANY);
        destAddr.sin_family = AF_INET;
        destAddr.sin_port = htons(port);
        addr_family = AF_INET;
        ip_protocol = IPPROTO_IP;
        inet_ntoa_r(destAddr.sin_addr, addr_str, sizeof(addr_str) - 1);

        int sock = socket(addr_family, SOCK_DGRAM, ip_protocol);
        if (sock < 0) {
            ESP_LOGE(TAG, "Unable to create socket: errno %d", errno);
            break;
        }
        ESP_LOGI(TAG, "Socket created");

        int err = bind(sock, (struct sockaddr *)&destAddr, sizeof(destAddr));
        if (err < 0) {
            ESP_LOGE(TAG, "Socket unable to bind: errno %d", errno);
        }
        ESP_LOGI(TAG, "Socket binded");

        int cnt = 1;
        uint32_t last = 0;
        struct sockaddr_in sourceAddr;
        socklen_t socklen = sizeof(sourceAddr);
        while (1) {

            int len = recvfrom(sock, rx_buffer, sizeof(rx_buffer), 0, (struct sockaddr *)&sourceAddr, &socklen);
            if (len < 0) {
                ESP_LOGE(TAG, "recvfrom failed: errno %d", errno);
                break;
            } else {
                uint32_t ticks = xTaskGetTickCount();
                printf( "%d,%d\n",  (ticks - last) / portTICK_PERIOD_MS, cnt++);
                last = ticks;
            }
        }

        if (sock != -1) {
            ESP_LOGE(TAG, "Shutting down socket and restarting...");
            shutdown(sock, 0);
            close(sock);
        }
    }
    vTaskDelete(NULL);
}

extern "C" void app_main() {
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    wifi_init_sta();
    xEventGroupWaitBits(app_event_group, WIFI_CONNECTED_BIT, false, true, portMAX_DELAY);
    xTaskCreatePinnedToCore(udp_server_task, "udp_server", 4096, NULL, 5, NULL, 0);  
    vTaskDelete(NULL);
}

```

## Debug Logs

```
On the lines with numbers, the left number is the delay since the last packet is received.
It should always be close to 50.

D (9060) phy_init: wifi mac time delta: 7819
D (9162) phy_init: wifi mac time delta: 96339
120,128
0,129
29,130
50,131
52,132
D (9367) phy_init: wifi mac time delta: 63719
74,133
24,134
D (9469) phy_init: wifi mac time delta: 18112
79,135
21,136
50,137
D (9674) phy_init: wifi mac time delta: 72996
132,138
1,139
18,140
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)

igrr, 2018-12-20:
Can you please try disabling Wi-Fi modem sleep and see if that makes any difference? 
`esp_wifi_wet_ps(WIFI_PS_NONE);`

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_wifi.html#_CPPv215esp_wifi_set_ps14wifi_ps_type_t

X-Ryl669, 2018-12-20:
Just a remark: to measure time, you should use `esp_timer_get_time` and not `xTaskGetTickCount` since the latter is only updated sparsely, you'll have a lot of jitter here (typically 10ms if you did not change the FreeRTOS scheduler frequency).