espressif/esp-idf #2853: How do I find out ESP32 transceiver state in BLE mode?

MegaGarynych, 2018-12-20:
Hi all,
Some time ago I have asked question
[https://github.com/espressif/esp-idf/issues/2768](url)

> We are going to use external RF power amplifier. We need to know which state ESP32 RF transceiver has currently (transmit or receive) so we could change RF amplifier mode. Does esp-idf has some functions to find out ESP32 transceiver state?

Code example from @Alvin1Zhang works as we need. But now we interested to use this solution in BLE mode.
**Does esp-idf has some functions to find out ESP32 transceiver state in BLE mode?**

Thanks,
Roman

MegaGarynych, 2018-12-24:
Hi @Alvin1Zhang, thanks for your reply.

In total, our project need contains WIFI and BLE modes at the same time with connected external RF power amplifier to increase the operation distance.

We plan to buy a small sample set of ESP32 chips for our project and need to be sure that function of  
find out ESP32 WIFI/BLE transceiver state for control exernal RF amplifier mode is present (or will implement soon) in esp-idf.

If it is imposible to use one RF power amplifier in WIFI and BLE modes at the same time, please help with some code examples for using RF power amplifier in only BLE mode (in WIFI mode it's works well). 

We have no any specific requirements on IDF version etc. 

Thanks,
Roman.

Alvin1Zhang, 2019-03-22:
@MegaGarynych Sorry that get back to you so late, would you share if your issue has been resolved? Thanks.

hdkim, 2019-04-15:
Hi

@Alvin1Zhang : I have same issue, I need a api for ESP32 transceiver state in BLE mode.

Do you have any plan for it?

Thanks

hdkim, 2019-04-18:
Hi, @Alvin1Zhang 

Is there any progress?

Alvin1Zhang, 2019-04-18:
@hdkim Hi, thanks for the feedback. Have tracked internally and will update if any. Thanks.

Alvin1Zhang, 2019-05-06:
@hdkim Hi, sorry for the late feedback. Would you please try using API esp_tx_state_out  (please make sure that PHY version >= 4090)? Thanks.

Alvin1Zhang, 2019-05-08:
@hdkim Hi, do you have time to have a try? Thanks.

hdkim, 2019-05-08:
@Alvin1Zhang Hi, Thank you for your feedback.

Now I am trying it, But I can't find any document about (esp_tx_state_out).

Can you share a link that I can refer to with me?

Thanks

Alvin1Zhang, 2019-05-08:
@hdkim Hi, thanks for the feedback. Would you please help check whether your PHY version >= 4090? Thanks.

kmmaran, 2019-08-09:
Hi @Alvin1Zhang @hdkim  do you guys find a solution for this. I have ESP32 BLE design for external RF module but can't make it work without the state info. Can you please share what is the progress for this. Thank you.

pk7639, 2022-03-24:
Hi folks, did anyone get any traction with this? I found esp_tx_state_out() in libphy.a, but I've no idea how it works...

jack0c, 2022-05-17:
It is not timely to use API to get the transceiver state. If I understand correctly,  there are two different problems:
1. switch between antennas
2. use external RF PA, maybe LNA also

For the first problem, it depends on the feedback from peer devices, we use RSSI as estimating info in a period, in this case we can use APIs to switch between antennas. There are still some small design and document issues in multi-antenna support.
For the second problem, it should be down by this, call an API to configure the GPIO control of external PA(or/and LNA), hardware can done the switch automatically.

copercini, 2022-12-07:
yeah esp_tx_state_out() exists in libphy.a, it would help a lot if Espressif add a header for this function here: https://github.com/espressif/esp-idf/blob/master/components/esp_phy/include/phy.h

I'm also interested in use Classic Bluetooth with an external power amplifier

xyzzy42, 2024-01-06:
Came across this looking for a way to know when BLE does a TX, for an RF coexistence system with an external RF device that is not Wifi or BT.

Looking at the code for ESP32-S3, `esp_tx_state_out()` has no parameters and is very simple.  It just does this:
```
void esp_tx_state_out(void) {
    tx_state_set(1);
}
```

So what does `tx_state_set()` do?  This function is also in phy lib on ESP32-S3 but I think it is in ROM on some processors.  It looks like it pokes around in four registers in the reserved `0x6000_6xxx` peripheral memory area.  Probably secret PHY registers.  There is a single argument to `tx_state_set()` and it is a 2-bit field that is put into some registers.

Maybe this is supposed configure GPIO control of external PA?  But I see nothing to control which GPIOs are used.  `esp_tx_state_out()` has no arguments, so it's not there.

xiongweichao, 2024-01-25:
Hi @xyzzy42 ,

Please provide the chip used and the idf commit. I will provide you with a patch.

Thanks

jack0c, 2024-02-07:
> Came across this looking for a way to know when BLE does a TX, for an RF coexistence system with an external RF device that is not Wifi or BT.
> 
> After disassembling the code for ESP32-S3, `esp_tx_state_out()` has no parameters and is very simple. It just does this:
> 
> ```
> void esp_tx_state_out(void) {
>     tx_state_set(1);
> }
> ```
> 
> So what does `tx_state_set()` do? This function is also in phy lib on ESP32-S3 but I think it is in ROM on some processors. It looks like it pokes around in four registers in the reserved `0x6000_6xxx` peripheral memory area. Probably secret PHY registers. There is a single argument to `tx_state_set()` and it is a 2-bit field that is put into some registers.
> 
> Maybe this is supposed configure GPIO control of external PA? But I see nothing to control which GPIOs are used. `esp_tx_state_out()` has no arguments, so it's not there. Neither `esp_tx_state_out()` nor `tx_state_set()` have any other parameters, or use any global variables, or call other functions, which could be the way the GPIO(s), if they exist, are configured. So it's still a great mystery how this can be used.

If you want ESP32S3+external RF device. Please check the external coexistence APIs https://github.com/espressif/esp-idf/blob/master/components/esp_coex/include/esp_coexist.h (@xiongweichao check why they are not in API reference). The software only set priorities on different time slots, and the coexistence arbitration will be controlled by hardware automatically. However, if you want to do the coexistence well, you still need knowledge about the external RF device. There is an ESP32S3+ESP32H2 solution as an example. Maybe @chshu can help more on this.

chshu, 2024-02-07:
The general ESP external coexistenace feature is introduced in this doc:
https://www.espressif.com.cn/sites/default/files/documentation/external_coexistence_design_en.pdf

The feature is enabled in our ESP Thread Border router solution, in order to reduce RF interference between Wi-Fi and 802.15.4 (Thread) on 2.4 GHz:
- https://github.com/espressif/esp-thread-br
- https://docs.espressif.com/projects/esp-thread-br

You may refer to this link on how to enable it: [Enable external coexistence](https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#rf-external-coexistence)

xiongweichao, 2024-02-22:
@xyzzy42  Do you want to know when BLE is in tx through the level of GPIO? If so, please provide the idf commit id you are currently using, and I will provide you with a test version. thanks！

trentp-igor, 2024-02-22:
> @xyzzy42 Do you want to know when BLE is in tx through the level of GPIO? If so, please provide the idf commit id you are currently using, and I will provide you with a test version. thanks！

Yes, this would useful, thank you.  I'm using 8171b22c40f39a75cd0db62165e7ef017f78c572 right now.

xyzzy42, 2024-02-22:
> @xyzzy42 Do you want to know when BLE is in tx through the level of GPIO? If so, please provide the idf commit id you are currently using, and I will provide you with a test version. thanks！

Yes, that would useful, thank you.  I'm using commit 8171b22c40f39a75cd0db62165e7ef017f78c572 right now.

xiongweichao, 2024-02-23:
@xyzzy42  Please try this patch to see if it meets your needs. [libbtdm_app.a](https://github.com/espressif/esp-idf/files/14382205/libbtdm_app_2b80152.txt) patch:$IDF_PATH/components/bt/controller/lib_esp32c3_family/esp32s3
 For usage instructions, please refer to [Wi-Fi Multiple Antennas](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-multiple-antennas).  BTW, some GPIOs may not be used.
```
// BT antnenna diversity addition. This function is called after Bluetooth initialization. 
static int configure_antennas(void)
{
    esp_bt_ant_gpio_config_t gpio_config = {0};
    gpio_config.gpio_cfg[0].gpio_select= 1;
    gpio_config.gpio_cfg[0].gpio_num= 18;
    gpio_config.gpio_cfg[1].gpio_select = 1;
    gpio_config.gpio_cfg[1].gpio_num = 17;
    esp_err_t err = esp_bt_set_ant_gpio(&gpio_config);
    if (err) {
        printf("%s Setting antenna GPIO failed.\n", __func__);
        return err;
    }
    esp_bt_ant_config_t ant_config = {
        .rx_ant_mode = ESP_BT_ANT_MODE_ANT0,
        .rx_ant_default = ESP_BT_ANT_ANT0,
        .tx_ant_mode = ESP_BT_ANT_ANT1,
        .enabled_ant0 = 1,
        .enabled_ant1 = 2
    };
    err = esp_bt_set_ant(&ant_config);
    if (err) {
        printf("%s Setting antenna failed.\n", __func__);
    }

    return err;
}
```

[set_ant_patch_20240223.txt](https://github.com/espressif/esp-idf/files/14382200/set_ant_patch_20240223.txt)
[libbtdm_app_2b80152.txt](https://github.com/espressif/esp-idf/files/14382205/libbtdm_app_2b80152.txt)

xiongweichao, 2024-04-09:
> @xyzzy42 Please try this patch to see if it meets your needs. [libbtdm_app.a](https://github.com/espressif/esp-idf/files/14382205/libbtdm_app_2b80152.txt) [...]

This feature has been merged into the [master](https://github.com/espressif/esp-idf/tree/master) branch. Reference documentation https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/phy.html

closed 2025-03-14 by Alvin1Zhang