espressif/esp-idf #2854: LL_CONNECTION_PARAMS_REQ while TX/RX BLE data causes link loss

jakkra, 2018-12-20:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32]
- IDF version f11ac037c4958fc68e67a50134a0ce9a1634008c
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description
Setup:
Central ESP32
Peripheral nRF52832

Central connects to peripheral and data transmission is started, the connection interval is 7.5ms.
After about ~4 seconds the peripheral sends a l2cap: Connection Parameter Update Request:
[image]

1. Which is accepted and the Master/Central/ESP32 sends a LL_CONNECTION_PARAM_REQ with min/max connection interval set to 50ms.
2. Peripheral responds with LL_UNKOWN_RSP (Not sure why it sends UNKNOWN_RSP, if it requested it, it should support it?)
3. Connection times out.
[image]

If we compare the above scenario when we are NOT sending data stright after connection estblished and instead waits until after the LL_CONNECTION_PARAM_REQ it looks like this:
[image]

As you can see in this scenario a LL_CONNECTION_UPDATE_IND is sent from the master. This is not sent if we are sending data while the negotiation is in progress.

Starting TX/RX after LL_CONNECTION_UPDATE_IND is sent works fine, connection is not lost.

If the initial connection interval is anything else than 7.5ms this problem does not seem to happen. If we initiate the connection with 8.75ms connection interval there is no problem. 

### Expected Behavior
No link loss while transmitting data while LL_CONNECTION_PARAM_REQ is negotiated.
### Actual Behavior
Link loss
### Steps to repropduce

See problem description.

## Debug Logs

## Other items if possible
Attached are the air traces for both scenarious described above.
[sniffer_logs.zip](https://github.com/espressif/esp-idf/files/2699039/sniffer_logs.zip)

Weijian-Espressif, 2018-12-25:
@jakkra , The server does not send response to master, the master has been retransmitting, causing the Bluetooth to disconnect. Data transmission causes the connection parameter update process to be delayed, which may cause the server timer to time out and the Bluetooth to be disconnected.

jakkra, 2019-01-16:
Why shouldn't this work while there is traffic? Shoudn't the LL pause transmision if it's necessary?

jakkra, 2019-02-04:
Thanks a lot, now fixed by Espressif and should soon be on GitHub.

closed 2019-02-04 by jakkra