espressif/esp-idf #2855: Sending HCI_Inquiry_Cancel while doing Inquiry sometimes causes crash

jakkra, 2018-12-20:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32]
- IDF version f11ac037c4958fc68e67a50134a0ce9a1634008c
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description
If we have both Bluetooth and WiFi enabled and doing inquiry we see every ~25-50 inquiry this crash:
```
0x4008444f: r_assert_err at ??:?
0x4003b0ae: ?? ??:0
0x40085e46: r_ld_fm_frame_isr at ??:?
0x40088386: r_rwbt_isr at ??:?
0x40088ec3: r_rwbtdm_isr_wrapper at intc.c:?
0x40082581: _xt_lowint1 at C:/Users/jakob.krantz/Documents/nina-esp/component/cb_rtsl_esp32/esp-idf/components/freertos/xtensa_vectors.S:1105
0x4000bfed: ?? ??:0

```
And this log: `ASSERT_ERR((EM_BT_RXDESC_OFFSET + ld_env.curr_rxdesc_index * REG_EM_BT_RXDESC_SIZE) == bt_et_currentrxdescptr_currentrxdescptr_getf()), in ld_inq.c at line 278`

The problem does not appear if WiFi is not initialized.

### Expected Behavior
No crash
### Actual Behavior
Crash
### Steps to repropduce
Have not tried to reproduce using anything other than our application.
But use our sdkconfig. Init WiFi and Bluetooth.
Start Inquiry with timeout something like 40s and then after 5s cancel the inquiry.
Repeat until it hopefully crashes.

## Debug Logs

## Other items if possible
[sdkconfig_debug.txt](https://github.com/espressif/esp-idf/files/2699068/sdkconfig_debug.txt)

HCI log only includes one Inquiry but you can see all Inquiry parameters and so on. 
Problem occurs when just repeating Inquiry start and Cancel.
[hci_log.zip](https://github.com/espressif/esp-idf/files/2699153/hci_log.zip)

blueMoodBHD, 2019-01-25:
Hi @jakkra , We have find the bug and fixed it, and it will be merged into ESP-IDF, soon. Glad to cooperate with you.

jakkra, 2019-02-04:
@blueMoodBHD Thanks alot, I'll close this issue now, as it's also tracked on other medium.

closed 2019-02-04 by jakkra