espressif/esp-idf #2856: BT Controller responds Connection Limit Exceeded (0x09) on 7th Classic connect

jakkra, 2018-12-20:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32]
- IDF version f11ac037c4958fc68e67a50134a0ce9a1634008c
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description
On the 7:th connect we get error from Espressif BT Controller that the maximum amount of classic links is reached. I want to remember that I got 7 links connected when I tried sometime before. Now when I test I can initiate and connect to 6 devices before Connection Limit Exceeded (0x09) however we still allow one device to connect to us, making the total number of connections 7. 

What is the expected behaviour? Shouldn't 7 Classic links be possible?

In the attached HCI log you can see that we only have 6 Classic Links connected, and in the end I inititate a connection from an other device, making the total number of connections 7.

### Expected Behavior
Possible to connect to 7 BT Classic devices
### Actual Behavior
When trying to connect to the 7th device we get error code: Connection Limit Exceeded (0x09).

### Steps to repropduce
Try connect to 7 BT Classic devices.

## Debug Logs

## Other items if possible
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2699176/sdkconfig.txt)

HCI log where we get error code when trying to connect to the 7th device, but we still allow one device to connect to us.
[bt_classic_7_links_failure.zip](https://github.com/espressif/esp-idf/files/2699288/bt_classic_7_links_failure.zip)

blueMoodBHD, 2018-12-21:
Hi @jakkra,  The 7th acl connect is used for inquiry scan, if you want to connect to the 7th device, you should set scan mode as no discoverable and no connectable, with this code
```
esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_NONE);
```

jakkra, 2018-12-21:
@blueMoodBHD Works like a charm, thank you!

closed 2018-12-21 by jakkra