espressif/esp-idf #2857: Ota update

DavideTomasella, 2018-12-20:
I have some problem with Ota update of my Esp32 board.
I have a complex project made of different files. When I send a new version to the board, the running program seems to change only when I made some changes in the main project file.
The modifies in the other file are not always applied. 
Is it a correct behavior?

Thanks, 
Davide Tomasella

eagi223, 2018-12-20:
When you're using OTA updates, your board should be downloading a binary file of the new firmware. The binary file comes from all your project and all the necessary ESP-IDF files being compiled, so that doesn't make much sense that only changes from the main file are being applied. I've got a few projects with around 30 source files (not including the ESP-IDF files) and when I perform an OTA update, all the new changes are there. My understanding is that the OTA update process does not clear the NVS flash memory, the boot loader, the partition table or the other application binaries flashed onto the device. After successfully downloading and writing an OTA update to flash, a flag is set and the boot loader then knows to boot from the partition in memory where the latest OTA update was stored.

This doesn't seem to me like it could be an issue with the OTA functionality. Some things that come to mind are:
-Something is happening so that you are not linking and/or compiling some of your files correctly, so the binary image that you are OTA'ing to doesn't actually have the changes you've made. If this is the case though, flashing the board directly would give you the same result as OTA'ing.
-The OTA process is failing for some reason and you don't actually have any of the changes you've made.
-The changes that you're expecting to see are in the form of data stored in the nvs partition, which wouldn't get updated unless your new firmware manually updates it.

My company is using the ESP OTA api for products that are in customers hands and it hasn't failed us yet. If this doesn't help, maybe someone else will have some better ideas. Maybe sharing your code (if possible) would illuminate the issue better.

DavideTomasella, 2018-12-20:
Thank you very much.
I wanted to exclude the possibility of an ota error in the board. Do you also know how the board can recognize when the avaible binary file is the same of the running one? It uses a checksum algorithm?

igrr, 2018-12-21:
Hi @DavideTomasella, please refer to examples/system/ota/native_ota_example, it shows how to get the version of running application and the downloaded application:

https://github.com/espressif/esp-idf/blob/78487123bf80852f4ce11c788a3ee7f44b873833/examples/system/ota/native_ota_example/main/native_ota_example.c#L185-L192

The example only displays both versions. In your application, you may instead cancel the update if versions are the same.

In practice, applications will usually check for update using some server-side API, providing current version and some unique identifier of the board. Server side logic will decide if an update needs to be performed, and respond with a URL pointing to the update binary.

closed 2019-03-01 by igrr