espressif/esp-idf #2858: WPA2-Enterprise PEAP-MSCHAPv2: "Method private structure allocated failure"

daniol, 2018-12-21:
## Environment

- Development Kit:  platformIO - espressif32
- Module or chip used: ESP32
- IDF version: 3.1.1, replacing the folder components\wpa_supplicant from the current master
- Operating System: Windows 7 x64
- Power Supply: USB
- Router: TL-WR841N
- Radius Server: TekRADIUS. User configuration: see https://pasteboard.co/HSIAZDR.png
- Server certificate: https://www.pastefile.com/QEdT1N
- Debug mode "verbose"

## Problem Description

I try to establish a connection to a WPA2-Enterprise with PEAP-EAP-MSCHAPv2 network (user/password authentication). The access point is using TekRADIUS as radius server. The connection on ESP32 doesn't work, it fails after some time with code 204 (handshake timeout).

### Actual Behavior

The connection with ESP32 doesn't work
According to the logs:

```
wpa: TLSv1: Record Layer - Failed to initialize HMAC
wpa: TLSv1: Failed to create a record
wpa: Method private structure allocated failure
```

### Expected Behavior

The connection should work, as my Windows computer and my Android phone does with no problems.
See the Windows network configuration which works: https://pasteboard.co/HSIBhWp.png
On Android I don't set up any certificate, just user/password and it works.
ESP32 shouldn't initialize TLS, as this is a PEAP-EAP-MSCHAPv2 (TTLS) connection.

### Code to reproduce this issue

See https://gist.github.com/daniol/6c489f8c5d7836f55c1ea2b115cac3a3

## Debug Logs

ESP32-Log: https://pastebin.com/QsxpdiTJ
TekRADIUS-Log: https://pastebin.com/P3WUGyXw

### Remarks

I tried everything, including the following modifications, without success:
- Removing CA certificate (without esp_wifi_sta_wpa2_ent_set_ca_cert) - it gives another error
- Removing call to esp_wifi_sta_wpa2_ent_set_new_password
- Removing server certificate from TekRADIUS - the router or TekRADIUS sends a "localhost" certificate

daniol, 2018-12-21:
I attach also a log of TekRADIUS of the successful Windows 7 connection: https://pastebin.com/MB83M4Uy

negativekelvin, 2018-12-22:
Does it also happen in master or only if you backport?

daniol, 2019-01-07:
> 
> 
> Does it also happen in master or only if you backport?

Both

sagb2015, 2019-09-27:
Apologies for late reply. Do you still this issue with current master? Or can you point the master commit that you used?

sagb2015, 2020-04-17:
@daniol  Closing because of lack of feedback. Feel free to open if you still see the issue.

closed 2020-04-17 by sagb2015