espressif/esp-idf #2863: HID example triggers driver error in Windows, does not work

hansmbakker, 2018-12-22:
## Environment

- Development Kit:      [DOIT ESP32 DEVKIT V1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version : 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         PlatformIO (https://github.com/platformio/platform-espressif32/commit/caadff09bc91c4d80076265de6134e68810824b0)
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Windows
- Power Supply:         USB

## Problem Description
I'm trying the HID example at https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/ble_hid_device_demo. Windows recognizes it as a HID device and I can pair with it.

The problem is that Windows sees there is something wrong with the device and does not start the driver.

### Expected Behavior
Device is recognized correctly by Windows and HID events work properly

### Actual Behavior

[image]

Error message in `Bluetooth Low Energy GATT-compatible HID device`:
```
Device BTHLEDevice\{00001812-0000-1000-8000-00805f9b34fb}_3c71bf4ce912\8&3b69fc53&14&002d was not migrated due to partial or ambiguous match.

Last Device Instance Id: USB\VID_045E&PID_0040\7&1ca76ca2&0&2
Class Guid: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
Location Path: 
Migration Rank: 0xF000FFFFFFFFF122
Present: false
Status: 0xC0000719

```

### Steps to repropduce

Use bug reproduction repository:
https://github.com/hansmbakker/EspIdfHidBugRepro

or follow these steps
1. Installl Platform.IO
2. Create a project.
3. Use `https://github.com/platformio/platform-espressif32.git` as `platform` in `platformio.ini`
4. Use `espidf` as `framework` in `platformio.ini`
5. Add `monitor_speed = 115200` in `platformio.ini`
6. Copy the `.c` and `.h` files from https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/ble_hid_device_demo/main to the `src` folder
7. Click `Upload and Monitor`

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5696
ho 0 tail 12 room 4
load:0x40078000,len:7768
ho 0 tail 12 room 4
load:0x40080000,len:5892
entry 0x40080318
␛[0;32mI (31) boot: ESP-IDF 3.30101.0 2nd stage bootloader␛[0m
␛[0;32mI (32) boot: compile time 00:49:54␛[0m
␛[0;32mI (40) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (41) boot: SPI Speed      : 40MHz␛[0m
␛[0;32mI (41) boot: SPI Mode       : DIO␛[0m
␛[0;32mI (44) boot: SPI Flash Size : 4MB␛[0m
␛[0;32mI (49) boot: Partition Table:␛[0m
␛[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (74) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (82) boot: End of partition table␛[0m
␛[0;32mI (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1f5c8 (128456) map␛[0m
␛[0;32mI (140) esp_image: segment 1: paddr=0x0002f5f0 vaddr=0x3ffc0000 size=0x00a20 (  2592) load␛[0m
␛[0;32mI (142) esp_image: segment 2: paddr=0x00030018 vaddr=0x400d0018 size=0x6d458 (447576) map␛[0m
␛[0;32mI (305) esp_image: segment 3: paddr=0x0009d478 vaddr=0x3ffc0a20 size=0x029cc ( 10700) load␛[0m
␛[0;32mI (309) esp_image: segment 4: paddr=0x0009fe4c vaddr=0x3ffc33ec size=0x00000 (     0) load␛[0m
␛[0;32mI (312) esp_image: segment 5: paddr=0x0009fe54 vaddr=0x40080000 size=0x00400 (  1024) load␛[0m
␛[0;32mI (321) esp_image: segment 6: paddr=0x000a025c vaddr=0x40080400 size=0x108b0 ( 67760) load␛[0m
␛[0;32mI (357) esp_image: segment 7: paddr=0x000b0b14 vaddr=0x400c0000 size=0x00000 (     0) load␛[0m
␛[0;32mI (357) esp_image: segment 8: paddr=0x000b0b1c vaddr=0x50000000 size=0x00000 (     0) load␛[0m
␛[0;32mI (374) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (374) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (375) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (379) cpu_start: Starting app cpu, entry point is 0x400813d8␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (389) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (396) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM␛[0m
␛[0;32mI (402) heap_init: At 3FFCCF08 len 000130F8 (76 KiB): DRAM␛[0m
␛[0;32mI (408) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (415) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (421) heap_init: At 40090CB0 len 0000F350 (60 KiB): IRAM␛[0m
␛[0;32mI (427) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (110) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (135) BTDM_INIT: BT controller compile version [7b0770a]
␛[0m
␛[0;32mI (135) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
␛[0;32mI (205) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0␛[0m
␛[0;32mI (495) HID_LE_PRF: esp_hidd_prf_cb_hdl(), start added the hid service to the stack database. incl_handle = 40␛[0m
␛[0;32mI (495) HID_LE_PRF: hid svc handle = 2d␛[0m
␛[0;32mI (505) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (505) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (515) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 ␛[0m
␛[0;32mI (525) gpio: GPIO[27]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 ␛[0m
␛[0;32mI (16205) HID_LE_PRF: HID connection establish, conn_id = 0␛[0m
␛[0;32mI (16205) HID_DEMO: ESP_HIDD_EVENT_BLE_CONNECT␛[0m
␛[0;32mI (16445) HID_DEMO: remote BD_ADDR: c49dedb61403␛[0m
␛[0;32mI (16445) HID_DEMO: address type = 0␛[0m
␛[0;32mI (16445) HID_DEMO: pair status = success␛[0m
␛[0;32mI (17535) HID_DEMO: Send the volume␛[0m
␛[0;32mI (22005) HID_DEMO: ESP_HIDD_EVENT_BLE_DISCONNECT␛[0m
```

hansmbakker, 2018-12-22:
Other people have the same issue, see https://esp32.com/viewtopic.php?f=13&t=4122&p=32683&hilit=hid#p32683

hansmbakker, 2019-01-11:
@Alvin1Zhang do you have any update on this?

Alvin1Zhang, 2019-02-09:
@hansmbakker Hi, sorry for the delay, we have been now on the Chinese traditional holiday -- New Spring Festival. Will get back to you when we are in office after February 10th. Thanks.

Marc-Aurele, 2019-02-11:
Hi,
I'm facing the same issue since i want only a consumer device report map in my code. I don't want my code working with windows especially but i have to assume that if it is working with windows it is more likely to work with a lot of devices so that's why i have tested it on windows and i can see exactly the same error. But i have also tested it with other devices and i don't have the same behavior, especially with my xiaomi phone ...

Yulong-espressif, 2019-02-21:
@hansmbakker We have every Windows 10 test environment, so can you delete the Vendor Report in the report map in HID and try again?

hansmbakker, 2019-02-21:
Do you mean to delete these lines?

https://github.com/espressif/esp-idf/blob/abea9e4c02bb17e86298aec4e299780399e4789f/examples/bluetooth/ble_hid_device_demo/main/hid_device_le_prf.c#L172-L181

hansmbakker, 2019-02-21:
Also, does this mean that you cannot reproduce the bug?

Yulong-espressif, 2019-02-22:
@hansmbakker Yes, delete these lines and test again. We can't find the test environment of window 10 now, it is not impossible to reproduce.

Weijian-Espressif, 2019-03-06:
@hansmbakker , we have reproduced the issue, Window10 do not support vendor report. we will fix the bug this week.

Weijian-Espressif, 2019-03-12:
@hansmbakker, please try 39c6b2f90495e7f53ac712c45e2f3b9afa97d150 or c1e50c8a2727e9301558c360d9a82f9ea0486b3b

Alvin1Zhang, 2019-03-13:
@hansmbakker Feel free to reopen if the issue is still not fixed. Thanks.

closed 2019-03-13 by Alvin1Zhang

hansmbakker, 2019-03-13:
@Weijian-Espressif @Alvin1Zhang I’ll test it soon and I’ll let you know the outcome

c-hamster, 2019-11-30:
I have the same question!

c-hamster, 2019-11-30:
@hansmbakker  do you solve the problem ?