espressif/esp-idf #2865: How to restore and save the BLE mesh state in between reboots

eminarcissus, 2018-12-22:
What kind of information need to stored and restored between reboots? Currently in example we have to re-provision all the device, is there any doc on how to save the state and restore? 
I've noticed there's model keys/group attribute, don't know only save those are sufficient or not.

chegewara, 2018-12-22:
This info suppose to be stored by mesh stack, but its not implemented yet. You need to store all provisioned data, but to be honest im not sure you can grab those info from application level.

PS could you add `[ble mesh]` in issue title for issues related to this topic, thanks

eminarcissus, 2018-12-22:
Got it,thx very much :)

Isl2017, 2018-12-24:
@eminarcissus, We will release a new BLE Mesh Version soon. It will include many new features, including `restoring the provision data of the node device`, `fast provisioning`, `wifi/ble_mesh coexistence`, etc... Please stay tuned!

Isl2017, 2019-05-20:
New version has been released [here](https://github.com/espressif/esp-idf/tree/ble_mesh_release/esp-ble-mesh-v0.6.1)

closed 2019-05-20 by Isl2017