espressif/esp-idf #2866: Core  0 panic'ed in mdns_parse_packet

GermanSheepDog, 2018-12-22:
## Environment

- Development Kit:      M5Stack
- Kit version (for WroverKit/PicoKit/DevKitC): -
- Module or chip used:  ESP32
- IDF version ESP-IDF v3.3-beta1-4-g23b6d40c5 and versions before!
- Build System:         Make
- Compiler version Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
                                Compiler version: 5.2.0
- Operating System:    macOS
- Power Supply:          USB

## Problem Description

I realized a small SmartHome ESP32-controller with connecting to Apple HomeKit. Homekit can read the values from ESP32 but every time the mdns-service crashed after a time.

### Expected Behavior
Here is the crash report: 
Important: The messages started with mdns_parse_packet are my new message!

> mdns_parse_packet 5-X1
mdns_parse_packet 5-K1
mdns_parse_packet 5-K2
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401b9463  PS      : 0x00060530  A0      : 0x801b9dc9  A1      : 0x3ffedd20  
0x401b9463: mdns_parse_packet at /Users/my/esp/esp-idf/components/mdns/mdns.c:4283
> A2      : 0x00000000  A3      : 0x00000010  A4      : 0x00000098  A5      : 0x3ffd5898  
A6      : 0x3ffd5b8d  A7      : 0x3ffd5c25  A8      : 0x801b9455  A9      : 0x3ffedd00  
A10     : 0x0000000a  A11     : 0x3ffd5b8d  A12     : 0x00000098  A13     : 0x00000001  
A14     : 0x00060120  A15     : 0x00000010  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  EXCVADDR: 0x00000004  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffa  
> Backtrace: 0x401b9463:0x3ffedd20 0x401b9dc6:0x3ffedda0 0x401b9e23:0x3ffeddd0 0x400918a1:0x3ffede00
0x401b9463: mdns_parse_packet at /Users/my/esp/esp-idf/components/mdns/mdns.c:4283
> 0x401b9dc6: _mdns_execute_action at /Users/my/esp/esp-idf/components/mdns/mdns.c:4283
> 0x401b9e23: _mdns_service_task at /Users/my/esp/esp-idf/components/mdns/mdns.c:4283
> 0x400918a1: vPortTaskWrapper at /Users/my/esp/esp-idf/components/freertos/port.c:403
>
> CPU halted.

### Actual Behavior

### Steps to reproduce and Code to reproduce this issue

Every time I started the app it crashed.

Here are the crash part in mdns.c:
```
void mdns_parse_packet(mdns_rx_packet_t * packet)
{
   static mdns_name_t n;
 ...
   } else if (ours) {
                    if (parsed_packet->questions && !parsed_packet->probe) {
                        _mdns_remove_parsed_question(parsed_packet, type, service);
                        continue;
                    }
                    //detect collision (-1=won, 0=none, 1=lost)
                    int col = 0;
                    if (clas > 1) {
                        col = 1;
                    } else if (!clas) {
                        col = -1;
                    } else {
          printf("HERE mdns_parse_packet HERE!\n");
          printf("clas=%d len=%d\n", clas, data_len);
          printf("HEXDUMP: ");
          for (int j=0; j<data_len; j++)
            printf("%02x ", (uint8_t) *(data_ptr+j) );
          printf("\n\n ASCIIDUMP:>");
          for (int j=0; j<data_len; j++)
            printf("%c", (uint8_t) *(data_ptr+j) );
          printf("<\n\n");
          /  goto clear_rx_packet; //dirty workaround
                        col = _mdns_check_txt_collision(service->service, data_ptr, data_len);
                    }
                    if (col && !_mdns_server->interfaces[packet->tcpip_if].pcbs[packet->ip_protocol].probe_running) {
          printf("mdns_parse_packet NEXT STEP\n");
                        do_not_reply = true;
                        _mdns_init_pcb_probe(packet->tcpip_if, packet->ip_protocol, &service, 1, true);
                    } else if (ttl > 2250 && !col && !parsed_packet->authoritative && !parsed_packet->probe && !parsed_packet->questions && !_mdns_server->interfaces[packet->tcpip_if].pcbs[packet->ip_protocol].probe_running) {
                        _mdns_remove_scheduled_answer(packet->tcpip_if, packet->ip_protocol, type, service);
                    }
                }
```

## Debug Logs
The output before the error occurred is:
```
mdns_parse_packet START
HERE mdns_parse_packet HERE!
clas=1 len=152
HEXDUMP: 97 4c 6f 78 4c 49 56 45 3a 20 2a 34 44 36 39 36 45 36 39 37 33 36 35
37 32 37 36 36 35 37 32 20 31 39 32 2e 31 36 38 2e 30 2e 36 32 3a 38 30 20 35 
30 34 46 39 34 31 30 31 46 33 38 20 31 30 2e 30 2e 39 2e 32 34 20 49 50 76 36
3a 2c 30 33 64 38 30 32 34 31 3a 39 30 30 30 38 32 32 2f 4f 2c 30 34 64 38 30 
37 62 33 3a 39 30 30 30 39 31 35 2f 4f 2c 30 36 64 38 30 31 63 32 3a 39 30 30 
30 38 32 32 2f 4f 2c 30 31 64 38 31 66 31 31 3a 39 30 30 30 38 32 32 2f 4f 

 ASCIIDUMP: ?LoxLIVE: *4D696E69736572766572 192.168.0.62:80 504F94101F38 
10.0.9.24 IPv6:,03d80241:9000822/O,04d807b3:9000915/
O,06d801c2:9000822/O,01d81f11:9000822/O

mdns_parse_packet NEXT STEP
```
## Other items if possible

The message which produced the error is a message from an uninvolved device called "loxone".
My dirty workaround is to stop the processing an goto  "clear_rx_packet".

david-cermak, 2018-12-23:
Hi @GermanSheepDog,

Thanks for reporting this issue. It looks like the "loxone" packet contains either standalone TXT or a TXT entry ahead of PTR/STR records and mdns parser is not resistant to this condition and needs to be fixed.

Just to confirm it is the case can you try to apply this condition
```
if (service) {
    col = _mdns_check_txt_collision(service->service, data_ptr, data_len);
}
```
to the original call of `_mdns_check_txt_collision`?

Also it would be helpful if you can provide entire packet that is causing this issue. Could be as pcap (if you can do wireshark in your network) or an output printf the same way you provided the section above. It is possible to inject the provided section only, but that causing crashes only to releases 3.1 and 3.2, interestingly it works with latest master/3.3 which means we miss something. It'd be highly appreciated if you could prove the entire packet (=udp payload only).

Thanks for your help,
David

GermanSheepDog, 2019-01-13:
Hello David,

thanks for your great answer and sorry for the long time I needed for an answer. The main reason was Christmas time.

My results:
- I implemented the the condition and it worked fine for me! 
- Every time I paired a new home kit device the "loxone" packet was sent.
- The wireshark output:
[image]

I saved the pcap-file and I can send you this file directly if you like.

Thanks for your help,
"GermanSheepDog"

GermanSheepDog, 2019-01-13:
Reopen of issue!

david-cermak, 2019-01-21:
Hi @GermanSheepDog 
Thanks for the followup. No need for the pcap (screenshot you've provided was just enough).
Please keep the issue opened, will be closed automatically when fixed.

commit af48977f (2019-05-10): mdns:  fixed mdns crashing on reception of txt packet without a corresponding service
closes #2866
closed 2019-05-10 by igrr