espressif/esp-idf #2876: Firmware crash -Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled Core 0 register dump: PC      : 0x4010c033  PS      : 0x00060630  A0      : 0x8010c264  A1      : 0x3ffc3c20  0x4010c033: pppos_input at esp-idf/components/lwip/netif/ppp/pppos.c:496

sijokwork, 2018-12-24:
## Environment

- Development Kit:      [ESP32-Custom Board-with GSM Module]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):v3.1
   - Operating System:     [Linux]
- Power Supply:         [external 5V]

## Problem Description
We are using ESP32 Module With GSM module on custom board for a IOT project ,the device would connect to MQTT broker and publish ,subscribe data with specified interval ,but some time i could see the device reset  because of gsm module network disconnect .
//Detailed problem description goes here.
For network connectivity i am using gsm module compatible with sim800 type. and the library using the connectivity for the gsm module is @loboris loboris pppos library. 
so far i am not using the Jtag for debugging but used the UART .

Our IOT device would initiate connection through GSM module ,(we are not using wifi and not enabled it) and publish data through MQTT ,some time the network issue the modem get disconnected 
ppp status callback function, ppp_status_cb with PPPERR_USER would call  in libGSM.c. 
because this reinitializing the pppapi_connect(ppp, 0); causing the below error 

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4010c033  PS      : 0x00060630  A0      : 0x8010c264  A1      : 0x3ffc3c20 
0x4010c033: pppos_input at ESP32Workspace/esp-idf/components/lwip/netif/ppp/pppos.c:496

A2      : 0x00000000  A3      : 0x3ffcc738  A4      : 0x000000af  A5      : 0x00000014 
A6      : 0x00000000  A7      : 0x3ffd24d2  A8      : 0x8010ea31  A9      : 0x3ffc3bf0 
A10     : 0x3ffc22a8  A11     : 0x3ffb2538  A12     : 0x3ffc230c  A13     : 0x00000000 
A14     : 0x3ffc237c  A15     : 0x00002522  SAR     : 0x00000004  EXCCAUSE: 0x0000001c 
EXCVADDR: 0x0000001c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff 

Backtrace: 0x4010c033:0x3ffc3c20 0x4010c261:0x3ffc3c40 0x400fc222:0x3ffc3c60
0x4010c033: pppos_input at  ESP32Workspace/esp-idf/components/lwip/netif/ppp/pppos.c:496

0x4010c261: pppos_input_sys at ESP32Workspace/esp-idf/components/lwip/netif/ppp/pppos.c:464 (discriminator 3)

0x400fc222: tcpip_thread at  ESP32Workspace/esp-idf/components/lwip/api/tcpip.c:481

 
 ## Debug Logs
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4010c033  PS      : 0x00060630  A0      : 0x8010c264  A1      : 0x3ffc3c20 
0x4010c033: pppos_input at ESP32Workspace/esp-idf/components/lwip/netif/ppp/pppos.c:496

A2      : 0x00000000  A3      : 0x3ffcc738  A4      : 0x000000af  A5      : 0x00000014 
A6      : 0x00000000  A7      : 0x3ffd24d2  A8      : 0x8010ea31  A9      : 0x3ffc3bf0 
A10     : 0x3ffc22a8  A11     : 0x3ffb2538  A12     : 0x3ffc230c  A13     : 0x00000000 
A14     : 0x3ffc237c  A15     : 0x00002522  SAR     : 0x00000004  EXCCAUSE: 0x0000001c 
EXCVADDR: 0x0000001c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff 

Backtrace: 0x4010c033:0x3ffc3c20 0x4010c261:0x3ffc3c40 0x400fc222:0x3ffc3c60
0x4010c033: pppos_input at  ESP32Workspace/esp-idf/components/lwip/netif/ppp/pppos.c:496

0x4010c261: pppos_input_sys at ESP32Workspace/esp-idf/components/lwip/netif/ppp/pppos.c:464 (discriminator 3)

0x400fc222: tcpip_thread at  ESP32Workspace/esp-idf/components/lwip/api/tcpip.c:481

 So i need to know the how to fix the solution ,please provide a way to fix the issue.

suda-morris, 2018-12-25:
I wonder why your modem disconnected but go into `PPPERR_USER` rather than `PPPERR_CONNECT`? Have you manually called `pppapi_close` function? Or after you entered PPP mode, you will exit to command mode from time to time? Can you enbale `CONFIG_PPP_DEBUG_ON=y` and give out more runtime information?

sijokwork, 2018-12-26:
Yes, After ppp_status_cb call back then call pppapi_close(ppp, 0);  function ,  re init the command mode for AT commands ,
This crash will not happen frequently for our device but its essential to avoid the crash for IOT device ,below i am pasting the CONFIG_PPP_DEBUG_ON=y with out error condition 

I (30510) [PPPOS CLIENT]: AT COMMAND: [AT..]
E (30820) [PPPOS CLIENT]: AT: TIMEOUT
W (30820) [PPPOS CLIENT]: Wrong response, restarting...
I (33920) [PPPOS CLIENT]: AT COMMAND: [AT..]
I (33950) [PPPOS CLIENT]: AT RESPONSE: [..+CPIN: READY....RDY....+CFUN: 1....SMS Ready....Call Ready..AT...OK..]
I (34050) [PPPOS CLIENT]: AT COMMAND: [AT+CIMI..]
I (34080) [PPPOS CLIENT]: AT RESPONSE: [AT+CIMI...404809911000126....OK..]
I (39190) [PPPOS CLIENT]: AT COMMAND: [ATE0..]
I (39220) [PPPOS CLIENT]: AT RESPONSE: [ATE0...OK..]
I (39320) [PPPOS CLIENT]: AT COMMAND: [AT+CFUN=1..]
I (39350) [PPPOS CLIENT]: AT RESPONSE: [..OK..]
I (40450) [PPPOS CLIENT]: AT COMMAND: [AT+CNMI=0,0,0,0,0..]
I (40480) [PPPOS CLIENT]: AT RESPONSE: [..OK..]
I (40580) [PPPOS CLIENT]: AT COMMAND: [AT+CPIN?..]
I (40610) [PPPOS CLIENT]: AT RESPONSE: [..+CPIN: READY....OK..]
I (40710) [PPPOS CLIENT]: AT COMMAND: [AT+CSQ..]
I (40750) [PPPOS CLIENT]: AT BAD RESPONSE: [..+CSQ: 20,0....OK..]
I (40750) [PPPOS CLIENT]: AT : [AT+CSQ
I (40750) [PPPOS CLIENT]: SIGNAL STRENGTH: [..+CSQ: 20,0....OK..]
I (40850) [PPPOS CLIENT]: AT COMMAND: [AT+CREG?..]
I (40890) [PPPOS CLIENT]: AT RESPONSE: [..+CREG: 0,1....OK..] 
I (42990) [PPPOS CLIENT]: AT COMMAND: [AT+CGDCONT=1,"IP","surfnet"..]
I (43030) [PPPOS CLIENT]: AT RESPONSE: [..OK..]
I (43130) [PPPOS CLIENT]: AT COMMAND: [AT+CGDATA="PPP",1..]
I (43180) [PPPOS CLIENT]: AT RESPONSE: [..CONNECT..~.}#.!}!}!} }<}!}$}&@}#}$.#}%}&....}"}&} } } } }'}"}(}"}7.~~~]
I (44180) [PPPOS CLIENT]: GSM initialized.
ppp phase changed[1]: phase=0
ppp_connect[1]: holdoff=0
ppp phase changed[1]: phase=1
pppos_connect: unit 1: connecting
ppp_start[1]
pppos_send_config[1]: out_accm=FF FF FF FF
ppp_send_config[1]
pppos_recv_config[1]: in_accm=FF FF FF FF
ppp_recv_config[1]
ppp: auth protocols: PAP=1 CHAP=0 CHAP_MD5=0 CHAP_MS=0 CHAP_MS2=0
pppos_write[1]: len=24
ppp_start[1]: finished
pppos_input[1]: got 175 bytes
pppos_write[1]: len=32
pppos_write[1]: len=32
pppos_write[1]: len=32
pppos_input[1]: got 46 bytes
netif_set_mtu[1]: mtu=1500
pppos_send_config[1]: out_accm=0 0 0 0
ppp_send_config[1]
pppos_recv_config[1]: in_accm=0 0 0 0
ppp_recv_config[1]
ppp phase changed[1]: phase=5
pppos_write[1]: len=10
pppos_input[1]: got 45 bytes
Remote message: TTP Com PPP - Password Verified OK
PAP authentication succeeded
ppp phase changed[1]: phase=7
ccp_set[1]: is_open=1, is_up=0, receive_method=0, transmit_method=0
pppos_write[1]: len=32
pppos_write[1]: len=18
pppos_input[1]: got 31 bytes
Protocol-Reject for 0x8057 received
pppos_input[1]: got 32 bytes
pppos_write[1]: len=26
pppos_write[1]: len=14
pppos_input[1]: got 28 bytes
pppos_write[1]: len=26
pppos_input[1]: got 28 bytes
sifvjcomp[1]: VJ compress enable=0 slot=0 max slot=0
sifup[1]: err_code=0
I (50780) [PPPOS CLIENT]: status_cb: Connected
I (50780) [PPPOS CLIENT]:    ipaddr    = 10.1.0.148
I (50780) [PPPOS CLIENT]:    gateway   = 10.0.0.1
I (50790) [PPPOS CLIENT]:    netmask   = 255.255.255.255
I (50800) [PPPOS CLIENT]:    ip6addr   = ::
local  IP address 10.1.0.148
remote IP address 10.0.0.1
primary   DNS address 66.198.145.145
secondary DNS address 129.250.35.250
ppp phase changed[1]: phase=8
I (50810) [SNTP-TIME_OBTAIN_MAIN]: Time is not set yet. Connecting to GSM and getting time over NTP.
I (55820) [SNTP]: Initializing SNTP
pppos_netif_output[1]: proto=0x21, len = 58
I (55820) [SNTP-TIME_OBTAIN]: Waiting for system time to be set... (1/25)
pppos_netif_output[1]: proto=0x21, len = 58
I (56890) [ENGINE_RPM_EVENT]:  Waiting Time update 
pppos_netif_output[1]: proto=0x21, len = 58
I (57820) [SNTP-TIME_OBTAIN]: Waiting for system time to be set... (2/25)
pppos_netif_output[1]: proto=0x21, len = 58
I (59820) [SNTP-TIME_OBTAIN]: Waiting for system time to be set... (3/25)
I (61820) [SNTP-TIME_OBTAIN]: Waiting for system time to be set... (4/25)
pppos_netif_output[1]: proto=0x21, len = 58
pppos_netif_output[1]: proto=0x21, len = 58
I (63820) [SNTP-TIME_OBTAIN]: Waiting for system time to be set... (5/25)
pppos_input[1]: got 128 bytes
ppp_input[1]: ip in pbuf len=122
pppos_netif_output[1]: proto=0x21, len = 76
pppos_input[1]: got 128 bytes
ppp_input[1]: ip in pbuf len=122
pppos_netif_output[1]: proto=0x21, len = 56
pppos_input[1]: got 81 bytes
ppp_input[1]: ip in pbuf len=76
I (65820) [SNTP-TIME_OBTAIN_MAIN]: TIME SET TO Wed Dec 26 08:14:29 2018
I (65820) [SNTP-TIME_OBTAIN_MAIN]: The current date/time in  is: Wed Dec 26 08:14:29 2018
I (65830) [MQTT_EVENT_HANDLER]: Dev ID:240AC4295CD0 
D (65830) MQTT_CLIENT: MQTT client_id=ESP240AC4295CD0 
pppos_netif_output[1]: proto=0x21, len = 64
pppos_netif_output[1]: proto=0x21, len = 64
pppos_netif_output[1]: proto=0x21, len = 64
pppos_netif_output[1]: proto=0x21, len = 64

suda-morris, 2018-12-26:
@sijokwork IIRC, the `ppp_close()` function should be called before the state go into `PPPERR_USER`. And in `PPPERR_USER`, what you can call is `ppp_free()`. Please check this document for more information.
http://www.nongnu.org/lwip/2_1_x/group__ppp.html

suda-morris, 2018-12-27:
Two suggestions:
1. Wait until the ppp close operation really take effect. I mean after the ppp state go into PPPERR_USER and then you just restart everything from begging.
2. When ppp go into PPPERR_CONNECT, just call ppp_connect to reconnect it.

sijokwork, 2019-01-02:
There is the solution that worked out.
In my case we were checking the sim contact issue with  the device  we installed where will have so much vibrate issue  and ppp state go into  PPPERR_USER ,
there was no differentiate connect between PPPERR_USER and PPPERR_CONNECT , so it always call ppp_connect to reconnect it without call pppapi_pppos_create() and it crashed .

Changed to restart everything from beginning when ppp go to PPPERR_USER.

unfortunately i have landed some another issue which crashing the firmware when if i call only ppp_connect to reconnect it with more than 3-4 times  in row with connect disconnect method ,

error header: 
ppp_free[3] CORRUPT HEAP: Bad head at 0x3ffca3e8. Expected 0xabba1234 got 0x3ffcd1c8 

i think i should have free some data block which initialized with malloc in my program ! ,or is it from the ppp stack ? do you have any other suggestions ?

 thank you for your valuable time and i appreciate :).

akbarhash, 2019-01-10:
I am working on the same. can you please share the changes made in libgsm.c to make more robust.

suda-morris, 2019-01-29:
@sijokwork @akbarhash IDF now has [a new example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client) showing how to establish PPPoS connection. Feel free to try it and welcome to push your ideas and suggestions.

sijokwork, 2019-02-01:
@suda-morris  definitely i try the new example  and share the experience
thanks

Alvin1Zhang, 2019-04-18:
@sijokwork Hi, would you help share if any updates or experience for this issue? Thanks.

suda-morris, 2020-05-06:
Feel free to open a new issue if you have met any problems in our esp-modem example. We're going to close this issue.

closed 2020-05-06 by suda-morris