espressif/esp-idf #2879: ESP32 freezes on ESD

ArTyGill, 2018-12-24:
After several times I hit the pin tuned to the input (I produce ESD), ESP32 hangs and sends garbage to the UART, even watchful timers do not save. Only restart helps.
How to fix this bug?

Hardware: esp32-wrover-i, with spi RAM enabled. 
spi-flash 4MB 40Mhz.

costaud, 2018-12-25:
On which pins and at what voltage did you hit the module ?

ArTyGill, 2018-12-27:
Test on pins io 34, 35, 36, 39
[image]

ArTyGill, 2018-12-27:
Does the internal circuit contain protection against electrostatic discharge on pins?

ArTyGill, 2019-01-23:
@me-no-dev help pls!

ArTyGill, 2019-01-23:
Yes, it's good, but for this it is enough to run an empty example, and it will hang even if you just knock tweezers on the pins :)
And even the watchdog timer does not save (

ArTyGill, 2019-01-23:
Even if the pins are set up on the exit and the tightening is turned on this does not help (and these pins can only work as input).
It saves only additional capacity, but it can be a solution that will solve the problem, but for me it is important that the controller does not hang 100%.

ArTyGill, 2019-01-23:
@me-no-dev  I can provide an oscillogram of what is happening on the pin.
Or just quickly sketch what it looks like.

me-no-dev, 2019-01-23:
I am trying to reproduce this with code that just prints a message every second to uart. So far I can not reproduce. Will try to find WROVER-I to test with. What version IDF are you using? Commit?

ArTyGill, 2019-01-23:
I advise you to try to take one end of the wire with your fingers, and with the other end of the wire, knock on the pin of ESP32.

I am using arduino-esp32 without IDF.

me-no-dev, 2019-01-23:
I got one piece of WROVER-I and will try to replicate with 
```cpp
void setup() {

}

void loop() {
  printf("ping\n");
  delay(1000);
}
```

Alvin1Zhang, 2019-03-14:
@ArTyGill Hi, sorry that we tried with all different boards and modules that we found but was not able to replicate this issue.

ArTyGill, 2019-04-11:
[image]

Everything works fine if you install protection against electrostatic discharge. I think it's worth thinking about installing a TVS diode on modules with ESP 32. And you also need protection on the antenna.

closed 2019-04-24 by ArTyGill