espressif/esp-idf #2880: Connection time?

Stasikiii, 2018-12-24:
Hello,

I would like to know if the problem with a long connection has been resolved with esp32 or is this problem still present? ESP8266 (on 80mhz) is able to get a connection in 200ms while ESP32 (on 230mhz) requires 900ms or more!

I heard that this problem is caused by the fact that even if it recognizes the access point, it continues to scan other channels, which takes more than 700ms of time from it.

Is this problem solved in new versions of ESP-IDF?

Stasikiii, 2018-12-28:
Hello?

Alvin1Zhang, 2018-12-28:
@Stasikiii Hi, Stasikiii, we have been looking into this issue, will update if any progress. Thanks.

TimXia, 2019-01-08:
@Stasikiii Please try this:
**    wifi_config_t wifi_config = {
        .sta = {
            .ssid     = AP_SSID,
           .password = AP_PASS,
           .scan_method = WIFI_FAST_SCAN,
           .bssid_set = 0,
        },
    };**

Stasikiii, 2019-01-08:
@liuzfesp  @zhangyanjiaoesp 
Today i tested one more time.

AP_ENCRYPTION_OFF / Tickrate 1000 / NO_ARP / TX-POWER: 20dBi

80MHZ
Connection time - **500 - 600ms**
UDP packet (1 Byte [8 bit]) - **80 - 84ms**

160MHZ 
Connection time - **300 - 400ms**
UDP packet (1 Byte [8 bit]) - **114 - 128ms**

240MHZ
Connection time - **200 - 250ms**
UDP packet (1 Byte [8 bit]) - **9ms - 14ms**

With encryption 900 - 1200 like i said before.

esp8266@80mhz
Encryption ON Connection time - 200ms
UDP - 9 - 10ms. 
But you have to use MAC and AP_CHANNEL in config.
Without MAC it takes 1200ms.

Stasikiii, 2019-03-03:
@zhangyanjiao no any news? :(

liuzfesp, 2019-03-04:
Hi @Stasikiii this code is in reviewing status, I think @zhangyanjiaoesp will merge it into IDF this week.

Stasikiii, 2019-03-04:
This is great news! Thank.
Could you tell us a little more about the problem?

closed 2019-03-18 by igrr

Stasikiii, 2019-05-22:
@liuzfesp 
Hello. I just wanna ask about release? Was this bug fix merged in some release?

Thx in advice!

zhangyanjiaoesp, 2019-05-22:
@Stasikiii This bugfix has been merged into mater, v3.3, v3.2, v3.1

Stasikiii, 2019-05-22:
@zhangyanjiaoesp my question was wrong. 
Is this bug already fixed and can i see the result? I'm user of Arduino and I do not see any results on the latest stable version. :(

zhangyanjiaoesp, 2019-05-23:
It seems that bugfix hasn't been mergerd into the Arduino. But you can found it in the release/v3.2. de6b7d6520e1759bffb16d2084ea90c8009d067a

Stasikiii, 2019-06-26:
@zhangyanjiaoesp Hello again. It's is posseble to see good effect from this FIX while using arduino as esp-idf component?
Thx again!