espressif/esp-idf #2881: open helloworld cmakelist.txt with clion  error occurred

dtboy1995, 2018-12-25:
build log 
```log
"D:\Program Files\JetBrains\CLion 2018.1.2\bin\cmake\bin\cmake.exe" -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - NMake Makefiles" E:\esp\hello_world
-- Checking Python dependencies...
Python requirements from E:/esp/esp-idf\requirements.txt are satisfied.
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Program Files/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/Program Files/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/Program Files/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: D:/Program Files/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: D:/Program Files/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at E:/esp/esp-idf/components/soc/CMakeLists.txt:4 (include):
  include could not find load file:

    /esp32/sources.cmake
Call Stack (most recent call first):
  E:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:105 (include)
  E:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:112 (expand_component_requirements)
  E:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:112 (expand_component_requirements)
  E:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:112 (expand_component_requirements)
  E:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:112 (expand_component_requirements)
  E:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:112 (expand_component_requirements)
  E:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:112 (expand_component_requirements)
  E:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:112 (expand_component_requirements)
  E:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:215 (expand_component_requirements)

CMake Error at E:/esp/esp-idf/CMakeLists.txt:36 (include):
  include could not find load file:

    E:/esp/hello_world/cmake-build-debug/component_depends.cmake

-- Component names: 
-- Component paths: 
CMake Error at E:/esp/esp-idf/tools/cmake/kconfig.cmake:83 (list):
  list sub-command SORT requires list to be present.
Call Stack (most recent call first):
  E:/esp/esp-idf/CMakeLists.txt:62 (kconfig_process_config)

CMake Error at E:/esp/esp-idf/CMakeLists.txt:65 (include):
  include could not find load file:

    E:/esp/hello_world/cmake-build-debug/config/sdkconfig.cmake

-- IDF_VER: v3.3-beta1-132-gdb8bc3eea
CMake Error at E:/esp/esp-idf/tools/cmake/targets.cmake:34 (if):
  if given arguments:

    "NOT" "esp32" "STREQUAL"

  Unknown arguments specified
Call Stack (most recent call first):
  E:/esp/esp-idf/CMakeLists.txt:75 (idf_check_config_target)

-- Configuring incomplete, errors occurred!
See also "E:/esp/hello_world/cmake-build-debug/CMakeFiles/CMakeOutput.log".
See also "E:/esp/hello_world/cmake-build-debug/CMakeFiles/CMakeError.log".

[Failed to reload]
```

david-cermak, 2018-12-26:
Hi @dtboy1995 

There could be some issues with environmental variables on windows. Please provide an exact version of IDF you use, so we can be more specific and provide fixes.

Generally you can work around those issues by providing IDF_PATH in CLion IDE under File->Settings->Build, Execution, Deployment -> CMake; either in Environment or directly to "CMake options", for example -DIDF_PATH=\"E:/esp/esp-idf\".

Thanks!

dtboy1995, 2018-12-26:
@david-cermak **Thank you very much**. I just saw your reply. I will try to give you feedback when I go to the company

dtboy1995, 2018-12-26:
@david-cermak  thank you very much. I'm using Ubuntu instead of Windows 10. There's no problem.

dtboy1995, 2018-12-27:
@david-cermak. there is still an error after configuration according to your method. thanks...How should you configure it in Windows?Ubuntu can be used without any configuration

thanks a lot

dtboy1995, 2018-12-27:
ubuntu 

[image]

dtboy1995, 2018-12-27:
win10

[image]
[image]

dtboy1995, 2018-12-27:
- log
```
"C:\Program Files\JetBrains\CLion 2018.3.2\bin\cmake\win\bin\cmake.exe" -DCMAKE_BUILD_TYPE=Debug -DIDF_PATH=D:/msys32/home/dtboy/esp/esp-idf -G "CodeBlocks - NMake Makefiles" D:\esp\hello_world
-- Building empty aws_iot component due to configuration
CMake Error at D:/msys32/home/dtboy/esp/esp-idf/components/azure-iot/CMakeLists.txt:6 (project):
  project command is not scriptable
Call Stack (most recent call first):
  D:/msys32/home/dtboy/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:151 (include)
  D:/msys32/home/dtboy/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:159 (expand_component_requirements)
  D:/msys32/home/dtboy/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:183 (expand_component_requirements)

CMake Error at D:/msys32/home/dtboy/esp/esp-idf/tools/cmake/project.cmake:66 (include):
  include could not find load file:

    D:/esp/hello_world/cmake-build-debug/component_depends.cmake
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!

[Previous CMake output restored: 2018/12/27 10:16]

```

closed 2018-12-27 by dtboy1995

david-cermak, 2018-12-28:
Thanks @dtboy1995 for the update and good news!

dtboy1995, 2018-12-28:
Thank you very much for your help, too @david-cermak

my-dudhwala, 2023-09-22:
CMake Error at CMakeLists.txt:7 (if):
  if given arguments:

    "STREQUAL" "linux"

  Unknown arguments specified

Can any guide me regarding this please?

igrr, 2023-09-22:
@my-dudhwala in which example project are you getting this error?

zikalino, 2023-10-02:
@my-dudhwala I am not sure which CMakeList you're using, but I am able to reproduce this exact problem,

Basically you should change:

    "STREQUAL" "linux"

to:

    STREQUAL "linux"

If you place STREQUAL in quotes it's not recognized as CMake keyword and CMake will complain with "Unknown arguments specified"