espressif/esp-idf #2886: Posting an event with data alloctaes a copy of the data but does not free it

i-franzki, 2018-12-26:
## Environment

- Development Kit:      ESP32-PICO-Kit
- Module or chip used:  ESP32-PICO-D4
- IDF version: v3.3-beta1-134-g1023ff73f
- Build System:  Make
- Compiler version: 1.22.0-80-g6c4433a5
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

When posting an even uisng esp_event_post() or esp_event_post_to() with data (i.e. parameter event_data != NULL and event_data_size > 0), a copy of the event data is allocated by post_instance_create() and attached to the post which is then sent to the event queue.

That copy of the event data is never freed, except when sending to the event queue has failed. 

I would expect that the event loop (i.e. esp_event_loop_run()) would free that copy of the event data after passing it to all registered event handlers, i.e at the end of the while loop. But this does not seem to happen. Instead the copy remains allocted, causing a memory leak if many such events are posted.

### Expected Behavior
Event data to be freed in event loop (i.e. esp_event_loop_run()) after passing it to all registered event handlers, i.e at the end of the while loop.

### Actual Behavior
Event data is not freed, leading to a memory leak.

### Steps to repropduce

Post an event using esp_event_post() or esp_event_post_to() with data.
E.g. like it is illustrated in example 'system/esp_event/default_event_loop': 
https://github.com/espressif/esp-idf/tree/master/examples/system/esp_event/default_event_loop

### Code to reproduce this issue
See example 'system/esp_event/default_event_loop':
https://github.com/espressif/esp-idf/tree/master/examples/system/esp_event/default_event_loop

## Debug Logs

None.

## Other items if possible

None.

i-franzki, 2019-01-03:
I also wonder why function `post_instance_create()` defines `event_data_copy` as `void **`, while the event data itself is just `void*`. Also in `struct esp_event_post_instance` it is defined as `void **`, although it should just be `void *` in my opinion.

Function `esp_event_loop_run()` then passed `post.data` to the handler, and `esp_event_handler_t` is defined with `void* event_data`. So `post.data` should really be `void *`, not `void **`. However, this does not really hurt.

i-franzki, 2019-01-03:
The minimal fix for this issue should be in functioon `esp_event_loop_run()`:

```
    while(xQueueReceive(loop->queue, &post, ticks_to_run) == pdTRUE) {
        esp_event_base_instance_t* base = NULL;
        esp_event_id_instance_t* event = NULL;
        ...
        bool exec = false;

        for (int i = LOOP_LEVEL_HANDLER; i <= EVENT_LEVEL_HANDLER; i++) {
            if (handlers_list[i] != NULL) {
                esp_event_handler_instance_t* it;
                ...
            }
            exec |= true;
        }

        free(post.data);    /* <---- add this line */

        if (ticks_to_run != portMAX_DELAY) {
           ...
```
Besides that `post.data` should be changed to just `void *` as well as `event_data_copy` in `post_instance_create()`.

detly, 2019-01-17:
I am running into this issue as well. It's a huge problem, since I've architected my entire code around ESP's event system. @i-franzki I'll try your fix and see if that helps.

detly, 2019-01-17:
Okay, @i-franzki's fix seems to have solved this issue for me (just the `free()` call, I didn't change the pointer types).

yrashk, 2019-01-17:
I am personally now leaning towards refactoring out of esp_event. While this is just one issue and this can be fixed easily as shown above, esp_event does not currently support posting for ISRs and this would require mixing esp_event with FreeRTOS queues for different messages (discriminating over their origin). 

If you think about it, this comes from the fact that esp_event effectively hides FreeRTOS queues and in order for it to support ISR, it needs to wrap more API territory, complicating it even further.

While with bare FreeRTOS you can effectively multiplex over multiple queues with queue sets, this won't mix nicely with esp_event, even though it is using queues underneath.

So, I am personally leaning towards a simple component that would allow to register callbacks over FreeRTOS queues / queue sets, ensuring it is thread-safe. This will also allow me to escape the confines of BASE/EVENT classification dichotomy.

detly, 2019-01-17:
> esp_event does not currently support posting for ISRs and this would require mixing esp_event with FreeRTOS queues for different messages

I noticed that too, but so far it hasn't been an issue. I imagine I'd just post small amounts of data from an ISR via a FreeRTOS queue into a simple task in the main code, but I don't know how well this works in practice.

ifranzki, 2019-01-17:
> esp_event does not currently support posting for ISRs and this would require mixing esp_event with FreeRTOS queues for different messages

That's correct, but this can be easily done using xTimerPendFunctionCallFromISR to run a deferred function in the RTOS timer daemon task that just posts the event (with data if needed):

Pseudo code: 

```
static void my_send_event_func(void *not_used, uint32_t event)
{
  esp_event_post(EVENT_BASE, event, NULL, 0, portMAX_DELAY));
}

static void IRAM_ATTR my_isr_handler(void* arg)
{
  ...
  xTimerPendFunctionCallFromISR(my_send_event_func, NULL, event_to_send, &xHigherPriorityTaskWoken);
}
```
I am using this in my project and it works well.

Passing event data won't work that way, unless the event data is static, or you make a copy of the data in a malloc'ed area, and free it in the deferred function. 

I am also using FreeRTOS queues directly for more complicated cases, where events are not sufficient.

i-franzki, 2019-01-17:
> esp_event does not currently support posting for ISRs and this would require mixing esp_event with FreeRTOS queues for different messages

That's correct, but this can be easily done using xTimerPendFunctionCallFromISR to run a deferred function in the RTOS timer daemon task that just posts the event (with data if needed):

Pseudo code: 

```
static void my_send_event_func(void *not_used, uint32_t event)
{
  esp_event_post(EVENT_BASE, event, NULL, 0, portMAX_DELAY));
}

static void IRAM_ATTR my_isr_handler(void* arg)
{
  ...
  xTimerPendFunctionCallFromISR(my_send_event_func, NULL, event_to_send, &xHigherPriorityTaskWoken);
}
```
I am using this in my project and it works well.

Passing event data won't work that way, unless the event data is static, or you make a copy of the data in a malloc'ed area, and free it in the deferred function. 

I am also using FreeRTOS queues directly for more complicated cases, where events are not sufficient.

closed 2019-01-31 by igrr