espressif/esp-idf #2888: windows10 cmake build error

dtboy1995, 2018-12-27:
```log
D:\esp\hello_world>idf.py build
Running ninja in directory D:\esp\hello_world\build
Executing "ninja all"...
[1/4] Generating hello-world.bin
FAILED: hello-world.bin
cmd.exe /C "cd /D D:\esp\hello_world\build && C:\Users\dtboy\.platformio\python27\python.exe D:/msys32/home/dtboy/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 2MB -o hello-world.bin hello-world.elf"
Traceback (most recent call last):
  File "D:/msys32/home/dtboy/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 35, in <module>
    import serial.tools.list_ports as list_ports
ImportError: No module named serial.tools.list_ports
[2/4] Performing build step for 'bootloader'
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-build bootloader/bootloader.elf bootloader/bootloader.bin bootloader/bootloader.map
cmd.exe /C "cd /D D:\esp\hello_world\build\bootloader && "D:\Program Files\CMake\bin\cmake.exe" --build ."
[1/1] Generating bootloader.bin
FAILED: bootloader.bin
cmd.exe /C "cd /D D:\esp\hello_world\build\bootloader && C:\Users\dtboy\.platformio\python27\python.exe D:/msys32/home/dtboy/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 2MB -o bootloader.bin bootloader.elf"
Traceback (most recent call last):
  File "D:/msys32/home/dtboy/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 35, in <module>
    import serial.tools.list_ports as list_ports
ImportError: No module named serial.tools.list_ports
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

dobairoland, 2018-12-27:
Hello @dtboy1995!

Please try to run `C:\Users\dtboy\.platformio\python27\python.exe -m pip install pyserial`. If that doesn't help you then please post here the output of `C:\Users\dtboy\.platformio\python27\python.exe -m pip show pyserial`

dtboy1995, 2018-12-27:
@dobairoland be deeply grateful. it works well

dtboy1995, 2018-12-27:
@dobairoland Thank you, I would like to ask you a question, my python environment variable is set to D:\Python27, why the path will be changed to C:\Users\dtboy\. Platformio \Python27 \ after using idf.py?

dtboy1995, 2018-12-28:
@dobairoland 
Sorry for not replying in time
This is a screenshot
[image]

dobairoland, 2019-01-02:
Hello @dtboy1995.

Yes, you are right. It seems that `esptool` was invoked with a different Python interpreter than `idf.py` was invoked with. However, I wasn't able to find the root cause, therefore, I reopened the issue to help track this as TODO.

dtboy1995, 2019-01-03:
thanks a lot @dobairoland

projectgus, 2019-03-28:
Hi @dtboy1995 & @dcm684w ,

Thanks for being patient while someone got back to you about this.

The root cause for this issue was fixed in commit d0b2d5ec95113fc7c9ec17787e16d30afb8aebf5. In some cases, the default interpreter registered in Windows for ".py" files was being used to execute a Python script instead of the correct interpreter.

In @dtboy1995's case, it looks like the python.exe in the `.platformio` directory was registered as the interpreter for .py files, but this was not the interpreter that was used to run idf.py.

If you update to the latest master branch, the same Python interpreter should be used throughout. Please leave another comment if that's not the case, and we'll reopen.

projectgus, 2019-03-28:
@dtboy1995 Oops, I missed you were using the 3.1.1 release. This hasn't been backported to v3.1 yet, but it will be. Will update this issue when that fix is merged.

closed 2019-03-28 by projectgus

dtboy1995, 2019-03-28:
@projectgus, thank you very much for your hard work.

projectgus, 2019-03-29:
@dtboy1995 Actually there's something a bit odd here, the error you got matches a fix we made in master branch but that bug was never in the 3.1 release branch (the bug was introduced after we had created the 3.1 release branch).

There are two possibilities:

1. This error has a different root cause
2 You were using master branch when you reported the issue, not the 3.1.1 release.

Could you please double check for me?

dtboy1995, 2019-03-29:
@projectgus I am very sorry, after that I used the Ubuntu system

projectgus, 2019-03-29:
@dtboy1995 No problems. Glad you found a workaround.

@dcm684w If you're still experiencing this bug, please let us know your IDF version and we can reopen the issue.