espressif/esp-idf #2889: ESP32-LyraTD-MSC A v2.2 Failed to connect to ESP32: Timed out waiting for packet header

vijaypacketp, 2018-12-27:
Hi All,
We got an ESP32-LyraTD-MSC A v2.2 dev board and followed ESP32-LyraTD-MSC V2.2 Getting Started Guide to set up, after setting up as per document
It reported error when I ran the following steps.
1), Connected it to laptop through USB cable.
2), Turned the board on.
3), Ran "make flash".
4), The following messages were printed out on terminal:"make flash

Flashing binaries to serial port COM9 (app at offset 0x10000 )...
esptool.py v2.6-beta1
Serial port COM9
Connecting......................................_____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
make: *** [/home/vkm/esp/esp-idf/components/esptool_py/Makefile.projbuild:63: flash] Error 2

Note: I connected the console throgh COM9 using teraterm/minicom and am able to see the existing application logs, there I press Reset button and release the Boot button
and made sure that

rst:0x1 (POWERON_RESET),boot:0x7 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

I also even tried with flash_download_tools_v3.6.5 and seen following message

..................................................[2018-12-27 15:17:22
,753][ESP8266Loader_spi[1]][espDownloader.py][line:351][ERROR]: Chip sync error: Failed to connect to ESP32: Timed out w
aiting for packet header
[2018-12-27 15:17:22,755][ESP8266Loader_spi[1]][espDownloader.py][line:439][ERROR]: ESP32 Chip sync error esp_sync_block
ing.

I tried in both Linux and Windows 10 machines and observed same  error message.

Could you please tell us what is missing in terms of configuration to detect the flash
Thank You
Kumar

krzychb, 2018-12-27:
Hi Kumar @vijaypacketp,

You tried most of things already including showing that the ESP32 is able to ender download mode. There are three more I do not see mentioned:

> It reported error when I ran the following steps.
1), Connected it to laptop through USB cable.
2), Turned the board on.
3), Ran "make flash".
4), The following messages were printed out on terminal:"make flash

*  Above description is missing a step described under Get Started / [Upload](https://docs.espressif.com/projects/esp-adf/en/latest/get-started/index.html#upload)
*  If you did it then I would try:
   * lowering baud rate
   * using a shorter USB cable

vijaypacketp, 2018-12-27:
Yes @krzychb  I did  hold down Boot button, momentarily press Reset button and release the Boot button, saw the wait for Download message and closed the console, will try with using a shorter USB cable 
Thank You @krzychb

vijaypacketp, 2018-12-28:
@krzychb  No luck even after using lower baud rates and  shorter  US cable ..

krzychb, 2018-12-29:
@vijaypacketp basing on all the tests you have made, it looks like this board is defective. 
Do you have any other ESP32 board to check if you are able to upload it?

closed 2019-03-15 by Alvin1Zhang