espressif/esp-idf #2892: PSRAM Cache Issue stills exist

neoniousTR, 2018-12-27:
We stumbled upon the fact that cache issue with PSRAM still exist, even in the newest development environment. This can produce random crashes, even if the code is 100 % valid.

This very small example program reproduces the problem easily, at least if compiled with newest ESP-IDF and toolchain under Mac OS X (did not try other environments):
https://github.com/neonious/memcrash-esp32/

(As a side note: We noticed this problem when we implemented the dlmalloc memory allocator in a fork of ESP-IDF. We worked around this problem (hopefully you can fix it correctly), and now have an ESP-IDF with far faster allocations. Take a look at the blog post here: https://www.neonious-basics.com/index.php/2018/12/27/faster-optimized-esp-idf-fork-psram-issues/ ).

Alvin1Zhang, 2018-12-29:
@neoniousTR Hi, neoniousTR, thanks for reporting this, we will look into this and update if any feedbacks. Also there is a topic about the issue on our forum at http://bbs.esp32.com/viewtopic.php?f=13&t=8628&sid=1acc8bd897e72cf450ad9eb71491d732. Thanks.

neoniousTR, 2019-01-26:
We updated the example project at https://github.com/neonious/memcrash-esp32
It is now leaner, more to the point, and most importantly, compiles out of the box.

We think this problem is urgent to fix, as random crashes can occur to anyone using the PSRAM of the ESP32.

There only seems to be two workarounds:

- Use only the first 2 MB of 4 MB of PSRAM (big penalty)
- End every function which stores to PSRAM with a memw instruction (slow). nops do not help.

Please take a look at the project, and hopefully you have a better idea.

Spritetm, 2019-01-29:
Fyi, we're working on this. For what it's worth, it seems to be caused by an interrupt (in your example, the FreeRTOS tick interrupt) firing while some cache activity is going on. We have our digital team running simulations to see what exactly is going on in the hardware; we hope to create a better workaround than the memw solution from that.

markwj, 2019-01-30:
We seem to be seeing this as a std::string memory corruption (all zeros, on a 4 byte boundary).

In our case, disabling the top 2MB of SPIRAM didn't seem to work. But pre-allocating 2MB (which we then never use) seemed to workaround the problem. Our code runs primarily on core #1.

This is impacting us quite badly. Lots of random corruptions and crashes with devices in the field.

neoniousTR, 2019-01-30:
Maybe whether the top or bottom of the RAM works depends on the core used.

dexterbg, 2019-01-30:
Confirmed: running our test project from #3006 with that 2 MB allocation and also starting the test task on core 0 shows the corruptions again. It seems core 0 can only work reliably with the lower 2 MB and core 1 only with the higher 2 MB.

neoniousTR, 2019-02-01:
@Spritetm or @Alvin1Zhang
As this issue does not happen in single core mode, do you know if the original PSRAM cache issue which is fixed with the flag and adds many nops and memws is also only in dual core mode?

If so, we will try to switch low.js to single core mode, this might even be faster at the end, because the JavaScript itself is single core anyhow and has the most load.

Also, how is the progress going? I'd think the chances are to get this fixed by modifying the interrupt handlers or the cache fetchers and savers (they are part of the ROM?).

xbary, 2019-02-11:
Hello, I wanted to add to the subject the error I observed in my application using PSRAM. Random error while retrieving the amount of free PSRAM memory. In my application, I check the amount of free PSRAM memory in the main loop, and differently I received in reply that, for example, 16 bytes of free memory, but at the next check, it actually answered ~ 4mb.

In my opinion, there must have been an erroneous random reading from PSRAM.

neoniousTR, 2019-02-19:
Our current status:

Load to cache/Write from cache does not seem to be interrupt-based.. Might be 100% hardware-based?

Added memw to interrupt handlers does not change anything.

Currently we believe Dual Core + PSRAM is a broken combination.

So we will completly switch to Unicore now.

Please answer:
Do you know if the original PSRAM cache issue also exist in unicore mode? Would be great if we can get rid of the nops and memws with this once and for all...

xbary, 2019-02-21:
I made such an experience, I rewrote the String class from the arduino project to use the PSRAM memory. I changed the name and changed the realloc in the changebuffer function. Suddenly, it turned out that my application did not regularly show 0x00 in one cell.
here is this changed String class: https://github.com/xbary/xb_StringPSRAM
I would like you to be able to replace the class with StringSRAM as part of the tests, you may be able to reproduce the repeatability of the error.

neoniousTR, 2019-02-22:
I have to confirm that the original PSRAM workaround is still required in Unicore mode.

So Workaround + Unicore is the only combination which works reliably with PSRAM. If I am wrong, I hope somebody will post. Otherwise we have to take this as a fact ...

me21, 2019-02-23:
Can dual core chip be switched to unicore mode?
Does this error manifest itself in Arduino framework? As far as I know, Arduino task is pinned to core 0, therefore, it can be effectively viewed as unicore. Am I right?

neoniousTR, 2019-02-23:
Yes the CPU can be configured for unicore.

Pinning to one core does not help, as in dual core mode the other 2 MB are handled by the other core.

Spritetm, 2019-02-26:
FWIW, we have a tentative solution for this; the existing workaround solution does actually seem to work but doesn't take calls/returns into account properly. We'll ship a toolchain with improved workaround code soon, but we want to have this fairly well tested so we don't have any other edge cases sneaking past us. I'll see if I can post a preliminary patch as soon as I have something halfway stable,

markwj, 2019-03-04:
Do have any idea of schedule for this, or an ability to get us a pre-release toolchain?

This is impacting us quite badly. The 2MB pre-allocation solves the problem for our code, but just shifts the problem to wifi running on the first core (which now experiences random errors and throughput problems).

xbary, 2019-03-04:
I confirm, the error still occurs at random moments, even hangs completely.

markwj, 2019-04-25:
Do have any idea of schedule for this, or an ability to get us a pre-release toolchain?

dexterbg, 2019-05-25:
@Spritetm We do appreciate your efforts in making sure your patch is perfect. But meanwhile our system has to bear a huge performance hit by the workaround, while the stability is still impacted by the bug. We're more than willing to help you in beta testing your patch by using it on our project. Please do a pre-release or share some update on the status. Thanks!

negativekelvin, 2019-05-25:
FWIW, for the example this seems to work

```
        // write
        crash_set_both(mem1, mem2, val);

        portENTER_CRITICAL(&memtest_spinlock);
        asm volatile (
        "nop \n"
        "nop \n"
        "nop \n"
        "nop \n");
        // read
        int val2 = *mem2;
        portEXIT_CRITICAL(&memtest_spinlock);
        
        if(val2 != val)
            printf("Try %d did not work, addresses in SPI RAM (base=0): %08x %08x!\n", tries, pos1, pos2);
        tries++;
```

Patrik-Berglund, 2019-05-26:
Also think an update is in place, we are awaiting to see if you are able to fix this bug or if it makes the PSRAM feature unusable. 

We need more RAM than internal available in the ESP32, so this is a deal breaker for our product.

negativekelvin, 2019-05-26:
Just wondering why if the original workaround should work in this case that forcing nops does not resolve it.

```
400d4b3c:	1047a5        	call8	400e4fb8 <crash_set_both>
400d4b3f:	f03d      	nop.n
400d4b41:	f03d      	nop.n
400d4b43:	f03d      	nop.n
400d4b45:	f03d      	nop.n
400d4b47:	0228      	l32i.n	a2, a2, 0
```

Also I noticed the workaround will add the nops even when there is already a memw barrier.

```
400d4e86:	03a9      	s32i.n	a10, a3, 0
400d4e88:	0020c0        	memw
400d4e8b:	01a9      	s32i.n	a10, a1, 0
400d4e8d:	f03d      	nop.n
400d4e8f:	f03d      	nop.n
400d4e91:	f03d      	nop.n
400d4e93:	f03d      	nop.n
400d4e95:	0020c0        	memw
400d4e98:	0138      	l32i.n	a3, a1, 0
```

Spritetm, 2019-05-29:
From what I can see, the load-store inversion doesn't occur there exactly... the issue has something to do with a cache miss around that time that has delayed effects later. Because a cache miss takes a while to resolve, you can fix it with nops but you'd need to put a gazillion of them there.

negativekelvin, 2019-05-29:
Ok thanks.  Other things I noticed when playing with the memcrash-esp32 example:

- If running on core 0 error will occur with mem2 (lower) in HIGH-LOW mode
- If running on core 1 error will occur with mem1 (upper) in HIGH-LOW mode
- If running on core 0 error will occur with both mem1 & mem2 even in EVEN-ODD mode
- If running on core 1 error will occur with both mem1 & mem2 odd in EVEN-ODD mode
- Error does not seem to happen in NORMAL mode on either core ( I don't know if this is supported or just giving a false result)

negativekelvin, 2019-05-30:
More info: the memw in the example actually does not prevent the error when the routine is running in parallel on both cores.  It is much more infrequent but still happens.  

Normal mode does have a performance cost as it is around 24% fewer tries/ms with the example on both cores, but no errors.

neoniousTR, 2019-06-01:
Normal mode has cache coherency issues according to documentation, so not an option.

negativekelvin, 2019-06-01:
Yes I see, makes sense.  Back to the drawing board then.

https://github.com/espressif/esp-idf/blob/a62cbfec9ab681bddf8f8e45d9949b1b1f67a2ec/components/esp32/spiram_psram.h#L39-L41

Spritetm, 2019-06-05:
Okay, I have a toolchain update that has a different psram workaround. You can grab the binaries for [Linux](https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-90-g26d9621-5.2.0-psramfix-20190605.tar.gz), [Windows](https://dl.espressif.com/dl/xtensa-esp32-elf-windows-1.22.0-90-g26d9621-5.2.0-psramfix-20190605.tar.gz) or  [OSX](https://dl.espressif.com/dl/xtensa-esp32-elf-macos-1.22.0-90-g26d9621-5.2.0-psramfix-20190605.tar.gz). By default, it should use the workaround; switch to the new one by putting

    CFLAGS+=-mfix-esp32-psram-cache-dupldst
    CXXFLAGS+=-mfix-esp32-psram-cache-dupldst

into your Makefile (or do the equivalent for cmake), recompile and test.

Note that this is not an 100% fix as the binary libraries (wifi/bt/libstdc/libc++/libgcc) at the moment are not compiled with this workaround, but I don't think I have seen any issues that originated there yet, so as a first test this should cover most use cases.

neoniousTR, 2019-06-05:
Thank you.
Can you explain the workaround? What does it do / what penalty does it have?

Spritetm, 2019-06-05:
It's actualy almost shamefully trivial for how well it works... every store of a 32-bit value to a memory location that is not volatile, will have a load from that memory location appended to it. I haven't ran any benchmarks, but I don't think it affects speed more than the NOP solution. In theory, it may break storing a word to a peripheral (as the load after may trigger something), but as those should be declared volatile, I don't think that happens in practice.

neoniousTR, 2019-06-05:
Thank you.

Is there a schedule when the binaries will exist fixed? Is it realistic that all code in ROM will be patched, too (btw was this already done completly with the last psram workaround? This would be good because we should not expect more code to be patched, just a new patch, correct?).

Spritetm, 2019-06-05:
We first need to know if this fix 1. actually fixes the issue and 2. doesn't introduce more problems than it fixes. Once that's established, we'll also recompile all binary libraries with it. The amount of fixed code should indeed be the same as the existing psram fix.

negativekelvin, 2019-06-05:
The windows binaries have symlinks instead of copies.  

It doesn't seem to work:
```mem_task start on core 1
mem_task start on core 0
At tick: 1000  tries per ms: 152.26
At tick: 1000  tries per ms: 152.26
At tick: 2000  tries per ms: 153.84
At tick: 2000  tries per ms: 153.84
At tick: 3000  tries per ms: 154.35
At tick: 3000  tries per ms: 154.35
At tick: 4000  tries per ms: 154.61
At tick: 4000  tries per ms: 154.61
At tick: 5000  tries per ms: 154.77
At tick: 5000  tries per ms: 154.77
At tick: 6000  tries per ms: 154.88
At tick: 6000  tries per ms: 154.88
At tick: 7000  tries per ms: 154.96
At tick: 7000  tries per ms: 154.96
mem1: Try 1204965 did not work on core 0, addresses in SPI RAM (base=0): 00231930 (1)
00019450! (1)
At tick: 8000  tries per ms: 155.01
At tick: 8000  tries per ms: 155.01
mem2: Try 1363835 did not work on core 0, addresses in SPI RAM (base=0): 0035ce70 (1)
001df184! (1)
mem1: Try 1368300 did not work on core 0, addresses in SPI RAM (base=0): 0035eddc (0)
001b4118! (0)
At tick: 9000  tries per ms: 155.04
At tick: 9000  tries per ms: 155.04
At tick: 10000  tries per ms: 155.08
At tick: 10000  tries per ms: 155.08
At tick: 11000  tries per ms: 155.12
At tick: 11000  tries per ms: 155.12
At tick: 12000  tries per ms: 155.14
At tick: 12000  tries per ms: 155.14
At tick: 13000  tries per ms: 155.16
At tick: 13000  tries per ms: 155.16
At tick: 14000  tries per ms: 155.18
At tick: 14000  tries per ms: 155.18
At tick: 15000  tries per ms: 155.20
At tick: 15000  tries per ms: 155.20
At tick: 16000  tries per ms: 155.21
At tick: 16000  tries per ms: 155.21
At tick: 17000  tries per ms: 155.23
At tick: 17000  tries per ms: 155.23
mem1: Try 2740709 did not work on core 0, addresses in SPI RAM (base=0): 0020f154 (0)
0005ea9c! (0)
At tick: 18000  tries per ms: 155.23
At tick: 18000  tries per ms: 155.23
At tick: 19000  tries per ms: 155.24
At tick: 19000  tries per ms: 155.24
mem2: Try 3008575 did not work on core 0, addresses in SPI RAM (base=0): 002dd3a0 (1)
00116174! (1)
At tick: 20000  tries per ms: 155.25
At tick: 20000  tries per ms: 155.25
At tick: 21000  tries per ms: 155.25
At tick: 21000  tries per ms: 155.25
At tick: 22000  tries per ms: 155.26
At tick: 22000  tries per ms: 155.26
mem1: Try 3472272 did not work on core 0, addresses in SPI RAM (base=0): 003f26b4 (1)
001133b0! (1)
mem1: Try 3520313 did not work on core 0, addresses in SPI RAM (base=0): 003ceb50 (0)
001b35ec! (0)
```

```
400e6bcc <crash_set_both>:
400e6bcc:	004136        	entry	a1, 32
400e6bcf:	0249      	s32i.n	a4, a2, 0
400e6bd1:	0248      	l32i.n	a4, a2, 0
400e6bd3:	0349      	s32i.n	a4, a3, 0
400e6bd5:	0348      	l32i.n	a4, a3, 0
400e6bd7:	f01d      	retw.n
400e6bd9:	000000        	ill
```

dexterbg, 2019-06-05:
The good news: using the workaround my test project from #3006 no longer produces the corruptions in the char buffer tests (`run_test_buffer`).

The bad news: the std::string tests have corruptions as before. Did you build the included libstdc++ using the workaround? As we mostly use std::string for buffers, we won't see much of an improvement without a fixed std::string class.

dexterbg, 2019-06-05:
**Update**: looking into the toolchain I found a secondary build of the libs in `xtensa-esp32-elf/xtensa-esp32-elf/sysroot/lib/esp32-psram`, so I gave those linker precedence and voila: the std::string tests now also work without corruptions!

So it seems the -m flag misses setting that path. I've added this to my `main/component.mk`:

`COMPONENT_ADD_LDFLAGS = -lmain -L$(HOME)/esp/xtensa-esp32-elf/xtensa-esp32-elf/sysroot/lib/esp32-psram`

There surely is a better method to add this to the project Makefile I haven't found yet.

I'll now try building our main project with the workaround.

dexterbg, 2019-06-05:
Not a single data corruption yet, I'm publishing the build to our beta testers now.

negativekelvin, 2019-06-06:
Yeah it only happens on the order of 1 per million tries when hammering the cache from both cores so it may be very rare in a practical application

@neoniousTR makes an interesting point about passing an extram pointer to a rom function which may perform a store and then return to user code which performs a load

Spritetm, 2019-06-06:
@negativekelvin That doesn't mesh with my tests. What setup do you use? I tried the sources at https://github.com/neonious/memcrash-esp32/ with only the Makefile modified to use the master esp-idf and to add the -mfix-esp32-psram-cache-dupldst flag. I've ran the test for a good million ticks now and haven't seen any errors.

Wrt ROM functions: All the relevant ROM functions (mostly Newlib) get placed in flash with the workaround compiled in when the PSRAM workaround is enabled. That's the case for the old workaround and will also be done for the new workaround as soon as I have an indication it works well enough and doesn't lead to other issues. Same goes for all other libraries.

negativekelvin, 2019-06-06:
@Spritetm my version here https://github.com/negativekelvin/memcrash-esp32

Spritetm, 2019-06-06:
Thanks, that one also fails for me. I'll look into what the issue is there...

negativekelvin, 2019-06-06:
Ah ok I think you got me there, ...brain fart.  I guess it is working then!

negativekelvin, 2019-06-06:
Just want to mention I am still seeing an error in EVENODD mode on the order of 1 in 100 million tries.  I don't have any particular need to use this mode though.

neoniousTR, 2019-06-06:
That EVENODD does not work 100% lets me think: Could the fix maybe not work if the address which you read from which is appended to the one you write in is in the other half of the memory? Right on the boundary?

negativekelvin, 2019-06-06:
Not sure, there are definitely more boundaries in EVENODD but the extra generated load is from the same address as the store.

As of now I also have observed 2 errors on one of two cores out of 1.6 billion tries using LOWHIGH.

dexterbg, 2019-06-08:
Test results so far: improved stability & network performance, no new issues. Looks very good.

Spritetm, 2019-06-10:
@negativekelvin I can replicate the few-times-in-a-billion error with your code. It's a weird error: while the previous errors could be influenced by interrupts (as in: more interrupts = more errors), this error is impervious to interrupt flooding, which makes me think it's caused by something else.

Spritetm, 2019-06-12:
New version of the toolchain is here: [Linux](https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-93-gf6c4cdf-5.2.0-psramfix-20190612.tar.gz) [MacOS](https://dl.espressif.com/dl/xtensa-esp32-elf-macos-1.22.0-93-gf6c4cdf-5.2.0-psramfix-20190612.tar.gz) [WIndows](https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-93-gf6c4cdf-5.2.0-psramfix-20190612.tar.gz) (note the Windows toolchain probably still has symlinks instead of copies)

In this version, the load-after-store is the default psram fix; there's no need to use `-mfix-esp32-psram-cache-dupldst` in your Makefile anymore. The old fix still is available: specify `-mfix-esp32-psram-cache-nops` to use it. Because of this change, all the compiler libraries (libgcc, libstdc++ etc) should also be compiled with the new fix.

neoniousTR, 2019-06-12:
A few times in a billion IMHO is too much. Customers expect uptimes of years (our PC servers manage that easily), and if such an error propagates to the file system, data might even be corrupted.
Have you tried turning down the Mhz? Maybe it is simply a bad signal problem.

neoniousTR, 2019-06-12:
Question 2: I think there are binary blobs in ESP-IDF, too, such as Wifi? If I am correct, how are these updated?

Spritetm, 2019-06-12:
@neoniousTR Agreed entirely, and we're not done with this issue yet. However, seeing that this already is a pretty large improvement on the previous issue and I can see the remaining issue potentially take a fair while to catch, I'd rather start giving out beta tests and updating early. WiFi and Newlib need to be updated in an esp-idf release; as soon as the gcc release is stable I'll see if I can get those recompiled as well.

neoniousTR, 2019-06-19:
@Spritetm: I just sent you a message via the contact form of your website SpritesMod.com, as I did not find any other way to contact you in private. Please look for this mail. Thank you!

Spritetm, 2019-06-21:
Another build; this one should be pretty close to final. The once-a-billion problem is fixed, thanks to some simulation and creative thinking on the side of my digital colleagues. Seems doing store-load-store solves all issues. Also included in this new build is a fix for an edge case where the original patch erroneously thought a memw wasn't necessary.

[Linux](https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-95-g5a1864d-5.2.0-psramfix-20190621.tar.gz)
 [MacOS](https://dl.espressif.com/dl/xtensa-esp32-elf-macos-1.22.0-95-g5a1864d-5.2.0-psramfix-20190621.tar.gz)
 [WIndows](https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-95-g5a1864d-5.2.0-psramfix-20190621.tar.gz) (note the Windows toolchain probably still has symlinks instead of copies)

dexterbg, 2019-06-21:
Either we need to do something different than before or something's wrong with that -95 version compared to the previous -93 version: simply rebuilding our project with the -95 leads to a crash reboot loop.

The crash happens right after the scheduler startup:

```
I (395) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
/home/balzer/esp/esp-idf/components/freertos/queue.c:1445 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40094b17 on core 0
0x40094b17: xQueueGenericReceive at /home/balzer/esp/esp-idf/components/freertos/queue.c:2037

Backtrace: 0x400906fc:0x3ffcbbf0 0x400908d3:0x3ffcbc10 0x40094b17:0x3ffcbc30 0x40094c55:0x3ffcbc70 0x400855b1:0x3ffcbc90 0x400856a9:0x3ffcbcc0 0x40123e9d:0x3ffcbce0 0x40118e1d:0x3ffcbfa0 0x40118919:0x3ffcbff0 0x40092e89:0x3ffcc020 0x40093b32:0x3ffcc050 0x40094db1:0x3ffcc070 0x400959a7:0x3ffcc090 0x40095aac:0x3ffcc0c0
0x400906fc: invoke_abort at /home/balzer/esp/esp-idf/components/esp32/panic.c:670

0x400908d3: abort at /home/balzer/esp/esp-idf/components/esp32/panic.c:670

0x40094b17: xQueueGenericReceive at /home/balzer/esp/esp-idf/components/freertos/queue.c:2037

0x40094c55: xQueueTakeMutexRecursive at /home/balzer/esp/esp-idf/components/freertos/queue.c:2037

0x400855b1: lock_acquire_generic at /home/balzer/esp/esp-idf/components/newlib/locks.c:155

0x400856a9: _lock_acquire_recursive at /home/balzer/esp/esp-idf/components/newlib/locks.c:169

0x40123e9d: _vfiprintf_r at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:860 (discriminator 2)

0x40118e1d: fiprintf at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c:50

0x40118919: __assert_func at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:59 (discriminator 8)

0x40092e89: vPortCPUAcquireMutexIntsDisabledInternal at /home/balzer/esp/esp-idf/components/freertos/tasks.c:3564
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/balzer/esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at /home/balzer/esp/esp-idf/components/freertos/tasks.c:4267

0x40093b32: vTaskPlaceOnEventListRestricted at /home/balzer/esp/esp-idf/components/freertos/tasks.c:3564

0x40094db1: vQueueWaitForMessageRestricted at /home/balzer/esp/esp-idf/components/freertos/queue.c:2429

0x400959a7: prvProcessTimerOrBlockTask at /home/balzer/esp/esp-idf/components/freertos/timers.c:484

0x40095aac: prvTimerTask at /home/balzer/esp/esp-idf/components/freertos/timers.c:484

Rebooting...
```

Chip is ESP32D0WDQ6 (revision 1)

The build from version -93 works flawlessly.

dexterbg, 2019-06-21:
Btw, for the record: the previous builds had few crashes left, most of them happening at random places within the LWIP subsystem while running on Wifi. Since yesterday I'm testing `CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST` disabled, too few data (read: no crashes) on that config yet.

I assume the next big step in stability can now be achieved by replacing the idf blobs.

PerMalmberg, 2019-06-24:
Can these patches be made available for gcc 8 as well? I'd like to test if they solve #3624.

PerMalmberg, 2019-07-16:
Can we please have an update on this? I still can't use TLS on the ESP as per my previous post (tried with today's IDF, v4.0-dev-1275-gfdab15dc7, xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0

dexterbg, 2019-07-22:
> rebuilding our project with the -95 leads to a crash reboot loop.

Solved: after upgrading our esp-idf clone to v3.3-beta3, builds with toolchain -95 now work without any problems so far.

dexterbg, 2019-07-22:
Sorry, I was too quick assuming no problems. My build just began crashing and won't stop. The -93 build before has been running fine for two days, so there still is an issue with -95.

leres, 2019-08-02:
For those of us who can't use a binary toolchain could we get the source version that results in crosstool-ng-1.22.0-93-gf6c4cdf? (And if we could get the -95 version we could investigate the boot crash that Michael reports.)

I'm currently using 1.22.0-80-g6c4433a5

dexterbg, 2019-08-03:
A fix for version -95 would be nice, as the once-in-a-billion problem apparently occurs in real life (I've had it this morning on my production module).

jcsbanks, 2019-08-14:
On Windows using esp-idf 3.3rc, how do I update the toolchain from xtensa-esp32-elf-win32-1.22.0-95-g5a1864d-5.2.0-psramfix-20190621.tar.gz download to test and how to go back if it doesn't work?

[121 more replies]

closed 2020-11-13 by igrr