espressif/esp-idf #2895: Git clone with shallow failed

larryli, 2018-12-28:
## Environment

- IDF version: v3.1.1 and other
- Operating System:     Windows|Linux|macOS

## Problem Description

Git clone with `--shallow-submodules` failed.

### Steps to repropduce

```shell
git clone -b v3.1.1 --recurse-submodules --shallow-submodules https://github.com/espressif/esp-idf.git
```

Failed:

```
error: Server does not allow request for unadvertised object 8bf852db77c360eebfa4b800754fdb90e29ea43e
Fetched in submodule path 'components/aws_iot/aws-iot-device-sdk-embedded-C', but it did not contain 8bf852db77c360eebfa4b800754fdb90e29ea43e. Direct fetching of that commit failed.
```

mahavirj, 2018-12-29:
@larryli 
- Release artifacts can also be downloaded from https://github.com/espressif/esp-idf/releases/tag/v3.1.1
- If not, then you may want try increasing depth, e.g. `git submodule update --depth 10`

Alvin1Zhang, 2019-01-02:
@larryli Hi, larryli, would you help share if any updates for this issue? Thanks.

larryli, 2019-01-02:
@Alvin1Zhang 

https://stackoverflow.com/questions/2144406/how-to-make-shallow-git-submodules/2169914

But I don't test the scripts in the answer.

kostyan5, 2019-10-18:
Was there any resolution to this issue?  I can't switch to 3.3 tag with same error.