espressif/esp-idf #2896: NVS base64 support

LaurentLouf, 2018-12-28:
## Environment

- Development Kit: Custom board
- Kit version Custom board
- Module or chip used:  ESP32-WROOM-32
- IDF version v3.1.1
- Build System: CMake
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: Linux Ubuntu
- Power Supply: USB

## Problem Description

I can't access the data that I have defined in a NVS partition, as base64 type. The base64 is not listed as part of the supported types in the documentation but can be found in some examples in the repository, see for example https://github.com/espressif/esp-idf/blob/master/components/nvs_flash/nvs_partition_generator/sample_multipage_blob.csv . 

I guess this is only a matter of time until the data type is supported, but any idea when ? Would be really really useful for certificates and private key that usually come as base64 strings.

### Expected Behavior

`nvs_flash_init_partition` should return ESP_OK and data should be accessible

### Actual Behavior

`nvs_flash_init_partition`returns ESP_ERR_NVS_NEW_VERSION_FOUND and data is therefore inaccessible.

### Steps to repropduce

1. Define a NVS partition for some data, for example 
```
# Custom ESP32 Partition Table with larger app
# Name,   Type, SubType, Offset,  Size
nvs,      data, nvs,     0x9000,  0x4000
security, data, nvs,     0xd000,  0x3000
factory,  app,  factory, 0x10000, 2M
```
2. Define some data in it, with some being `base64` types 
```
key,type,encoding,value
security,namespace,,
username,data,string,54ssd
client_id,data,string,pouirge5465
password,data,string,8787gdfg
private_key,data,base64,MIIEowIBAA
certificate,data,base64,MIIDnTCCAo=
```
3. Generate binary partition with `nvs_partition_gen.py` then flash it at specified address
4. Try to initialize the partition 

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"
#include "nvs.h"
#include "nvs_flash.h"
const char* security_partition_namespace_label = "security";

void app_main()
{
esp_err_t nvs_error = nvs_flash_init_partition(security_partition_namespace_label);
    if (nvs_error == ESP_OK) {
        nvs_handle nvs;
        nvs_error = nvs_open_from_partition(security_partition_namespace_label,
                                            security_partition_namespace_label, NVS_READONLY, &nvs);
        if (nvs_error == ESP_OK) {
            size_t required_size;
            nvs_get_str(nvs, "username", NULL, &required_size);
            char* username = (char*)malloc(required_size);
            nvs_get_str(nvs, "username", username, &required_size);
            printf("Retrieved username %s\r\n", username);
            fflush(stdout);
        } else {
            printf("Error in nvs_open_from_partition %d\r\n", nvs_error);
            fflush(stdout);
        }
    } else {
        printf("Error in nvs_flash_init_partition %d\r\n", nvs_error);
        if (nvs_error == ESP_ERR_NVS_NO_FREE_PAGES) {
            printf("ESP_ERR_NVS_NO_FREE_PAGES\r\n");
        } else if (nvs_error == ESP_ERR_NOT_FOUND) {
            printf("ESP_ERR_NOT_FOUND\r\n");
        } else if (nvs_error == ESP_ERR_NVS_NEW_VERSION_FOUND) {
            printf("ESP_ERR_NVS_NEW_VERSION_FOUND\r\n");
        }
        fflush(stdout);
    }    
}

```

## Debug Logs

Flashing
```
Compressed 4096 bytes to 2128...
Wrote 4096 bytes (2128 compressed) at 0x0000d000 in 0.1 seconds (effective 604.2 kbit/s)...
Hash of data verified.
```

Booting
```
I (28) boot: ESP-IDF v3.1.1 2nd stage bootloader
I (28) boot: compile time 12:06:52
I (28) boot: Enabling RNG early entropy source...
I (32) boot: SPI Speed      : 40MHz
I (37) boot: SPI Mode       : DIO
I (41) boot: SPI Flash Size : 4MB
I (45) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (63) boot:  1 security         WiFi data        01 02 0000d000 00003000
I (70) boot:  2 factory          factory app      00 00 00010000 00200000
I (78) boot: End of partition table
```

Testing
```
Error in nvs_flash_init_partition 4368
ESP_ERR_NVS_NEW_VERSION_FOUND
```

mahavirj, 2018-12-28:
@LaurentLouf This error `ESP_ERR_NVS_NEW_VERSION_FOUND` looks bit different though (as in nothing to do with data type formatting). Could you please try adding `--version=v1` to you nvs partition generator command? (ref: https://github.com/espressif/esp-idf/tree/release/v3.1/components/nvs_flash/nvs_partition_generator)

LaurentLouf, 2018-12-28:
@mahavirj yup, sorry, I thought about that a bit after posting the issue, testing now and probably closing the issue right after :)

LaurentLouf, 2018-12-28:
Yup, that was it.

closed 2018-12-28 by LaurentLouf