espressif/esp-idf #2897: Cannot create UART3....UART3 not working

hetvishah08, 2018-12-28:
**Hardware:**
Board:	ESP32 DevkitC
Core Installation/update date:	12 December 2018
IDE name:	Arduino IDE
Flash Frequency:	40Mhz
PSRAM enabled: no
Upload Speed:	115200
Computer OS: Windows 10

**Description**:
I am trying to develop an application on ESP32 DevkitC where i need 4 UART ports.I want to use GPIO4 and GPIO5 as my 4th UART port. I am not able to create it in ESP-IDF.

In Arduino IDE there is a provision to use other GPIOs for UART by using SoftwareSerial library.I have tried it and was successful using those GPIOs.
I couldn't find a way in ESP-IDF. Help me by suggesting a way.
 When i add UART_NUM_3 it shows me error;

uart_param_config(UART_NUM_3, &uart_config);

**Debug**:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
"uart_num error", ESP_FAIL

I checked the uart.c file and saw this line:

static DRAM_ATTR uart_dev_t* const UART[UART_NUM_MAX] = {&UART0, &UART1, &UART2};

is it possible to add UART3 here and make changes to the library where necessary and create UART3?
Or like SoftwareSerial in Arduino IDE is there any other way to create new UART port?

sysizlayan, 2018-12-28:
ESP32 has 3 hardware UART peripherals as you have already said and this cannot be increased.
There is no SoftwareSerial class written for ESP32 as far as I know.

I you need additional UART ports, you may us multiplexer or another microcontroller which can be programmed to transfer incoming data from I2C/SPI to UART.

koobest, 2019-01-02:
Hi @hetvishah08
ESP32 only have three hardware uart, UART0, UART1 and UART2.

hetvishah08, 2019-01-02:
@sysizlayan there is a SoftwareSerial library available for ESP32 on Arduino IDE

Here is the link and it works fine.
https://github.com/akshaybaweja/SoftwareSerial

hetvishah08, 2019-01-02:
@koobest  Yes i do know that.I wanted to know if there is any way possible to add one more serial like we can in Arduino IDE using Software Serial library as i am developing my application using ESP-IDF.

koobest, 2019-01-04:
Hi, @hetvishah08
Can we know your application scenario? If only write data, you can use time division multiplexing for a serial port. Two tx_pins are connected to two different receivers and switch the signal to the corresponding pin before sending the data.

xiongyumail, 2019-01-11:
I think you can use a hardware timer and gpio to simulate a serial port.

Alvin1Zhang, 2019-03-21:
@hetvishah08 Hi, would you help share if any updates for the issue? Thanks.

hetvishah08, 2019-03-25:
@koobest  and @xiongyumail  Thanks for your help.
@Alvin1Zhang I wasn't able to develop a solution similar to Software Serial in Arduino.

sysizlayan, 2019-04-05:
@hetvishah08 I do not know your communication topology. If you are allowed to change the communication pattern in your design, you can develop a solution using an addressable communication. What I mean, you can specify the address an address at the beginning of the message package and wait for the response, similar to Modbus or I2C. By this, you can use the same UART bus to interface multiple devices.

borud, 2020-01-11:
If someone asks for software serial, please don't waste their time by demanding to know what they are trying to do or suggesting something that probably isn't going to help them.  Please have some respect and assume that they have a legitimate reason to ask. 

_I came here while looking for a usable software serial implementation for ESP32, as I imagine many will, and there is nothing more annoying than people being this unhelpful.  Suggesting they change the hardware (which isn't an option if you can't alter the design) or that they switch to some other programming environment is not being helpful.  Please think about this before responding._

chegewara, 2020-01-11:
What the heck was that? Trolling on github?

borud, 2020-01-11:
No, it was a plea to have some respect for people who ask questions.

o-marshmallow, 2021-11-17:
To sum up and as said previously, ESP32 has 3 hardware UARTs. If you would like to have more UART interfaces, you would need to bitbang some GPIOs to simulate a UART. ESP-IDF doesn't have such library embedded, you will have to find a third-party one or write one yourself.
Closing this issue. If the question is still relevant, please open a new issue.

closed 2021-11-17 by o-marshmallow