espressif/esp-idf #2901: Error during make flash with esp-idf-v3.1.1

siara-cc, 2018-12-30:
## Environment

- Development Kit: ESP32
- IDF version: v3.1.1 
- Build System: Make
- Compiler version : (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:  macOS
- Power Supply: USB

## Problem Description

When trying to do `make flash` getting following errors during compilation:

```
CC build/app_trace/app_trace.o
In file included from /Users/arun/esp/esp-idf-v3.1.1/components/soc/esp32/include/soc/soc.h:20:0,
                 from /Users/arun/esp/esp-idf-v3.1.1/components/app_trace/app_trace.c:160:
/Users/arun/esp/esp-idf-v3.1.1/components/soc/esp32/include/soc/io_mux_reg.h: In function 'PIN_PULLUP_DIS':
/Users/arun/esp/esp-idf-v3.1.1/components/esp32/include/esp_assert.h:24:13: error: implicit declaration of function 'assert' [-Werror=implicit-function-declaration]
             assert(#MSG && (CONDITION));                                                                    \
             ^
/Users/arun/esp/esp-idf-v3.1.1/components/soc/esp32/include/soc/soc.h:146:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/Users/arun/esp/esp-idf-v3.1.1/components/soc/esp32/include/soc/soc.h:177:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_CLR_BIT);                                                                    \
             ^
/Users/arun/esp/esp-idf-v3.1.1/components/soc/esp32/include/soc/io_mux_reg.h:95:5: note: in expansion of macro 'REG_CLR_BIT'
     REG_CLR_BIT(PIN_NAME, FUN_PU);
     ^
cc1: some warnings being treated as errors
make[1]: *** [app_trace.o] Error 1
make: *** [component-app_trace-build] Error 2
```
### Expected Behavior

No error during compilation (this error comes from code belonging to esp-idf)

### Actual Behavior

See above

### Steps to reproduce

1. Download esp-idf-v3.1.1.zip that includes submodules and unzip
2. `cd esp-idf-v3.1.1`
3. `git submodule add https://github.com/siara-cc/esp32-idf-sqlite3 components/esp32-idf-sqlite3`
4. `git submodule add https://github.com/siara-cc/esp32-idf-sqlite3-examples examples/esp32-idf-sqlite3-examples`
5. `cd examples/esp32-idf-sqlite3-examples/spiffs`
6. `make flash`

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2717120/sdkconfig.txt)

igrr, 2018-12-30:
This is probably because the component you are including into the build, `esp32-idf-sqlite3`, provides a file called `assert.h`, which shadows `assert.h` provided by newlib.

In that file (https://github.com/siara-cc/esp32-idf-sqlite3/blob/master/include/assert.h), if NDEBUG is not defined, there is an `include <assert.h>` which will include the same file. In this case, `assert` macro ends up being undefined.

https://github.com/siara-cc/esp32-idf-sqlite3/blob/2527c3c630daa423dcb64985289a53510f0b1102/include/assert.h#L6

You can replace that `include` with `include_next`, then it should work as you have intended, i.e. include the "real" assert.h file.

siara-cc, 2018-12-31:
The problem stems from `newlib`'s definition of `assert` which is `# define assert(__e) ((void) sizeof(__e))` when `NDEBUG` is defined.  This expects __e to be valid when NDEBUG is defined, which creates the problem with the way `sqlite3.c` uses assert macro (refer to line number 106235).

I am inclined to think that `newlib`'s `assert` should not expect `__e` to be valid when `NDEBUG` is defined.  In this case I am removing the component's assert.h and handling by modifying `sqlite3.c` and closing this issue, but I am mentioning this because it may arise with other 3rd party components in future.  Also I am not sure if the components include files should affect other components.

Thank you for looking into it.

siara-cc, 2018-12-31:
Linking same issue faced with ESP32 Arduino SDK: https://github.com/espressif/esp-idf/issues/2758

closed 2018-12-31 by siara-cc

igrr, 2018-12-31:
> Also I am not sure if the components include files should affect other components

Component may have private include directories (which will be added to the compiler command line when compiling this component only), and public include directories (exported to other components). You can create a directory, say include_private, in your component, place assert.h there, and add this directory to COMPONENT_PRIVINCLUDEDIRS. Then this modified assert.h will only be used by your components

siara-cc, 2018-12-31:
I added
`COMPONENT_PRIV_INCLUDEDIRS=private_include`
to `component.mk` and moved `assert.h` and `osapi.h` to `private_include`.
It works fine - Thanks!!