espressif/esp-idf #2902: ble disconnect reason 0x0100

chegewara, 2018-12-31:
What is disconnect reason 0x0100?
`BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=515 reason=0x0100`

negativekelvin, 2018-12-31:
https://github.com/espressif/esp-idf/blob/135eef767f70b6fc1db00c3f44b07b18d4bff976/components/bt/bluedroid/stack/include/stack/gatt_api.h#L125

closed 2019-01-01 by chegewara

tanchakra, 2023-06-02:
Hi team,
I am working on ESP32 EVK, facing an issue in connecting TPMS sensor to the ESP32 module.
it is scanning the sensor device but not able to connect to it.
I have shared the error logs below
 **I am not understanding the fail status.
E (33592) GATTC_MULTIPLE_DEMO: connect device failed, status 133
ESP_GATT_ERROR                  =   0x85,   /* 0x85 */**

 I (3542) GATTC_MULTIPLE_DEMO: 80 00 00 00 00 05
I (3542) GATTC_MULTIPLE_DEMO: TPMS1_000005
I (3542) GATTC_MULTIPLE_DEMO:

I (3552) GATTC_MULTIPLE_DEMO: Searched device TPMS1_000005
I (3562) BT_L2CAP: L2CA_ConnectFixedChnl() CID: 0x0004  BDA: 800000000005
I (3562) GATTC_MULTIPLE_DEMO: Stop scan successfully
W (33572) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x100
W (33572) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x100
W (33572) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x100
I (33582) GATTC_MULTIPLE_DEMO: esp_gattc_cb
I (33582) GATTC_MULTIPLE_DEMO: Start scanning again
I (33582) GATTC_MULTIPLE_DEMO: SVM : start_scan
I (33592) GATTC_MULTIPLE_DEMO: esp_gattc_cb
E (33592) GATTC_MULTIPLE_DEMO: connect device failed, status 133

can you please suggest what could be the reason for this issue?
Kind Ragards,
Tanuja

zhp0406, 2023-11-03:
@chegewara 
I will check it,could you please give me complete log?and how can I reproduce it?