espressif/esp-idf #2904: Why ESP-IDF does use old version of submodules?

IBeRyUS, 2019-01-03:
We are using latest stable version of EDP-IDF but i noticed submodules are not using latest versions. 

Esp-Idf version I'm using is v3.1.2 and it's the latest one provided by the espressif at the moment. Release date of v3.1.2 is 02 January 2019.

Here is the list of some submodules still using the old version:

**1. AWS IoT SDK:**  commit 8bf852d on 08 January 2018. espressif/aws-iot-device-sdk-embedded-C repo has latest AWS version on 06 August 2018 and commit 2991832. I'm testing AWS and using latest version of AWS making my compile dirty. 

**2. BT lib:** sdk included commit 2bf7a0620f on 14 December 2018. Repo latest stable release commit bc66c9d on the same day but inluding more bug fixes.

**3. mbedtls:** sdk included commit b9ada30675 on 30 July 2018. Repo latest stable release commit 19eb57f7 on 18 September 2018

Other submodules also using old versions but they are not forked into espressif repositories, but all the submodules I mentioned above are forked in espressif repos and that commits are made by espressif employees. They are checked and release on submodule's repos but not included in the esp-idf sdk. 

Is there any particular reason for not tracking submodules? 

Thanks,
Ibrahim.

dobairoland, 2019-01-03:
Hello @IBeRyUS. v3.1.2 is a bugfix-only release which means that the submodules are updated only if they contain a bugfix for the v3.1 release. It is normal that it has older submodules in comparison with later releases (new submodule features of v3.2 or v3.3 are not required in v3.1).

It is normal even for master to not have the latest submodule versions because new features/bugfixes are firstly added to the submodule and only later to master.

IBeRyUS, 2019-01-03:
Hi @dobairoland. V3.1.2 is bugifx for release/v3.1 yes it's true. but the submodules i mentioned they are already bug-fixes for v3.1. if you have enough time please check v3.1.1 and v3.1.2 submodule versions. 

one example. if you check esp32 lib you will see same versioning with esp-idf. All the new commits i mentioned are either using version number correlation with esp-idf or standalone repo. 

or why don't you check aws-iot repo on espressif repos. it's updated months ago by @projectgus but not even master branch including latest version. 

I'm manually changing to latest versions of submodules but it's making my compiles dirty and that's annoying. Deploying esp-idf on other machines is another hassle too.

dobairoland, 2019-01-03:
Sorry, maybe I don't get your point or I was not clear with my explanation. Let's take a look at your example:

> 2. BT lib: sdk included commit 2bf7a0620f on 14 December 2018. Repo latest stable release commit bc66c9d on the same day but inluding more bug fixes.

You are right. If we take a look at the release/v3.1 branch of the BT library then we see bc66c9d as the newest commit which is not included in v3.1.2 of ESP-IDF (and not even in the release/v3.1 branch of ESP-IDF). The reason is simple. We have an internal PR which hasn't been merged yet. After merge it gets into the release/v3.1 branch of ESP-IDF and after testing it gets into some of the subsequent v3.1.x releases. 

If you need a bugfix from a submodule then you have to wait until it gets released with ESP-IDF or you can manually checkout-out the given fixes (as you do). But be advised, that untested combination of IDF and submodules may yield issues.

You are right regarding third-party submodules. Those are not updated automatically. We are updating them on request (or plan). The update also may take a while because it cause incompatibility issues and has to be tested with the other components.

IBeRyUS, 2019-01-03:
Esp submodules updating regularly. I know you are making stable versioning for v3.1.x. Esp32-lib, bt-lib and esptool-py are followed by esp-idf submodules but third-party submodules are not followed in the esp-idf.

Another thing if documentation says **latest** and **stable** version is v3.1.x then I expect all submodules to be updated as well.

I need to change AWS, mbedtls after every pull because old versions not working properly with AWS servers. Doing this manual update makes my compilation with dirty flag and sometimes it's hard to understand is it because of my alterations on the sdk or this updates.

If you update third-party submodules by request, can you check and update aws, json and mbedtls ASAP. 

Other third-party submodules also needs checking too.

projectgus, 2019-01-03:
Hi @IBeRyUS,

There are three different things to discuss here: versioning, internal & external libraries.

> Another thing if documentation says latest and stable version is v3.1.x then I expect all submodules to be updated as well.

We have a "master" branch which is the equivalent of the "latest" version in the docs, and stable releases which are older than this. Currently master (new development) is v3.3-dev, we have a prerelease of v3.2-beta1 and the latest stable release ("stable") is v3.1.2. You can read more about the versioning scheme here: https://docs.espressif.com/projects/esp-idf/en/latest/versions.html

For internal (Espressif) libraries (esp32-wifi-lib & esp32-bt-lib), the matching submodule version for a particular commit (version) of ESP-IDF should always be used not any other version or branch from the repo. The very latest esp32-wifi-lib and esp32-bt-lib commits on GitHub usually correspond with the master branch (ie new development, not stable). Sometimes they may even be pushed to github ahead of the corresponding (necessary) source changes in ESP-IDF itself which update the submodule. If these internal library submodules don't match the commits set in the ESP-IDF repo, ESP-IDF will probably not work as expected.

For third party libraries, we try to track upstream versions but there are sometimes delays because we need to test any updates still work with ESP-IDF. For the supported stable (tested) version of ESP-IDF, please use the submodules at the versions provided. If you find something doesn't work (either because the library is out of date or for some other reason), please open an issue with details and we'll fix it (possibly by updating the library submodule or possibly by some other method).

For bugfix releases like v3.1.2, we only update to fix bugs found since the v3.1 release, to avoid unexpected behaviour changes for users. This applies to both ESP-IDF code and upstream libraries.

So currently AWS IoT is based on v2.2.1 in the v3.1.x releases and based on AWS IoT v3.0.1 (the current stable AWS IoT release) in master. If something isn't working in ESP-IDF v3.1.2 with the provided AWS IoT version, please open an issue with the details.

For mbedTLS, ESP-IDF v3.1.x releases are using the v2.12.0 mbedTLS release and ESP-IDF master is using mbedTLS v2.13.1. I notice we're about six weeks behind on mbedTLS minor versions, so we will update the mbedTLS version used in the master branch soon - thanks for the reminder. However we probably won't update mbedTLS to a new minor version in ESP-IDF bugfix releases unless there are known bugs which are fixed by updating - so if you experience problems with the provided mbedTLS version then please open an issue with those details as well.

projectgus, 2019-01-06:
> Sorry but I'm not evaluating and testing esp32 and sdk for  single chip. It's for massive production and I wouldn't use master or  any other dev branch for that.

I certainly wouldn't argue with this approach, this is what we recommend as well. However it's also why we don't update third party libraries to the latest versions automatically - code changes are risky, and we're conservative about doing them to already released versions.

> v3.1, v3.2 and v3.3 how do different they are? If you make these you should expect different modules or functionalities between them or changing your structure so they won't be compatible like v2.x and v3.x.

The main difference is v3.2 and v3.3 have not been released yet (they're beta and development versions, respectively).

We follow Semantic Versioning, so updating to a new minor version (ie v3.1 -> v3.2) will add functionality but should not require any code changes (no breaking of existing public APIs). You can read more about this in the [ESP-IDF Documentation](https://docs.espressif.com/projects/esp-idf/en/latest/versions.html#versioning-scheme)

However, many customers prefer to stay with a single minor version (ie v3.1) to minimize any re-testing requirements on their end. For this reason we also make bugfix releases (ie v3.1.1->v3.1.2). These don't add additional functionality, only fix bugs.

> If you look from my perspective I have two options to use your SDK.
>
> 1- Use official release v3.1.x and stuck with older versions of external libraries and write so many > workaround code because I will have more reliable and stable BT&WiFi library. (Yes we are testing > coexistence of BT and WiFi and planning to use on our product)
>
> 2- Use release candidate or dev or even master branch for latest external libraries but have more buggy and less relieable BT&WiFi functionality.

Yes. This is essentially the tradeoff of almost all software development - use the very latest development (and risk the "bleeding" part of the "bleeding edge"), or use the older stable version and miss out on new features.

Depending on your development schedule you may find a good compromise is to start using a prerelease branch (ie v3.2-beta2) for now and report any bugs you find, and update to v3.2 when the final version is released (in the coming few weeks).

> I looked for porting AWS library before then i realised you already ported newer version of it but not included in v3.1.x. Instead of porting i manually pulled the latest commit.

What functionality do you need which is not in AWS IoT SDK v2.2.1?

> Reporting issues is another problem. I submitted two bugs. One solved for master branch not included in v3.1.2, second one nobody even acknowledge the issue and write a comment.

Sorry about that. I'll find out why there's been no reply to the second issue yet.

projectgus, 2019-03-12:
Closing due to lack of response.

AWS IoT SDK v3.0.1 (latest upstream release) can be found in ESP-IDF master branch and prereleases v3.2-beta3 and v3.3-beta2. This version will be v3.2 & v3.3 final releases once they are available.

closed 2019-03-12 by projectgus