espressif/esp-idf #2907: What is the meaning of MBEDTLS_ERR_SSL_CONN_EOF?

sysizlayan, 2019-01-03:
When I try to connect with https client, I am taking this error log

D (17906) esp-tls: host:192.168.1.137: strlen 13
D (17956) esp-tls: handshake in progress...
E (17963) esp-tls: mbedtls_ssl_handshake returned -0x7280
E (17963) esp-tls: Verify Cert!
E (17963) esp-tls: Test1
I (17965) esp-tls: Certificate verified.
E (17971) esp-tls: Failed to open new connection
E (17975) TRANS_SSL: Failed to open a new connection
E (17981) HTTP_CLIENT: Connection failed, sock < 0
I (17986) K_pairThread: HTTP_EVENT_DISCONNECTED

The error code -0x7280 means MBEDTLS_ERR_SSL_CONN_EOF according to mbedtls files. What is the meaning of that? I could not find any explanation for that.

Thanks in advance.

Alvin1Zhang, 2019-03-13:
@sysizlayan Hi, would you help provide any updates for the issue? Thanks.

Alvin1Zhang, 2019-05-17:
@sysizlayan Hi, would you help provide any updates for the issue? Are you able to access any logs or other information from the server end? Thanks.

sysizlayan, 2019-05-17:
The problem was because of the server which I tried to reach does not recognize the root certificate that I have signed myself. As a result, the server rejects the connection when I try to reach.

I forgot to close the issue. Sorry for the late response.

closed 2019-05-17 by Alvin1Zhang

MATTYGILO, 2023-07-17:
> hi! i am getting this error i tried lot to rectify it but i was unable to do it
> 
> Traceback (most recent call last): File "", line 49, in File "", line 42, in create File "urequests.py", line 120, in post File "urequests.py", line 62, in request OSError: (-29312, 'MBEDTLS_ERR_SSL_CONN_EOF')
> 
> > > >
> 
> can any one help me out from this ....pls
> 
> CODE :
> 
> `import network import time import urequests
> 
> timeout = 0
> 
> wifi = network.WLAN(network.STA_IF) wifi.active(True)
> 
> wifi.connect("DLink_39_9","vgdlink@1234")
> 
> if not wifi.isconnected(): print('connecting..') while (not wifi.isconnected() and timeout < 5): print(5 - timeout) timeout = timeout + 1 time.sleep(1)
> 
> if(wifi.isconnected()): print('Connected') req = urequests.post('https://www.example.com') print(req.status_code) print(req.text) else: print('Time Out') print('Not Connected')`

Did you find a fix for this?

Rainbow-Cai, 2024-03-20:
> It means the underlying TCP socket was closed (returned an EOF - End Of File - result). It looks like the initial TCP connection was established ("handshake in progress..." line), and then the server closed the socket during the handshake for some reason (maybe something with the parameters offered by the client?).
> 
> Are you able to access any logs or other information from the server end?

Hi ，I just verified the setting of system properties.
IoTHubMessage_SetContentTypeSystemProperty(msg_handle,
            "application/JSON")

Whenever "/" character is present in system property I am getting the following error.
The Error log report “E (89095) Dynamic Impl: mbedtls_ssl_fetch_input error=29312”

If I use "application@JSON" or "application-JSON" it's working filen. May I know the reason for rejecting "/" character?

Rainbow-Cai, 2024-03-20:
> It means the underlying TCP socket was closed (returned an EOF - End Of File - result). It looks like the initial TCP connection was established ("handshake in progress..." line), and then the [...]

Hi @projectgus 

I just verified the setting of system properties.
IoTHubMessage_SetContentTypeSystemProperty(msg_handle,
            "application/JSON")

Whenever "/" character is present in system property I am getting the following error.
The Error log report “E (89095) Dynamic Impl: mbedtls_ssl_fetch_input error=29312”

If I use "application@JSON" or "application-JSON" it's working filen. May I know the reason for rejecting "/" character?