espressif/esp-idf #2909: Reliable Promiscuous CSI extraction

dakami, 2019-01-03:
It's implied in wifi.rst that CSI can be extracted from sniffed packets, and I do intermittently get boots where random frames fire the CSI callback.  I'm definitely getting promiscuous frames (*that* callback fires), and playing with the promisc filters doesn't seem to yield more frames.  (I do eventually get "unknown CSI error!" though.)  I've also tried various orderings of esp_wifi_start and esp_wifi_connect, as well as both the NULL and STA modes.

Any ideas?

Alvin1Zhang, 2019-03-14:
@dakami Hi, would you help share if any updates for this issue? Thanks.

jonathanmuller, 2019-04-03:
@TimXia Here is a demo, I would be glad to give you more informations if needed
@dakami What filters did you try ?

### TLDR : 
I have the same issue and a stable setup to reproduce it!

### Details :
Hi, I have exactly the same issue :
- CSI callback is called for a non-CSI frame in promiscuous mode

## Hardware : 
ESP-WROVER-KIT V4.1 and SparkFun ESP32 Thing (both have the same comportment)
IDF : Master (03.04.2019), commit 2f8b6cfc77bdb2c5c412f17529d5fddf9b68b325
## Code to reproduce the error : 
https://github.com/jonathanmuller/ESP32-gather-channel-state-information-CSI-/blob/master/csi/main/csi.c
## Example of the error :
Here is the content the "wifi_csi_info_t *data" field "buf" for consecutive frames 
```
void receive_csi_cb(void *ctx, wifi_csi_info_t *data) {
	wifi_csi_info_t received = data[0];
	char senddMacChr[LEN_MAC_ADDR] = {0}; // Sender
	sprintf(senddMacChr, "%02X:%02X:%02X:%02X:%02X:%02X", received.mac[0], received.mac[1], received.mac[2], received.mac[3], received.mac[4], received.mac[5]);
	printf("Received CSI from adress %s\n", senddMacChr); 
	uint8_t* my_ptr=&(data->buf);
	printf("0000 ");
	for(int i=0;i<data->len;i++){
		printf("%02x ", my_ptr[i]);
	}
	printf("\n\n");
}
```
Result (note that the leading "0000" is added by my code to facilitate conversion to Wireshark)
```
Received CSI from adress 8C:0C:90:01:1E:C8
0000 14 a0 fc 3f 80 00 d8 3a f9 ff fd 3f f8 ff fd 3f 00 94 05 4e 00 00 00 00 90 72 fb 3f f9 ff fd 3f f8 ff fd 3f fd 8f 80 86 03 2f 3d 6a 6a be be 46 9d c4 d2 86 75 5a bd 76 de 1c 5e 7b 21 db 5f 7b 9a 68 70 a5 51 cd 79 18 3f 79 dc 1b b8 82 b9 1d 9d 9a 63 76 4f e5 9c 35 81 5a c9 c0 31 c3 e2 e9 2b c3 03 e2 0a 74 39 40 12 10 e2 5e f4 c1 48 95 57 9d 29 8e 41 87 28 1c 30 13 fd 60 79 ec 8c c9 

Received CSI from adress 00:00:90:01:1E:C8
0000 c0 a8 fc 3f 80 00 5c 6a f9 ff fd 3f f8 ff fd 3f 00 00 01 01 00 00 00 00 90 72 fb 3f f9 ff fd 3f f8 ff fd 3f b6 ca 28 14 55 9a d1 0a a4 ca 52 26 55 aa 0f c9 b4 d1 b3 39 d3 84 0c 1b 80 3c e6 91 5e 12 54 7d 38 51 bd ae d0 17 e6 e6 65 9d 9f 65 e2 54 71 88 d2 66 11 ba 81 73 2d f5 fe fe b8 a7 30 7f 72 de 20 1d 5d 59 66 16 54 fc e8 12 3d 95 d5 f1 84 ad 31 28 37 f4 2e c1 41 c6 04 5a 85 59 

Received CSI from adress 8C:0C:90:01:1E:C8
0000 6c b1 fc 3f 80 00 d8 3a f9 ff fd 3f f8 ff fd 3f 00 94 05 4e 00 00 00 00 90 72 fb 3f f9 ff fd 3f f8 ff fd 3f fd 8f 80 86 03 2f 3d 6a 6a be be 46 9d c4 d2 86 75 5a bd 76 de 1c 5e 7b 21 db 5f 7b 9a 68 70 a5 51 cd 79 18 3f 79 dc 1b b8 82 b9 1d 9d 9a 63 76 4f e5 9c 35 81 5a c9 c0 31 c3 e2 e9 2b c3 03 e2 0a 74 39 40 12 10 e2 5e f4 c1 48 95 57 9d 29 8e 41 87 28 1c 30 13 fd 60 79 ec 8c c9 

Received CSI from adress 00:00:90:01:1E:C8
0000 18 ba fc 3f 80 00 5c 6a f9 ff fd 3f f8 ff fd 3f 00 00 01 01 00 00 00 00 90 72 fb 3f f9 ff fd 3f f8 ff fd 3f 14 72 d7 0e ce 5d 91 c0 5c 91 c0 5b 91 c0 5a 10 f0 78 5f a4 e0 ac fe 6b 05 67 9d 1f 5d 10 67 bd 94 d0 9d f0 99 92 a7 0a 60 a6 6c 4f 57 66 5a ff 2d 32 13 67 a4 a3 72 46 1a 32 fa dd 8c f4 07 22 7e 17 7e ac 7a f3 1a a5 96 56 33 ba 8d 1a c9 47 1a 66 6c 29 11 cf d4 6c 95 80 65 e6 

Received CSI from adress 8C:0C:90:01:1E:C8
0000 1c d4 fc 3f 80 00 6d 3f f9 ff fd 3f f8 ff fd 3f 00 c2 d7 41 00 00 00 00 90 72 fb 3f f9 ff fd 3f f8 ff fd 3f 82 38 46 dd c7 f0 f3 33 ed a9 7a 6b 18 b4 1b 0e ed 7a b3 ee b5 10 dd 9b ff 38 90 77 8f 2d 32 f5 3d ad 61 64 11 ae f0 17 f7 32 00 00 f6 c8 58 00 f9 ff fd 3f f8 ff fd 3f 00 00 00 00 a3 00 01 41 49 ef 8b 09 f5 c7 8f b0 00 80 20 70 02 26 60 00 8c 0c 90 01 1e c8 01 00 6c b1 fc 3f 

Received CSI from adress F0:79:60:0E:C7:FC
0000 c8 dc fc 3f 80 00 40 9f f9 ff fd 3f f8 ff fd 3f 00 be 9c 85 00 00 00 00 90 72 fb 3f f9 ff fd 3f f8 ff fd 3f ca f4 a9 8e ed 69 94 12 47 79 80 df 1e b4 a8 8f 26 41 a1 c9 93 5d 43 7f 35 30 11 46 14 18 b4 9a 3b 8e 9d db 5a b0 e3 14 2c 37 0d d2 c3 1b 7b 0a 71 6c 9c b8 48 fe 07 cf 9c cb b4 67 0c d7 4f af e5 96 b8 e8 ef 98 1e 63 c7 68 ae 28 15 a5 09 aa b5 1c 75 58 d0 0f 00 c1 a6 66 71 25 

Received CSI from adress 8C:0C:90:01:1E:C8
0000 74 e5 fc 3f 80 00 d8 3a f9 ff fd 3f f8 ff fd 3f 00 94 05 4e 00 00 00 00 90 72 fb 3f f9 ff fd 3f f8 ff fd 3f fd 8f 80 86 03 2f 3d 6a 6a be be 46 9d c4 d2 86 75 5a bd 76 de 1c 5e 7b 21 db 5f 7b 9a 68 70 a5 51 cd 79 18 3f 79 dc 1b b8 82 b9 1d 9d 9a 63 76 4f e5 9c 35 81 5a c9 c0 31 c3 e2 e9 2b c3 03 e2 0a 74 39 40 12 10 e2 5e f4 c1 48 95 57 9d 29 8e 41 87 28 1c 30 13 fd 60 79 ec 8c c9 

Received CSI from adress 00:00:90:01:1E:C8
0000 d8 7e fb 3f 80 00 5c 6a f9 ff fd 3f f8 ff fd 3f 00 00 01 01 00 00 00 00 90 72 fb 3f f9 ff fd 3f f8 ff fd 3f fc b5 31 6a a6 cf 3a 80 26 11 bd 6b 59 40 f4 2e ba 20 5c 57 6b 87 18 29 18 28 94 54 1e 3c 03 ca 43 6c f5 37 5c a1 d9 db 53 a3 d8 26 7e ed 59 6e cd 66 8a cd d7 ec a6 2f 2b 81 08 fd 08 e5 5c b0 73 2b 53 69 a4 17 6d 12 cf e6 ba 42 93 30 65 01 88 54 2f 26 28 d8 b3 72 e0 1d 57 bc 

Received CSI from adress F0:79:60:0E:C7:FC
0000 30 90 fb 3f 80 00 30 00 f9 ff fd 3f f8 ff fd 3f 00 01 1e c8 00 00 00 00 90 72 fb 3f f9 ff fd 3f f8 ff fd 3f 08 00 45 00 05 dc f6 38 40 00 35 06 d4 99 11 fd 4f ca 0a a0 08 e3 00 50 ec 85 8c 77 8b 2b 5c 6a 36 a3 80 10 00 79 29 34 00 00 01 01 08 0a 84 45 3c b4 1b b6 3d 5c 69 64 d1 e6 8b 7a 14 0b b8 3b f2 e4 40 94 e9 2d 34 d2 ba 87 38 ca 6d 25 32 1d 9e f5 c9 46 59 9d 44 98 5b f0 36 11 

Received CSI from adress 00:00:67:AF:C7:FC
0000 b8 74 fc 3f 80 00 8c 74 f9 ff fd 3f f8 ff fd 3f 00 00 01 01 00 00 00 00 90 72 fb 3f f9 ff fd 3f f8 ff fd 3f f9 ff fd 3f f8 ff fd 3f f9 ff fd 3f f8 ff fd 3f 08 00 45 00 05 dc f6 38 40 00 35 06 d4 99 11 fd 4f ca 0a a0 08 e3 00 50 ec [... cut ...]  01 74 14 82 eb b7 e2 de f3 4b 12 66 dc 7e 76 fb 90 90 
Which is equivalent to this mask : (_=changes, m x=same)
0000 __ __ x_ xx xx xx __  __ xx xx xx xx xx xx xx xx xx __ __ __ xx xx xx xx xx xx xx xx xx xx xx xx __ __ __ __ __ __ __ ...

The "80 00" (beacon frame) at the start of the frame let me think that a mask is wrong, which result in this semi-masking 
In addition, sometime I get the ESSID of my network in the CSI data, so I expect the masking is wrong for some beacon frames.

Thank you

jonathanmuller, 2019-04-03:
After further investigation CSI frames with 
rx_ctrl.sig_mode==1 are valid
rx_ctrl.sig_mode==0 are valid

So the bug is probably somewhere in here

kraoa, 2019-04-16:
Hi even I'm noticing the same issue as above namely in the CSI callback, I'm getting non CSI frames. 
Can someone from espressif acknowledge that this bug exists and also how do we solve this. 

Check out the following link
https://github.com/espressif/esp-idf/issues/3267

Also do we need to enable promiscuous mode to receive CSI. I disabled promiscuous mode and only enabled CSI mode. The code wasn't crashing but in the CSI callback I wasn't receiving any frames.

zhangyanjiaoesp, 2020-12-09:
@jonathanmuller 

1. 
> Hi, I have exactly the same issue :
> CSI callback is called for a non-CSI frame in promiscuous mode

What do you mean the `non-CSI frame` in the CSI callback? How do you judge it?

2. 
I have tested locally using the demo    https://github.com/jonathanmuller/ESP32-gather-channel-state-information-CSI-/blob/master/gather_csi_in_promiscuous/main/csi.c, and I get the following data:
`
<CSI><address>18:FE:34:00:1A:4D</address><len>384</len>
0000 108 -60 70 0 0 0 0 0 0 0 -1 -2 -2 -5 -2 -5 -2 -6 -2 -6 -2 -7 -2 -7 -2 -8 -2 -8 -1 -8 -1 -8 -1 -9 -1 -9 -1 -9 -1 -9 -1 -9 -1 -9 -1 -9 -1 -9 -1 -10 -1 -10 -1 -10 -1 -10 -2 -10 -2 -10 -2 -10 -2 -10 -1 -5 -2 -10 -2 -10 -2 -10 -2 -10 -2 -10 -2 -10 -2 -10 -2 -10 -3 -10 -3 -10 -3 -9 -3 -9 -2 -9 -3 -9 -2 -9 -1 -10 -1 -9 -2 -8 -2 -8 -2 -8 -2 -8 -2 -7 -2 -7 -1 -7 -1 -6 -1 -6 -1 -2 0 0 0 0 0 0 0 0 -1 -1 -1 -1 -6 -4 -5 -5 -5 -5 -4 -5 -3 -6 -3 -6 -2 -7 -2 -8 -1 -8 -1 -8 -1 -9 0 -10 0 -10 0 -11 0 -12 0 -13 0 -14 1 -14 0 -15 0 -16 -1 -17 -1 -18 -1 -19 -1 -20 -1 -21 -1 -22 -2 -23 -2 -24 -3 -26 -3 -28 -2 -29 -4 -30 -5 -32 -5 -33 -5 -34 -4 -36 -4 -37 -4 -38 -4 -39 -4 -40 -5 -40 -5 -41 -5 -42 -6 -43 -6 -44 -6 -45 -6 -46 -6 -48 -4 -48 -4 -47 -4 -48 -4 -48 -4 -48 -4 -49 -5 -51 -4 -53 -1 -55 -1 -15 0 0 0 0 0 0 -1 -1 0 0 0 0 -1 -1 -1 -1 -1 -1 -1 -3 -3 -9 -3 -10 -3 -11 -3 -12 -3 -12 -3 -13 -3 -14 -2 -14 -2 -15 -2 -16 -1 -16 -1 -17 -1 -17 -1 -17 -1 -17 -1 -17 -1 -17 -2 -17 -1 -18 -1 -18 -1 -18 -1 -18 -2 -18 -2 -19 -2 -19 -3 -19 -2 -19 -2 -19 -2 -19 -3 -19 -3 -18 -3 -19 -3 -19 -3 -19 -3 -19 -4 -19 -4 -18 -4 -18 -3 -18 -4 -18 -4 -17 -4 -16 -5 -16 -3 -16 -2 -16 -3 -15 -3 -14 -3 -14 -2 -14 -2 -13 -2 -13 -1 -12 -1 -11 1 -11 1 -11 1 -10 1 -9 0 -2 
`
`
<CSI><address>18:FE:34:00:1A:4D</address><len>384</len>
0000 108 -60 70 0 0 0 0 0 0 0 0 -1 3 -4 4 -4 5 -5 5 -5 6 -5 6 -5 7 -5 7 -5 7 -5 8 -5 8 -5 8 -5 9 -5 9 -5 9 -5 9 -5 9 -5 9 -5 9 -5 9 -6 9 -6 9 -6 9 -6 10 -7 10 -7 10 -7 5 -4 10 -7 9 -7 9 -7 9 -7 9 -7 9 -8 9 -8 9 -8 8 -8 8 -8 8 -8 8 -8 8 -7 8 -7 9 -6 10 -5 8 -6 7 -7 7 -7 6 -6 6 -6 6 -6 6 -5 6 -5 6 -5 5 -4 1 -1 0 0 0 0 0 0 0 0 -1 -1 -1 -1 0 -8 0 -7 1 -7 2 -7 2 -7 3 -7 4 -7 5 -7 6 -6 7 -6 7 -7 8 -7 9 -7 9 -8 10 -8 11 -9 12 -9 13 -9 13 -11 14 -12 15 -13 16 -14 17 -14 18 -15 18 -17 19 -18 21 -18 21 -19 22 -20 23 -22 27 -20 25 -23 25 -26 25 -28 26 -29 29 -29 31 -31 32 -33 33 -34 33 -34 33 -36 34 -36 35 -36 34 -37 35 -38 36 -40 37 -40 38 -40 39 -41 40 -40 40 -40 40 -41 40 -43 40 -44 40 -43 44 -40 51 -39 13 -10 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 1 -2 7 -8 8 -8 9 -8 10 -9 11 -9 12 -9 13 -9 14 -9 15 -9 15 -9 16 -9 16 -9 17 -9 17 -10 17 -9 17 -10 17 -10 17 -10 17 -11 17 -11 18 -11 18 -11 18 -11 18 -12 19 -12 18 -13 19 -13 18 -12 18 -13 18 -13 17 -13 17 -13 17 -13 17 -14 17 -14 16 -14 16 -14 16 -14 16 -13 16 -13 15 -13 17 -13 19 -14 16 -13 13 -12 13 -12 12 -11 12 -11 12 -10 12 -10 11 -9 11 -8 10 -7 11 -6 10 -6 9 -5 8 -4 0 -1 
`
`
<CSI><address>18:FE:34:00:1A:4D</address><len>384</len>
0000 108 -60 70 0 0 0 0 0 0 0 -1 -2 -1 -6 -1 -6 -1 -6 -1 -7 -1 -7 -1 -8 -1 -8 0 -9 0 -9 0 -9 -1 -10 0 -10 0 -10 0 -10 0 -10 0 -11 0 -11 0 -11 0 -11 0 -11 0 -11 0 -11 0 -11 0 -12 -1 -12 -1 -12 -1 -6 0 -12 -1 -12 -1 -12 -1 -12 -1 -12 0 -12 0 -12 -1 -12 -1 -12 -2 -12 -2 -12 -1 -11 -1 -11 -1 -11 -2 -12 -2 -14 -1 -12 -1 -10 -1 -10 -1 -9 -1 -9 -1 -9 -1 -9 -1 -8 -1 -8 0 -8 0 -2 0 0 0 0 0 0 0 0 -1 -1 -1 -1 -4 -2 -4 -2 -3 -2 -2 -3 -2 -3 -2 -3 -2 -3 -1 -3 -1 -4 -1 -4 0 -4 0 -5 0 -5 0 -5 0 -5 1 -6 1 -7 1 -7 1 -7 1 -8 1 -8 1 -8 1 -9 1 -10 1 -10 1 -11 1 -11 1 -12 1 -13 1 -14 2 -15 1 -15 1 -16 1 -16 1 -17 1 -18 2 -19 3 -19 3 -20 3 -20 3 -21 3 -22 3 -23 3 -23 2 -24 2 -24 1 -25 2 -28 4 -27 6 -23 6 -21 8 -21 6 -24 6 -25 9 -25 8 -26 6 -26 1 -7 -1 -1 -1 -1 0 0 0 0 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -2 -1 -6 -1 -6 -1 -6 -1 -7 -1 -7 -1 -7 -1 -8 -1 -8 0 -8 0 -9 -1 -9 -1 -9 0 -9 0 -10 0 -10 0 -10 0 -10 0 -10 0 -10 0 -11 0 -11 0 -11 0 -11 0 -11 0 -11 -1 -12 -1 -12 0 -11 0 -12 0 -12 -1 -11 0 -11 0 -11 -1 -11 -1 -11 -1 -11 -1 -11 -1 -11 -1 -11 -1 -11 -1 -11 -2 -11 -3 -12 -2 -11 -1 -10 -1 -9 -1 -9 -1 -9 -1 -8 -1 -8 -1 -8 -1 -8 0 -7 0 -7 0 -7 0 -6 0 -6 0 -1 
`
`
<CSI><address>18:FE:34:00:1A:4D</address><len>384</len>
0000 108 -60 70 0 0 0 0 0 0 0 1 0 4 0 5 0 5 0 6 0 6 0 7 1 7 1 7 1 8 1 8 1 8 2 9 2 9 2 9 2 9 2 9 2 9 3 9 3 10 2 10 3 10 3 10 3 10 3 11 3 11 3 11 3 5 1 11 3 11 3 11 3 11 3 11 3 11 3 11 3 11 3 10 3 10 3 11 2 11 2 11 2 11 2 9 2 9 2 9 3 9 2 9 3 8 3 8 3 8 3 7 3 7 3 7 2 6 2 1 0 0 0 0 0 0 0 0 0 -1 -1 0 -1 5 -7 6 -7 6 -6 6 -5 5 -4 6 -3 6 -2 6 -1 6 0 6 0 6 1 6 1 7 2 7 3 8 4 8 5 8 6 8 6 9 7 10 7 11 7 12 8 13 9 13 10 14 11 15 11 15 11 16 12 17 13 19 14 20 15 20 16 22 17 23 17 24 18 24 20 24 22 25 22 26 22 27 24 29 26 32 27 32 26 31 26 31 27 32 30 31 33 32 32 34 32 31 35 26 31 26 28 28 36 30 47 33 51 31 50 26 50 6 12 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 2 0 9 1 9 1 10 1 11 2 12 2 12 2 13 2 14 3 14 3 15 4 16 4 16 5 17 5 17 5 17 5 18 6 18 6 18 5 18 5 18 6 18 6 18 6 19 6 20 6 20 6 21 5 21 6 20 6 21 6 21 6 20 6 20 6 20 6 20 6 20 6 20 6 20 6 20 6 20 6 20 5 20 6 17 6 14 6 17 5 18 5 16 5 16 5 15 5 15 5 15 5 14 5 13 5 12 5 13 6 12 5 10 6 9 6 1 0
`
I didn't see the mask you said.

3. 

> After further investigation (and help) CSI frames with
> rx_ctrl.sig_mode==1 are valid
> rx_ctrl.sig_mode==0 are INvalid
 
What do you mean by saying `valid` and `invalid` ?

jonathanmuller, 2020-12-14:
@zhangyanjiaoesp Hi, 

> What do you mean the `non-CSI frame` in the CSI callback? How do you judge it?
I can't say if they are CSI frames filled with gibberish data or other type of frame but 
- There content make no sense (plotting them shows no continuity on the phase or the amplitude)
- "Coincidently" they contain some field of beacon data (beacon header and ESSID)

So I believe there is data in those frame that don't bellong to them 
 
> I have tested locally using the demo https://github.com/jonathanmuller/ESP32-gather-channel-state-information-CSI-/blob/master/gather_csi_in_promiscuous/main/csi.c, and I get the following data
> I didn't see the mask you said.

Maybe you were lucky because your environment doesn't match mine, maybe it was fixed.
Maybe do you have an simple setup we can use to check one from another ?

> > After further investigation (and help) CSI frames with
> > rx_ctrl.sig_mode==1 are valid
> > rx_ctrl.sig_mode==0 are INvalid
> 
> What do you mean by saying `valid` and `invalid` ?

Valid = phase and amplitude are continuous
Invalid = gibberish (that looks part of a beacon frame) data in the frame that makes the phase and amplitude random (and not continuous)

zhangyanjiaoesp, 2020-12-14:
@jonathanmuller 

My test info is as follows:
1. demo: https://github.com/jonathanmuller/ESP32-gather-channel-state-information-CSI-/blob/master/gather_csi_in_promiscuous/main/csi.c, your original link https://github.com/jonathanmuller/ESP32-gather-channel-state-information-CSI-/blob/master/csi/main/csi.c can be found
2. master IDF, commit is 75d32ec0519fb24b658ec55e9d79ac78dc53ffbf
3. ESP32_DevKitc_v4 / ESP-WROVER-KIT V4.1
3. I didn't do any special setup, just enable the CSI in menuconfig `Component config -> WiFi -> WiFi CSI` and then run the demo on esp32.

Alvin1Zhang, 2021-02-24:
@dakami Thanks for reporting and sharing updates, will close due to short of feedback, feel free to reopen with more updates.

closed 2021-02-24 by Alvin1Zhang