espressif/esp-idf #2910: Building clean project via Eclipse fails

william-ferguson-au, 2019-01-04:
## Environment

- Development Kit:      None
- Kit version NA
- Module or chip used:  ESP32-WROOM-32
- IDF version v3.3-beta1-136-g97eecfa1b

- Build System:         Make
- Compiler version  (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     Windows
- Power Supply:         NA

## Problem Description

//Detailed problem description goes here.
Building clean via Eclipse fails with:

```
16:16:32 **** Incremental Build of configuration Default for project wt_starter_esp_idf_2 ****
make -j8 all 
including /home/William/esp/esp-idf/components/app_update/Makefile.projbuild...
including /home/William/esp/esp-idf/components/bootloader/Makefile.projbuild...
including /home/William/esp/esp-idf/components/bootloader_support/Makefile.projbuild...
including /home/William/esp/esp-idf/components/coap/Makefile.projbuild...
including /home/William/esp/esp-idf/components/esp32/Makefile.projbuild...
including /home/William/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including /home/William/esp/esp-idf/components/mbedtls/Makefile.projbuild...
including /home/William/esp/esp-idf/components/nghttp/Makefile.projbuild...
including /home/William/esp/esp-idf/components/partition_table/Makefile.projbuild...
including /home/William/esp/esp-idf/components/ulp/Makefile.projbuild...
including /home/William/esp/esp-idf/components/unity/Makefile.projbuild...
MAKEFLAGS="" CC=cc LD=ld \
/usr/bin/make -C /home/William/esp/esp-idf/tools/kconfig
make[1]: Entering directory '/home/William/esp/esp-idf/tools/kconfig'
cc -c  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MMD  /home/William/esp/esp-idf/tools/kconfig/mconf.c -o mconf.o
cc: error: spawn: No such file or directory
make[1]: *** [Makefile:171: mconf.o] Error 1
make[1]: Leaving directory '/home/William/esp/esp-idf/tools/kconfig'
make: *** No rule to make target '/home/William/esp/esp-idf/tools/kconfig/conf-idf', needed by '/home/William/esp/wt_starter_esp_idf_2/build/include/config/auto.conf'.  Stop.
"make -j8 all" terminated with exit code 2. Build might be incomplete.

16:16:37 Build Failed. 2 errors, 0 warnings. (took 5s.838ms)
```
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2726156/sdkconfig.txt)

In order to allow Eclipse to build, I first need to execute: make -j8 all via the Mingw32 command line.

### Expected Behavior
Eclipse should be able to build projects from clean too.

### Actual Behavior
Eclipse fails with error above

### Steps to reproduce

1.  Import project into Eclipse following instructions from https://docs.espressif.com/projects/esp-idf/en/latest/get-started/eclipse-setup.html
2. Build clean from Eclipse

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)

yongjizhou, 2019-02-28:
I am having the same issue

prasad-alatkar, 2019-03-07:
Hi @william-ferguson-au , I did not observe the issue. 
``` 
Build Finished. 0 errors, 0 warnings. (took 1m:46s.817ms)
```
Eclipse version: 2018-09(4.9.0)

william-ferguson-au, 2019-04-27:
I'm pretty sure this was caused by the Eclipse clean wiping out the files generated by ```make menuconfig```.

Running ```make menuconfig``` then allowed the build to work.

closed 2019-04-27 by william-ferguson-au