espressif/esp-idf #2916: event: unexpected system event 63!

i-franzki, 2019-01-05:
## Environment

- Development Kit:       ESP32-PICO-Kit
- Module or chip used:  ESP32-PICO-D4
- IDF version: v3.3-beta1-172-g52e35538f
- Build System:         Make
- Compiler version: 1.22.0-80-g6c4433a5
- Operating System:     Windows
- Power Supply:         USB or external 5V

## Problem Description

I have an ESP32 application that initializes WiFi in AP mode. Once a station (e.g. my laptop computer) connects to the AP, I get the following warning/error messages (via make monitor):
```
SYSTEM_EVENT_AP_START
W (40074) event: unexpected system event 63!
E (40075) event: mismatch or invalid event, id=63
E (40075) event: default event handler failed!
SYSTEM_EVENT_AP_STACONNECTED
station:10:0b:a9:f0:33:94 join, AID=1
SYSTEM_EVENT_AP_STAIPASSIGNED
```
Note: The messages without the log prefix are from my application and show the expected system events before and after the error messages.

This happens since I updated my esp-idf from the master branch at January 4, 2019. Before that I was running with the version from master pulled at December 26, 2018. It did not happen back then. Only since I updated it on January 4, 2019 it happens. Looks like some changes pushed since December 28, 2018 causes this messages.

Other than these messages it does not cause any harm, at least I did not notice anything.

### Expected Behavior
No warning/error messages in the log

### Actual Behavior
```
SYSTEM_EVENT_AP_START
W (40074) event: unexpected system event 63!
E (40075) event: mismatch or invalid event, id=63
E (40075) event: default event handler failed!
SYSTEM_EVENT_AP_STACONNECTED
station:10:0b:a9:f0:33:94 join, AID=1
SYSTEM_EVENT_AP_STAIPASSIGNED
```
### Steps to repropduce
Start WiFi in AP mode, then connect with another WiFi station to that network. As soon as the station got accepted, the messages appear.

## Debug Logs
```
SYSTEM_EVENT_AP_START
W (40074) event: unexpected system event 63!
E (40075) event: mismatch or invalid event, id=63
E (40075) event: default event handler failed!
SYSTEM_EVENT_AP_STACONNECTED
station:10:0b:a9:f0:33:94 join, AID=1
SYSTEM_EVENT_AP_STAIPASSIGNED
```

i-franzki, 2019-01-05:
For info: The same messages are shown in the log for issue https://github.com/espressif/esp-idf/issues/2915

detly, 2019-01-13:
Git bisect says the culprit is e3833999ea6bc622347f576afba13d62dc469208

detly, 2019-01-15:
@liuzfesp That's what it seemed like, thanks for confirming.

fanhuafeng, 2019-02-09:
I have a  esp32_devkitc_v4 board, I use wifi manager example "OnDemandConfigPortal.ino",when board start display info:
 Starting
*WM: 
*WM: Configuring access point... 
*WM: OnDemandAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
E (26277) event: mismatch or invalid event, id=63
E (26278) event: default event handler failed!
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4011195c: 0008e0fe 00000000 ffffaa00
Core 0 register dump:
PC      : 0x40111960  PS      : 0x00060830  A0      : 0x80112796  A1      : 0x3ffb5570  
A2      : 0x3ffc3a70  A3      : 0x3ffc8250  A4      : 0x3ffc82d8  A5      : 0x400df924  
A6      : 0x3ffc3ca4  A7      : 0x3ffc82d8  A8      : 0x80111960  A9      : 0x3ffb5550  
A10     : 0x00000000  A11     : 0x3ffc3a38  A12     : 0x00000006  A13     : 0x3ffc8280  
A14     : 0x0000002f  A15     : 0x3ffc8250  SAR     : 0x00000010  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x40111960:0x3ffb5570 0x40112793:0x3ffb5660 0x401320a5:0x3ffb5680 0x400887ce:0x3ffb56d0 0x4008e749:0x3ffb5710

Rebooting...
ets Jun  8 2016 00:22:57

liuzfesp, 2019-03-04:
@fanhuafeng please update your IDF to latest one.

Alvin1Zhang, 2019-03-22:
@i-franzki @fanhuafeng Hi, would you help share if any updates for the issue? Has your issue been resolved? Thanks.

i-franzki, 2019-03-24:
Updating the IDF solved the problem for me. The messages are no longer appearing.

Alvin1Zhang, 2019-03-24:
@i-franzki Thanks for the feedback.

closed 2019-03-24 by Alvin1Zhang

treii28, 2019-03-27:
that's kind of a big bug for a wifi device (if you can't use it as an access point without crashing it). Any idea of a timeline on when it will make it to release?

detly, 2019-03-28:
@treii28 It shouldn't cause a crash, just a spurious error message in the serial log. Is it possible your application is doing something else as a result of an unrecognised error?

treii28, 2019-03-28:
How would I find out? Is there a call back available for when a device connects? My device reboots as soon as I try to connect. No additional messages. It happens after that error. I can add more debugging feedback, but I'm not sure where I could.

liuzfesp, 2019-04-01:
Hi @treii28 I also suspect the crash is related to your application code. Could you help to provide the crash log and the *elf file for analysis?