espressif/esp-idf #2921: OTA using SSL Certs fails when using HTTP BasicAuth

WayneKeenan, 2019-01-07:
## Environment

- Development Kit:      Adafruit HUZZAH32
- Module or chip used:  ESP32-WROOM-32
- IDF version: v3.3-beta1-132-gdb8bc3eea
- Build System:         Make
- Compiler version:  xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     macOS
- Power Supply:         USB

## Problem Description

Downloading firmware OTA via SSL using server certificates is working but when I try to use HTTP BasicAuth then it never works.

### Expected Behavior

Firmware downloaded via SSL when username and password are used to authenticate.

### Actual Behavior

WebServers says unauthorised (HTTP CODE 401)

```
x.x.x.x - - [07/Jan/2019:10:35:43 +0000] "GET /firmware.bin HTTP/1.1" 401 204 "-" "ESP32 HTTP Client/1.0"
```

### Steps to repropduce

call `esp_https_ota()` in Code2 example below. 

### Code to reproduce this issue

Code 1: snippet of OTA config that WORKS:
```cpp
    esp_http_client_config_t config = {
            .url = CONFIG_FIRMWARE_UPGRADE_URL,
            .transport_type = HTTP_TRANSPORT_OVER_SSL,
            .cert_pem = (char *)server_cert_pem_start,
            //.event_handler = bbl_ota_http_event_handler,
    };
    esp_err_t ret = esp_https_ota(&config);

```

Code 2: Config that FAILS (url, username and password are correct) 

```cpp
    esp_http_client_config_t config = {
            .url = CONFIG_FIRMWARE_UPGRADE_URL,
            .transport_type = HTTP_TRANSPORT_OVER_SSL,
            .username = USERNAME,
            .password = PASSWORD,
            .auth_type = HTTP_AUTH_TYPE_BASIC,
            .cert_pem = (char *)server_cert_pem_start,
            //.event_handler = bbl_ota_http_event_handler,
    };
    esp_err_t ret = esp_https_ota(&config);

```

## Debug Logs

```
I (48644) mbedtls: ssl_tls.c:8021 => handshake
I (48644) mbedtls: ssl_cli.c:3405 client state: 0
I (48644) mbedtls: ssl_tls.c:2751 => flush output
I (48654) mbedtls: ssl_tls.c:2763 <= flush output
I (48654) mbedtls: ssl_cli.c:3405 client state: 1
I (48664) mbedtls: ssl_tls.c:2751 => flush output
I (48664) mbedtls: ssl_tls.c:2763 <= flush output
[... 219 lines cut ...]
I (51474) mbedtls: ssl_tls.c:8647 <= write
I (51604) mbedtls: ssl_tls.c:8207 => read
I (51604) mbedtls: ssl_tls.c:4305 => read record
I (51604) mbedtls: ssl_tls.c:2532 => fetch input
I (51604) mbedtls: ssl_tls.c:2693 in_left: 0, nb_want: 5
I (51614) mbedtls: ssl_tls.c:2717 in_left: 0, nb_want: 5
I (51614) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
I (51624) mbedtls: ssl_tls.c:2738 <= fetch input
I (51634) mbedtls: ssl_tls.c:2532 => fetch input
I (51634) mbedtls: ssl_tls.c:2693 in_left: 5, nb_want: 452
I (51644) mbedtls: ssl_tls.c:2717 in_left: 5, nb_want: 452
I (51654) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 447 (-0xfffffe41)
I (51654) mbedtls: ssl_tls.c:2738 <= fetch input
I (51664) mbedtls: ssl_tls.c:1790 => decrypt buf
I (51674) mbedtls: ssl_tls.c:2372 <= decrypt buf
I (51674) mbedtls: ssl_tls.c:4379 <= read record
I (51684) mbedtls: ssl_tls.c:8495 <= read
I (51684) esp_https_ota: Starting OTA...
I (51694) esp_https_ota: Writing to partition subtype 17 at offset 0x220000
I (51784) esp_https_ota: esp_ota_begin succeeded
I (51784) esp_https_ota: Please Wait. This may take time
E (51784) esp_ota_ops: OTA image has invalid magic byte (expected 0xE9, saw 0x3c
E (51794) esp_https_ota: Error: esp_ota_write failed! err=0x0
E (51794) bbl_ota: Firmware Upgrades Failed, 1503
```

jitin17, 2019-01-07:
@WayneKeenan Thanks for reporting this, we are looking into it.

WayneKeenan, 2019-03-03:
Hi @jitin17 I was wondering what has been found out? Thanks.

closed 2019-05-10 by igrr