espressif/esp-idf #2922: Ubuntu setup - make: *** No rule to make target 'flash'.  Stop.

YashPandit4u, 2019-01-07:
## Environment

- Module or chip used:  ESP32-WROOM-32
- IDF version:   v3.3-beta1-171-g577f8d852
- Build System:         Make
- Compiler version  1.22.0-80-g6c4433a
- Operating System:   Linux/Ubuntu
- Power Supply:        USB

## Problem Description

I have followed the official ESP IDF get started guide until the build and flash section. Until then all went fine, the make menuconfig command also runs fine. But when I run 'make flash' it takes time compiles everything and at the end give the stated error.

### Expected Behavior

The code should compile and get flashed to the esp32 chip and reset the chip once flashed.

### Actual Behavior
```
default@yash-Inspiron-5559:~/esp/esp-idf$ printenv IDF_PATH
/home/default/esp/esp-idf
default@yash-Inspiron-5559:~/esp/esp-idf$ printenv PATH
/home/default/esp/xtensa-esp32-elf/bin:/home/default/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/default/esp/xtensa-esp32-elf/bin
default@yash-Inspiron-5559:~/esp/esp-idf$ sudo make flash
make: *** No rule to make target 'flash'.  Stop.
default@yash-Inspiron-5559:~/esp/esp-idf$  
```

YashPandit4u, 2019-01-07:
> Please check that you follow the steps in "Start Project" section: https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#start-a-project. You should run 'make' commands in the project directory, not in $IDF_PATH.

I tried to check it. Now I used the make command in the project directory. The make, make all command works fine if executed individually but when used with flash i.e. ```sudo make flash``` it doesn't work.
But the command with the esptool.py that is suggested by the make command works fine and I can upload code to the esp32.

```
printenv IDF_PATH
/home/default/esp/esp-idf
default@yash-Inspiron-5559:~/esp/esp-idf/examples/get-started/hello_world$ printenv PATH
/home/default/esp/xtensa-esp32-elf/bin:/home/default/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/default/esp32/xtensa-esp32-elf/bin
default@yash-Inspiron-5559:~/esp/esp-idf/examples/get-started/hello_world$ make
Toolchain path: /home/default/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /home/default/esp/esp-idf/requirements.txt are satisfied.
App "hello-world" version: v3.3-beta1-171-g577f8d852
To flash all build output, run 'make flash' or:
python /home/default/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /home/default/esp/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin 0x10000 /home/default/esp/esp-idf/examples/get-started/hello_world/build/hello-world.bin 0x8000 /home/default/esp/esp-idf/examples/get-started/hello_world/build/partitions_singleapp.bin
default@yash-Inspiron-5559:~/esp/esp-idf/examples/get-started/hello_world$ sudo make flash
Makefile:8: /make/project.mk: No such file or directory
make: *** No rule to make target '/make/project.mk'.  Stop.
```
But this command works.

```
sudo python /home/default/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /home/default/esp/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin 0x10000 /home/default/esp/esp-idf/examples/get-started/hello_world/build/hello-world.bin 0x8000 /home/default/esp/esp-idf/examples/get-started/hello_world/build/partitions_singleapp.bin
esptool.py v2.6-beta1
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____.
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 24128 bytes to 14454...
Wrote 24128 bytes (14454 compressed) at 0x00001000 in 1.3 seconds (effective 150.5 kbit/s)...
Hash of data verified.
Compressed 145664 bytes to 70024...
Wrote 145664 bytes (70024 compressed) at 0x00010000 in 6.2 seconds (effective 187.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 1731.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
Any idea on why the ```sudo make flash``` command is not working even if the /make/project.mk file exists.

dobairoland, 2019-01-07:
One of the problem could be that you are using it with `sudo` and the privileges are not propagated. `make flash` is not intended to be used with `sudo`. If you add your user to the group which owns the serial port then `make flash` should work without `sudo`. Further information can be found here:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/establish-serial-connection.html#adding-user-to-dialout-on-linux

Mr-Andersen, 2019-03-01:
I have very similar problem: I did everything by the tutorial, but when it comes to `make menuconfig` I have this:
```
~/e/hello_world> make menuconfig
Makefile:8: /make/project.mk: No such file or directory
make: *** No rule to make target '/make/project.mk'.  Stop.
```

dobairoland, 2019-03-01:
@Mr-Andersen Probably your IDF_PATH is not defined. Please see https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#setup-path-to-esp-idf.

Mr-Andersen, 2019-03-01:
It is
```
~> echo $IDF_PATH 
/home/aka_dude/esp/esp-idf/
```

Mr-Andersen, 2019-03-01:
Funny. This was fixed with `make IDF_PATH=$IDF_PATH` ¯\_(ツ)_/¯

yvedant, 2019-04-29:
I have followed the official ESP IDF get started guide until the build and flash section. Until then all went fine, the make menuconfig command also runs fine. But when I run 'make flash' it takes time compiles everything and at the end give the stated error.

make: *** No rule to make target '/home/Vedant/esp/esp-idf/hello_world/build/ldgen_libraries', needed by '/home/Vedant/esp/esp-idf/hello_world/build/esp32/esp32.project.ld'.  Stop.

please suggest me something thanks in advance

yvedant, 2019-04-29:
i am doing it in windows.

renzbagaporo, 2019-04-30:
Hi @yvedant. This looks like a separate issue from this one. Would you mind opening another issue for it?

mahavirj, 2019-05-20:
Closing this issue, at-least original one is already answered and/or resolved. If there are any followup questions please ask them at esp32.com forum. Thanks.

closed 2019-05-20 by mahavirj

XxKavosxX, 2019-12-14:
> Funny. This was fixed with `make IDF_PATH=$IDF_PATH` ¯_(ツ)_
Thanks!! Works for me. I follow step-by-step official espressif tutorial, but i get this error.

*"Makefile:8: /make/project.mk: No such file or directory
make: *** No rule to make target '/make/project.mk'.  Stop."*

It solved "make IDF_PATH=$IDF_PATH"

sarmientosaurus, 2020-03-31:
Yes, the workaround make IDF_PATH=$IDF_PATH works. Still: the doc should be changed or the issued should be solved, not worked around.

I would re-open this issue!

renzbagaporo, 2020-04-03:
@sarmientosaurus Is this on master?

sarmientosaurus, 2020-04-03:
No, this is resolved. I apologize. Thank you!

ardubit, 2020-07-06:
macOS Catalina uses .zprofile

sudo nano .zprofile

`export IDF_PATH="~/esp/esp-idf:$IDF_PATH"`

^O (F3) Write the current file to disk