espressif/esp-idf #2923: Add callback to WPS events.

carlessole, 2019-01-07:
### Board : ESP32 Dev Module

## Issue:
I don't know how to set a callback for WPS events in ESP32.
On ESP8266 NON_OS_SDK exists a function called `bool wifi_set_wps_cb(wps_st_cb_t cb)` where you could define a callback for WPS events. So in a ESP8266 project I was able to do:

```
void myFunction(void)
{
	wifi_wps_disable();
	bool b = wifi_wps_enable(WPS_TYPE_PBC);
	if (b)
	{
		b = wifi_set_wps_cb(wps_callback);
		if (b)
		{
			b = wifi_wps_start();
			if (b)
				printf("WPS SUCCESS\r\n");
			else
				printf("WPS FAILS ON wifi_wps_start\r\n");
		}
		else
			printf("WPS FAILS ON wifi_set_wps_cb\r\n");
	}
	else
		printf("WPS FAILS ON wifi_wps_enable\r\n");
}
```

How I can do that with ESP32?
Right now I have this code:

```
uint8_t myFunction(void)
{
    uint8_t ret = true;
    esp_wps_config_t config = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);
    esp_event_loop_init(my_event_handler, NULL);
    esp_err_t res = esp_wifi_wps_disable();
    if(res == ESP_OK)
    {
        ESP_LOGD(TAG, "WPS DISABLE OK!\r\n");
        res = esp_wifi_wps_enable(&config);
        if(res == ESP_OK)
        {
            ESP_LOGD(TAG, "WPS ENABLE OK!\r\n");
            res = esp_wifi_wps_start(0);
            if(res == ESP_OK)
            {
                ESP_LOGD(TAG, "WPS START OK!\r\n");
                ret = true;
            }
            else
            {
                ESP_LOGD(TAG, "WPS START: Error num: %d\r\n", res);
                ret = false;
            }
            
        }
        else
        {        
            ESP_LOGD(TAG, "WPS ENABLE: Error num: %d\r\n", res);
            // ESP_LOGD(TAG, "Error: %s\r\n", esp_err_to_name(res));
            ret = false; 
        } 
    }
    else
    {
        ESP_LOGD(TAG, "WPS DISABLE: Error num: %d\r\n", res);
        // ESP_LOGD(TAG, "Error: %s\r\n", esp_err_to_name(res));
        ret = false; 
    }
    return ret;       
}
```
As it can be seen, I tried to use  `esp_event_loop_init(my_event_handler, NULL);` but it never goes through `my_event_handler ` function.

Thank in advance

TimXia, 2019-01-08:
@carlessole Event is used instead of callback in ESP32. Please refer to esp-idf/example/wifi/wps.

carlessole, 2019-01-08:
@TimXia thanks.
And how it can be done? After taking a look to that example I see that they use the same function that I use: `esp_event_loop_init(event_handler, NULL);`
The code they are using is:
```
/*init wifi as sta and start wps*/
static void start_wps(void)
{
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_start());
    
    ESP_LOGI(TAG, "start wps...");
    
       
    ESP_ERROR_CHECK(esp_wifi_wps_enable(&config));
    ESP_ERROR_CHECK(esp_wifi_wps_start(0));
}
```
What I'm missing?
Thank you

AmrutaCh, 2019-01-15:
@carlessole you just need to check more events in the same `event_handler`

```
static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch (event->event_id)
    {
    case SYSTEM_EVENT_STA_START:
        break;
    case SYSTEM_EVENT_AP_START:
        break;
    case SYSTEM_EVENT_STA_WPS_ER_SUCCESS:
        break;
....
....
....
    default:
        break;
    }
    return ESP_OK;
}
```

Alvin1Zhang, 2019-02-02:
@carlessole Feel free to reopen the issue, thanks.

closed 2019-02-02 by Alvin1Zhang