espressif/esp-idf #2924: #if DHCPS_DEBUG is missing in dhcpserver.c

luc-github, 2019-01-07:
`#if DHCPS_DEBUG` is missing 
So Debug message is alway appearing on serial output
https://github.com/espressif/esp-idf/blob/ea89b8c55700e8b0b7c7aae93f5451bd6f79e4b8/components/lwip/apps/dhcpserver/dhcpserver.c#L527

luc-github, 2019-02-18:
@zhangyanjiaoesp 3.1.3 esp-idf is just released, may I know when V4.0 is planned ?

Is there a reason to wait for a major release to add the define line?

**#if DHCPS_DEBUG**
        DHCPS_LOG("dhcps: send_offer>>udp_sendto result %x\n", SendOffer_err_t);
**#endif**

luc-github, 2019-03-13:
@igrr Thank you
I think there is another one  looking at code https://github.com/espressif/esp-idf/blob/27cc0d1f91e16980698aac1d1448b131a158f2c1/components/lwip/apps/dhcpserver/dhcpserver.c#L604
but never saw it in output, so not sure it can be a problem

zhangyanjiaoesp, 2019-03-14:
@luc-github I'm sorry that I didn't check all the "DEBUG_LOG" code, this place need to be changed also. Thanks for your reminding.

closed 2019-03-17 by igrr