espressif/esp-idf #2926: esp_restart not working properly

ammaree, 2019-01-08:
## Environment

- Development Kit:     [none]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):  Latest master as of today
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a5
- Operating System:     [Windows]
- Power Supply:        [external 3.3]

## Problem Description
esp_restart does not work cleanly

### Expected Behavior
No additional timeouts and crashes after calling

### Actual Behavior
2x Guru Meditation Errors, the second within core dump

### Steps to reproduce
Code is running normally, all tasks functioning properly.
A: Successful firmware upgrade; or
B: UART console command
initiates a soft restart that consistently results in the events below.

### Code to reproduce this issue
Part of a very large application but the key component initiating the restart is below:
`	static	int32_t RestartState = 0 ;
	bool	myApplicationIdleHook(void) {
		if (xRtosCheckStatus(flagAPP_RESTART)) {			// Must check for restart flag
			if (RestartState == 0) {
				vRtosSetDeleteState(maskTASKS2DELETE) ;		// start by terminating most tasks
				RestartState = 1 ;
				return true ;								// stop stage, give time to execute
			}
			if (RestartState == 1) {
				if (xActuatorRunningCount() == 0) {			// check NO actuators running...
					vRtosSetDeleteState(taskACTUATE) ;		// set to delete
					vRtosSetRunState(taskACTUATE) ;			// enable run to trigger delete
					RestartState = 2 ;
				}
				return true ;								// stop state, give time to execute
    		        }
			if ((RestartState == 2) && (uxTaskGetNumberOfTasks() < 11)) { // IRMACOS tasks terminated
				esp_restart() ;							// then restart
			}
		}
		return true ;
	}
`
## Debug Logs

```
@359 vTaskActuator() [delete]
@389 vTaskMqttTx() [delete]
@408 vTaskSensors() [delete]
@128 vTaskMqttRx() [delete]
Wrote nvs/irmacos/restart: '666'
Committing update in nvs
@336 vTaskConsole() [delete]
@566 vTaskHttp() [delete]
@427 vTaskTelnet() [delete]
@161 vTaskEvents() [delete]
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008fb83  PS      : 0x00060934  A0      : 0x80090f45  A1      : 0x3ffc7140
0x4008fb83: vPortCPUAcquireMutexIntsDisabledInternal at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/tasks.c:5093
 (inlined by) vPortCPUAcquireMutexIntsDisabled at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/tasks.c:4231

[... 134 lines cut ...]
Re-entered core dump! Exception happened during core dump!
Rebooting...

@131 app_main() [start]
@292 vTaskConsole() [start]
@455 vTaskHttp() [start]
@288 vTaskTelnet() [start]
@214 vSntpTask() [start]
@275 vTaskMqttTx() [start]
@114 vTaskMqttRx() [start]
@30 vTaskEvents() [start]
@389 vTaskSensors() [start]
@168 app_main() [delete]
@264 vTaskActuator() [start]
```

As can be seen at the start and end of the debug logs above, all the application tasks (except SNTP) shutdown orderly.
And again at the bottom of the logs all the tasks (including SNTP) start up again.
The reason why the SNTP task does not shutdown is due to the absence of the xTaskAbortDelay() function. (see [https://github.com/espressif/esp-idf/issues/1453](url) ) but even if I remove the SNTP task the symptom are exactly the same.

Although the outcome is that the device restarts, (after the 2x Guru Meditation Errors) the code returned by esp_reset_reason() is incorrect causing the application to believe it crashed and hence do a core dump to the cloud host.

The [delete] message from each task is the second last line in each task, the last being vTaskDelete() in all cases.

All help appreciated..

igrr, 2019-01-08:
Hi @ammaree, there seem to be multiple issues here, but the first one (which triggers the chain of events) is that esp_restart is called from Idle task. esp_restart is a blocking function (as you see from the backtrace it calls esp_wifi_stop, which attempts to take a mutex). However FreeRTOS does not permit calling blocking functions from the Idle task. See https://www.freertos.org/RTOS-idle-task.html ("It is therefore imperative that the hook function does not call any API functions that might cause the idle task to block").

ammaree, 2019-01-09:
Hi @igrr 

The problem is that when I had this as part of my console/control task, the task terminates but then nothing, no reboot, no error message. See code below

> Wrote nvs/irmacos/ssid: 'irmacos'
Committing update in nvs
Wrote nvs/irmacos/pswd: '1rm@C0$1'
Committing update in nvs
@566 vTaskHttp() [delete]
Wrote nvs/irmacos/restart: '666'
Committing update in nvs
@376 vTaskConsole() [delete]

`
   {
#define	maskTASKS2DELETE	(taskSNTP | taskSENSORS | taskEVENTS | taskMQTT_RX | taskMQTT_TX | taskHTTP | taskTELNET)
 static	int32_t RestartState = 0 ;
   	if (xRtosCheckStatus(flagAPP_RESTART)) {			// Must check for restart flag
    	if (RestartState == 0) {
    		vRtosSetDeleteState(maskTASKS2DELETE) ;		// start by terminating most tasks
    		RestartState = 1 ;
    		goto alldone ;								// stop stage, give time to execute
    	}
    	if (RestartState == 1) {
    		if (xActuatorRunningCount() == 0) {			// check NO actuators running...
    			vRtosSetDeleteState(taskACTUATE) ;		// set to delete
    			vRtosSetRunState(taskACTUATE) ;			// enable run to trigger delete
    			RestartState = 2 ;
    		}
    		goto alldone ;								// stop stage, give time to execute
        }
    	if ((RestartState == 2) && (uxTaskGetNumberOfTasks() < 12)) {
    		vRtosSetDeleteState(taskCONSOLE) ;
    	}
    }
alldone:
	vTaskDelay(pdMS_TO_TICKS(consoleINTERVAL_MS)) ;
    }
   if (SysTimer) {
	xTimerStop(SysTimer, 0) ;
	xTimerDelete(SysTimer, 0) ;
    }
     halSTORAGE_WriteKeyValue(halSTORAGE_STORE, halSTORAGE_KEY_RESTART, (x32_t) halSTORAGE_VALUE_PLANNED, vfUXX) ;
    // should not need to lock, all tasks (especially HTTP & TNET) winding down
    IF_TRACK(debugAPPL_THREADS, debugAPPL_MESS_DN) ;
    esp_restart() ;
    vTaskDelete(NULL) ;
`

ammaree, 2019-05-03:
Think we can close this one, problem seems to have gone away..

ammaree, 2019-07-09:
@igrr 
It seemed as if the issue went away but it very soon came back. Since our firmware was in a freeze period we did not need to do field updates hence we ignored the issue temporarily.

Since we are soon coming out of the freeze period we have investigated the issue and have the following to report back.
The shutdown procedure for all tasks have been changed to ensure:
a) all timers are stopped and deleted.
b) all queues are deleted
c) all open sockets are closed
In addition we have changed our SNTP task logic so ALL user our applications tasks are now deleted before we try to restart

The restart logic has been moved to appmain()  as called from the "main" task.
We have traced the logic and found that that it hangs in esp_restart() when calling the first (and only) shutdown handler being  esp_wifi_stop().

> -------------------------------------------------------------------------------------------
 7:20.494: #0 SNTP vSntpTask [delete]
 7:20.509: #0 Sensors vTaskSensors [delete]
 7:20.579: #0 mqttRX vTaskMqttRx [delete]
 7:20.600: #0 TNET vTaskTelnet [delete]
 7:20.700: #0 Events vTaskEvents [delete]
 7:21.203: #1 mqttTX vTaskMqttTx [delete]
 7:21.425: #1 HTTP vTaskHttp [delete]
 7:22.295: #0 Actuator vTaskActuator [delete]
 7:22.298: #0 main app_main [delete]
 7:22.497: #0 Ctrl vTaskControl [delete]
 No| Pc/Pb |-TaskName-| S | LoStk|MCU| %Util| System Ticks |Stack Base|-Task TCB-|
 1 | 22/22 |esp_timer | B | 2,976| 0 |  0.09|       437,875|0x3ffaeab0|0x3ffaf8b4|
 2 | 24/24 |ipc0      | B | 1,092| 0 |  0.00|        11,265|0x3ffbd27c|0x3ffafbd0|
 3 | 24/24 |ipc1      | B | 1,088| 1 |  3.78|    16,770,358|0x3ffbd880|0x3ffafd9c|
 5 |  1/ 1 |main      | R | 5,184| 0 |  0.00|        18,898|0x3ffbe430|0x3ffbfe34|
 6 |  0/ 0 |IDLE0     | P |   796| 0 | 83.16|   368,670,399|0x3ffbffa4|0x3ffc04a8|
 7 |  0/ 0 |IDLE1     | P |   792| 1 | 95.38|   422,847,657|0x3ffc0618|0x3ffc0b1c|
 8 |  4/ 4 |Tmr Svc   | B | 2,201| 0 |  0.00|        22,822|0x3ffc0d88|0x3ffc1b90|
12 | 18/18 |tiT       | B | 2,332| X |  0.31|     1,389,298|0x3ffc2784|0x3ffc3588|
13 | 20/20 |sys_evt   | B | 1,580| 0 |  0.00|         1,144|0x3ffc3a84|0x3ffc4488|
14 | 23/23 |wifi      | B | 1,524| 0 |  0.88|     3,942,915|0x3ffc4f3c|0x3ffc5d40|
Util:  0=0.97%  1=3.78%  X=0.31%  T=5.06%

@241 esp_restart() i=0 0x401120d8
0x401120d8: esp_wifi_stop at ??:?
> -------------------------------------------------------------------------------------------

At the time our application tasks have been shut down we have not changed anything related to the wifi configuration and the connection is still configured and running.

Since it seems to be wifi related can you please give my some pointer where we can start looking.

Thanks

reopened this issue.

liuzfesp, 2019-07-10:
@ammaree, could you let me know your IDF commit ID?

ammaree, 2019-07-10:
Latest as of yesterday morning: 138c941fad431d3ef2ccbe3eaabf08b96cdc4d0b

liuzfesp, 2019-07-10:
HI @ammaree could you provide the backtrace of last time your see the crash?

ammaree, 2019-07-10:
There is no backtrace since it does not crash.
The output pasted above is taken at the point just when esp_stop_wifi() is called.
After that no output is seen (even if left for 10 minutes) and a hard reset is required to recover.

ammaree, 2019-07-10:
@liuzfesp 

I removed some sections of code and was able to get a step further.
`36.017: #0 SNTP vSntpTask [delete]
36.026: #0 Sensors vTaskSensors [delete]
36.028: #1 mqttRX vTaskMqttRx [delete]
36.221: #0 HTTP vTaskHttp [delete]
36.234: #0 TNET vTaskTelnet [delete]
36.515: #0 Events vTaskEvents [delete]
36.518: #1 mqttTX vTaskMqttTx [delete]
36.617: #0 Actuator vTaskActuator [delete]
36.619: #0 main app_main [delete]
36.817: #0 Ctrl vTaskControl [delete]
No | Pc/Pb |-TaskName-| S | LoStk|MCU| %Util| System Ticks |Stack Base|-Task TCB-|
 1 | 22/22 |esp_timer | B | 2,976| 0 |  0.09|        34,269|0x3ffaeab0|0x3ffaf8b4|
 2 | 24/24 |ipc0      | B | 1,100| 0 |  0.00|         2,404|0x3ffbf2a4|0x3ffafbd0|
 3 | 24/24 |ipc1      | B | 1,096| 1 |  0.14|        53,171|0x3ffbf8a8|0x3ffafd9c|
 5 |  1/ 1 |main      | R | 4,920| 0 |  0.11|        44,867|0x3ffc0458|0x3ffc1e5c|
 6 |  0/ 0 |IDLE0     | P |   740| 0 | 86.76|    32,641,825|0x3ffc1fcc|0x3ffc24d0|
 7 |  0/ 0 |IDLE1     | P |   784| 1 | 96.98|    36,484,198|0x3ffc2640|0x3ffc2b44|
 8 |  4/ 4 |Tmr Svc   | B | 2,544| 0 |  0.00|         2,102|0x3ffc2db0|0x3ffc3bb8|
12 | 18/18 |tiT       | B | 2,328| X |  0.30|       114,294|0x3ffc4428|0x3ffc522c|
13 | 20/20 |sys_evt   | B | 1,576| 0 |  0.00|         1,148|0x3ffc5728|0x3ffc612c|
14 | 23/23 |wifi      | B | 1,528| 0 |  3.63|     1,365,783|0x3ffc6bf8|0x3ffc79fc|
Util:  0=3.83%  1=0.14%  X=0.30%  T=4.27%

@241 esp_restart() i=0 0x40112234
0x40112234: esp_wifi_stop at ??:?

37.658: #0 sys_evt tcpip_adapter handle_sta_disconnected 197 esp_wifi_internal_reg_rxcb ret=0x3014
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400932a5  PS      : 0x00060030  A0      : 0x80092de1  A1      : 0x3ffc3ac0
0x400932a5: uxListRemove at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/list.c:214

A2      : 0x3ffcec24  A3      : 0x74d17539  A4      : 0x74d17539  A5      : 0x00000000
A6      : 0x3ffc2d6c  A7      : 0x00000000  A8      : 0x3ffcec4c  A9      : 0xffffffff
A10     : 0x3ffc7b6c  A11     : 0x3ffc3aec  A12     : 0x80091cc8  A13     : 0x3ffc5de0
A14     : 0x00000000  A15     : 0x3ffc2dc0  SAR     : 0x00000000  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000007  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

ELF file SHA256: eb5afafbfbcea5cadd97c644efaa0a2f9210f80ec4bf0dd526dd20585f63a40e

Backtrace: 0x400932a2:0x3ffc3ac0 0x40092dde:0x3ffc3ae0 0x40092ebf:0x3ffc3b20
0x400932a2: uxListRemove at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/list.c:214

0x40092dde: prvProcessReceivedCommands at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/timers.c:485

0x40092ebf: prvTimerTask at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/timers.c:485
                                                                                                                                                                                                                                   W (58) boot:
PRO CPU has been reset by WDT.
W (58) boot: WDT reset info: PRO CPU PC=0x40093284
0x40093284: vListInsert at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/list.c:188 (discriminator 1)

W (58) boot: WDT reset info: APP CPU PC=0x4008fee6
0x4008fee6: xTaskGetCurrentTaskHandle at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/tasks.c:5063
`

ammaree, 2019-07-10:
The previous crash report was as a result of a console initiated reboot.
The following crash report, very similar, came as a result of a restart after a successful OTA.

`16:47.411: #1 SNTP vSntpTask [delete]
16:47.469: #0 Sensors vTaskSensors [delete]
16:47.519: #0 mqttRX vTaskMqttRx [delete]
16:47.710: #0 TNET vTaskTelnet [delete]
16:47.760: #0 Events vTaskEvents [delete]
16:47.763: #1 mqttTX vTaskMqttTx [delete]
16:47.964: #0 HTTP vTaskHttp [delete]
16:50.011: #0 Actuator vTaskActuator [delete]
16:50.014: #0 main app_main [delete]
16:50.211: #0 Ctrl vTaskControl [delete]
No | Pc/Pb |-TaskName-| S | LoStk|MCU| %Util| System Ticks |Stack Base|-Task TCB-|
 1 | 22/22 |esp_timer | B | 2,976| 0 |  0.09|       924,011|0x3ffaeab0|0x3ffaf8b4|
 2 | 24/24 |ipc0      | B | 1,100| 0 |  1.23|    12,481,491|0x3ffbf2a4|0x3ffafbd0|
 3 | 24/24 |ipc1      | B | 1,096| 1 |  0.67|     6,794,352|0x3ffbf8a8|0x3ffafd9c|
 5 |  1/ 1 |main      | R | 4,920| 0 |  0.00|        44,994|0x3ffc0458|0x3ffc1e5c|
 6 |  0/ 0 |IDLE0     | P |   676| 0 | 91.93|   929,473,322|0x3ffc1fcc|0x3ffc24d0|
 7 |  0/ 0 |IDLE1     | P |   704| 1 | 90.59|   915,926,611|0x3ffc2640|0x3ffc2b44|
 8 |  4/ 4 |Tmr Svc   | B | 2,209| 0 |  0.00|        48,522|0x3ffc2db0|0x3ffc3bb8|
12 | 18/18 |tiT       | B | 2,328| X |  0.27|     2,742,094|0x3ffc4428|0x3ffc522c|
13 | 20/20 |sys_evt   | B | 1,576| 0 |  0.00|         1,123|0x3ffc5728|0x3ffc612c|
14 | 23/23 |wifi      | B | 1,528| 0 |  0.67|     6,807,341|0x3ffc6bf8|0x3ffc79fc|
Util:  0=1.99%  1=0.67%  X=0.27%  T=2.93%

@241 esp_restart() i=0 0x40112234
0x40112234: esp_wifi_stop at ??:?

16:51.053: #0 sys_evt tcpip_adapter handle_sta_disconnected 197 esp_wifi_internal_reg_rxcb ret=0x3014
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
Core 0 register dump:
PC      : 0x400932ab  PS      : 0x00060630  A0      : 0x80092de1  A1      : 0x3ffc3ac0
0x400932ab: xt_ints_on at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/xtensa_intr_asm.S:159

A2      : 0x3ffcec24  A3      : 0x17ff8893  A4      : 0x17ff8893  A5      : 0x00000000
A6      : 0x3ffc2d6c  A7      : 0x00000000  A8      : 0x00000003  A9      : 0x3f40c388
A10     : 0x400f0fa4  A11     : 0x3ffc3aec  A12     : 0x80091cc8  A13     : 0x3ffc5de0
0x400f0fa4: lwip_ioctl_r_wrapper at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/port/esp32/vfs_lwip.c:58 (discriminator 7)

A14     : 0x00000000  A15     : 0x3ffc2dc0  SAR     : 0x0000000a  EXCCAUSE: 0x00000003
EXCVADDR: 0x3f40c38c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

ELF file SHA256: eb5afafbfbcea5cadd97c644efaa0a2f9210f80ec4bf0dd526dd20585f63a40e

Backtrace: 0x400932a8:0x3ffc3ac0 0x40092dde:0x3ffc3ae0 0x40092ebf:0x3ffc3b20
0x400932a8: xt_ints_on at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/xtensa_intr_asm.S:143

0x40092dde: prvProcessReceivedCommands at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/timers.c:485

0x40092ebf: xTimerGetExpiryTime at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/timers.c:485
                                                                                                                                                                                                                                   W (58) boot:
PRO CPU has been reset by WDT.
W (58) boot: WDT reset info: PRO CPU PC=0x4009327e
0x4009327e: uxListRemove at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/list.c:218

W (58) boot: WDT reset info: APP CPU PC=0x4008fed7
0x4008fed7: __getreent at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/tasks.c:5063
`

ammaree, 2019-07-13:
@liuzfesp 

Some progress with some consistency. The issue seems linked to FreeRTOS timers, possibly LWIP timers. At the time that the crash occurs ALL my applications tasks, queues and timers have been deleted. See the crash log below.

> -----------------------------------------------------------------------------------
 2:30.355: #0 SNTP vSntpTask [delete]
 2:30.360: #0 Sensors vTaskSensors [delete]
 2:30.362: #1 mqttRX vTaskMqttRx [delete]
 2:30.371: #1 TNET vTaskTelnet [delete]
 2:30.648: #0 Events vTaskEvents [delete]
 2:30.857: #1 HTTP vTaskHttp [delete]
 2:31.148: #0 mqttTX vTaskMqttTx MQTT Timer stopped
 2:31.151: #0 mqttTX vTaskMqttTx MQTT Timer deleted
 2:31.153: #0 mqttTX vTaskMqttTx MQTT Queue empty
@457 vTaskMqttTx() TIMER stopped
@462 vTaskMqttTx() MQTT Disconnecting
@466 vTaskMqttTx() MQTT Disconnected
@472 vTaskMqttTx() SOCK closed
 2:31.157: #1 mqttTX vTaskMqttTx MQTT TX break
 2:31.161: #0 mqttTX vTaskMqttTx MQTT Queue deleted
 2:31.163: #0 mqttTX vTaskMqttTx [delete]
 2:32.155: #0 Actuator vTaskActuator [delete]
 2:32.355: #0 Ctrl vTaskControl [delete]
No | Pc/Pb |-TaskName-| S | LoStk|MCU| %Util| System Ticks |Stack Base|-Task TCB-|
 1 | 22/22 |esp_timer | B | 2,976| 0 |  0.09|       149,050|0x3ffaeab0|0x3ffaf8b4|
 2 | 24/24 |ipc0      | B | 1,100| 0 |  0.00|        13,798|0x3ffbf1c4|0x3ffafbd0|
 3 | 24/24 |ipc1      | B | 1,096| 1 |  0.03|        50,820|0x3ffbf7c8|0x3ffafd9c|
 5 |  1/ 1 |main      | R | 4,920| 0 |  0.02|        38,447|0x3ffc0378|0x3ffc1d7c|
 6 |  0/ 0 |IDLE0     | P |   788| 0 | 87.14|   133,460,535|0x3ffc1eec|0x3ffc23f0|
 7 |  0/ 0 |IDLE1     | P |   784| 1 | 98.51|   150,877,453|0x3ffc2560|0x3ffc2a64|
 8 |  4/ 4 |Tmr Svc   | B | 2,209| 0 |  0.00|         8,069|0x3ffc2cd0|0x3ffc3ad8|
12 | 18/18 |tiT       | B | 2,228| X |  0.29|       449,265|0x3ffc43cc|0x3ffc51d0|
13 | 20/20 |sys_evt   | B | 1,580| 0 |  0.00|         1,141|0x3ffc5654|0x3ffc6058|
14 | 23/23 |wifi      | B | 1,532| 0 |  1.32|     2,022,292|0x3ffc6b14|0x3ffc7918|
Util:  0=1.43%  1=0.03%  X=0.29%  T=1.75%

 2:33.192: #0 main app_main [delete]
 2:33.197: #0 sys_evt tcpip_adapter handle_sta_disconnected 197 esp_wifi_internal_reg_rxcb ret=0x3014
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
Core 0 register dump:
PC      : 0x4009328f  PS      : 0x00060c30  A0      : 0x80092dc5  A1      : 0x3ffc39e0
0x4009328f: uxListRemove at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/list.c:215

A2      : 0x3ffceb40  A3      : 0x8e468c39  A4      : 0x8e468c39  A5      : 0x00000000
A6      : 0x3ffc2c8c  A7      : 0x00000000  A8      : 0x3ffcf3ac  A9      : 0x3f40c444
A10     : 0x400f1060  A11     : 0x3ffc3a0c  A12     : 0x80091cac  A13     : 0x3ffc5d10
0x400f1060: cyclic_timer at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip/lwip/src/core/timeouts.c:381

A14     : 0x00000000  A15     : 0x3ffc2ce0  SAR     : 0x0000000a  EXCCAUSE: 0x00000003
EXCVADDR: 0x3f40c448  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

ELF file SHA256: bbe4152b7b442bec602346990df32f5cde2bdde38fe1dcaad9ebbaef91dcd76f

Backtrace: 0x4009328c:0x3ffc39e0 0x40092dc2:0x3ffc3a00 0x40092ea3:0x3ffc3a40
0x4009328c: uxListRemove at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/list.c:215

0x40092dc2: prvProcessReceivedCommands at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/timers.c:485

0x40092ea3: prvTimerTask at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/timers.c:485

W (58) boot: PRO CPU has been reset by WDT.
W (58) boot: WDT reset info: PRO CPU PC=0x40093268
0x40093268: vListInsert at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/freertos/list.c:188 (discriminator 1)
W (58) boot: WDT reset info: APP CPU PC=0x4000bfdf

ammaree, 2020-07-01:
By improving the shutdown process (deleting all timers & queues) and adding some delays in the process, restarts now happen reliably hence closing the issue.

closed 2020-07-01 by ammaree