espressif/esp-idf #2927: IDF 3.2 app with older bootloader compatability issues

jakkra, 2019-01-08:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32]
- IDF version: 1c4d9f2acacc666a4eb02f1450abd400e7df58c3
- Build System:         [Make]
- Compiler version: (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description
Commit a8e46775c65f768c56d05a0fefe74a356450481c breaks backwards compatibility for our bootloader. 

In our bootloader based on your esp-idf 2.1 bootloader, but with some additions, for example FW upgrade.  We are facing problems with below code. As `REG_CLR_BIT( RTC_CNTL_WDTCONFIG0_REG, RTC_CNTL_WDT_FLASHBOOT_MOD_EN );` is not equivalent with `rtc_wdt_disable();` causing us to get a WD reset when we do FW upgrade through our bootloader. 

This happens after esp_restart is called and we enter the bootloader for FW upgrade.

```
@@ static esp_err_t bootloader_main()
- REG_CLR_BIT( RTC_CNTL_WDTCONFIG0_REG, RTC_CNTL_WDT_FLASHBOOT_MOD_EN );
+ rtc_wdt_disable();
```

```
@@ -270,14 +271,15 @@ void IRAM_ATTR esp_restart_noos()
    xt_ints_off(0xFFFFFFFF);

    // Enable RTC watchdog for 1 second
   - REG_WRITE(RTC_CNTL_WDTWPROTECT_REG, RTC_CNTL_WDT_WKEY_VALUE);
   - REG_WRITE(RTC_CNTL_WDTCONFIG0_REG,
   -        RTC_CNTL_WDT_FLASHBOOT_MOD_EN_M |
   -        (RTC_WDT_STG_SEL_RESET_SYSTEM << RTC_CNTL_WDT_STG0_S) |
   -        (RTC_WDT_STG_SEL_RESET_RTC << RTC_CNTL_WDT_STG1_S) |
   -        (1 << RTC_CNTL_WDT_SYS_RESET_LENGTH_S) |
   -        (1 << RTC_CNTL_WDT_CPU_RESET_LENGTH_S) );
   - REG_WRITE(RTC_CNTL_WDTCONFIG1_REG, rtc_clk_slow_freq_get_hz() * 1);
    + rtc_wdt_protect_off();
    + rtc_wdt_disable();
    + rtc_wdt_set_stage(RTC_WDT_STAGE0, RTC_WDT_STAGE_ACTION_RESET_RTC);
    + rtc_wdt_set_stage(RTC_WDT_STAGE1, RTC_WDT_STAGE_ACTION_RESET_SYSTEM);
    + rtc_wdt_set_length_of_reset_signal(RTC_WDT_SYS_RESET_SIG, RTC_WDT_LENGTH_200ns);
    + rtc_wdt_set_length_of_reset_signal(RTC_WDT_CPU_RESET_SIG, RTC_WDT_LENGTH_200ns);
    + rtc_wdt_set_time(RTC_WDT_STAGE0, 1000);
    + rtc_wdt_enable();
    +rtc_wdt_protect_on();
```

As only `REG_CLR_BIT( RTC_CNTL_WDTCONFIG0_REG, RTC_CNTL_WDT_FLASHBOOT_MOD_EN );` is called in the bootloader is not enough to stop the WD from timeout. 

How would you suggest we solve this issue?
I would guess that we are not the only one which will stumble onto this issue.

Thank you.

### Steps to repropduce

1. call esp_restart() in application.
2. Wait some time in the bootloader and the WD will reset.

## Other items if possible

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2737280/sdkconfig.zip)

KonstantinKondrashov, 2019-01-10:
Hi @jakkra. I try to figure out what is wrong here. As I understand you have:
* bootloader from v3.1 (you rewrited the bootloader for your purpose).
* app from v3.3. 

You gave the steps to repropduce it:
1) call esp_restart() in application.
2) Wait some time in the bootloader and the WD will reset. Why should we waite time in bootloader? We should firstly do a reset of WDT or disable them.

I tried to reproduce it but I can not get the same result as you described. Which a reset do you see in console? `rst:0xc (SW_CPU_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)`

> As REG_CLR_BIT( RTC_CNTL_WDTCONFIG0_REG, RTC_CNTL_WDT_FLASHBOOT_MOD_EN ); is not equivalent with rtc_wdt_disable();

`rtc_wdt_disable()` includes the ` REG_CLR_BIT( RTC_CNTL_WDTCONFIG0_REG, RTC_CNTL_WDT_FLASHBOOT_MOD_EN );`  https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/rtc_wdt.c#L52.

Could you give more information about it logs or a bit of code?

jakkra, 2019-01-10:
Hi thanks for looking into this @KonstantinKondrashov let me clarify:

- Bootloader is based on 2.1
- App is esp-idf 3.2

In our bootloader straight after start we do below, just like in your bootloader pre 
```
/* disable watch dog here */
REG_CLR_BIT( RTC_CNTL_WDTCONFIG0_REG, RTC_CNTL_WDT_FLASHBOOT_MOD_EN ); //<-- This line was removed and replaced rtc_wdt_disable() in the commit mentioned before
REG_CLR_BIT( TIMG_WDTCONFIG0_REG(0), TIMG_WDT_FLASHBOOT_MOD_EN );
```

```
// This comes after we call esp_restart() from app
rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0248,len:16
load:0x3fff0258,len:2164
ho 0 tail 12 room 4
load:0x40078000,len:15236
load:0x40080000,len:216
entry 0x40080020
ets Jun  8 2016 00:22:57
```
Now we are in the bootoader, we won't start the application as we now are going to do a FW upgrade through the bootloader. But very soon the esp32 is reset which looks like the rtc WD resets the device. Which I assume is because REG_CLR_BIT( RTC_CNTL_WDTCONFIG0_REG, RTC_CNTL_WDT_FLASHBOOT_MOD_EN ); is not enough to counter the new additions in esp_restart_noos.
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0248,len:16
load:0x3fff0258,len:2164
ho 0 tail 12 room 4
load:0x40078000,len:15236
load:0x40080000,len:216
entry 0x40080020
```

For reproducing this my guess is that if you use your bootloader from esp-idf 2.1 (I think any bootloader pre the commit mentioned before will work).

1. Call esp_restart in an application based on idf 3.2
2. Make some modifcation in your bootloader which won't start the appliacation straight away to simulate our FW upgrade procedure where we send a new FW over uart to our bootloader.
3. I would gess that you also will get the same RTCWDT_RTC_RESET now.

KonstantinKondrashov, 2019-01-10:
Thanks, @jakkra for reporting. I have created MR in our internal gitlab. It will be reviewed and merged as soon as possible. You can use this path to resolve this issue. [fix_wdt_settings_in_esp_reset_noos.zip](https://github.com/espressif/esp-idf/files/2745287/fix_wdt_settings_in_esp_reset_noos.zip).

jakkra, 2019-01-11:
@KonstantinKondrashov Thanks for fixing the issue! I can confirm that the patch solved our problems.

closed 2019-02-22 by igrr