espressif/esp-idf #2928: The introduction of "Linker Script Generation" breaks my project.

ataweg, 2019-01-08:
I'm working in a DOS shell of Windows 10.

By updating to a version (commit b00be955ee0e06e6c04439f3dc9a4abe90bb7704) before the introduction of "Linker Script Generation" my project was successfully created.
Update to the current head revision the build fails with:
make: *** No rule to make target '/f/working/Build/esp32/esp32.common.ld', needed by 'F:/working/Build/Wifi-Camera.elf'.  Stop.

esp32.common.ld was never created. Why? 
I found out that C:\Espressif\esp-idf\components\esp32\Makefile.projbuild together with C:\Espressif\esp-idf\make\ldgen.mk must create the file. But they didn't do their job.
After a hard day's work I got a fix. When I changed the variable BUILD_DIR_BASE in my Makefile from F:/working/Build to /F/working/Build it works fine.
Why is this needed to create esp32.common.ld but not for all other build steps?

renzbagaporo, 2019-01-14:
Hi @ataweg, just to clarify, you are doing a make build with the command line? The reason I ask is for make builds, we actually recommend using the MSYS environment (which can be setup using the instructions [here](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html) or [here](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup-scratch.html)).

Alvin1Zhang, 2019-04-18:
@ataweg Hi, would you help provide any updates for the issue? Has your issue been fixed? Thanks.

ataweg, 2019-04-18:
As stated in my first comment I have a fix. But the question at the end of this comment is still open.

closed 2019-04-18 by ataweg