espressif/esp-idf #2929: Get destioation IP addr from a socket (getsockname always returns 0.0.0.0)

olegantonyan, 2019-01-09:
The goal is: in APSTA mode if a user of esp_http_server came to STA interface - ask a password. But if they came to AP interface don't ask anything. Essentially, I don't want to annoy users by asking passwords twice: first when connecting to "private" AP and the second when opening webui. At the same time I want to ask a password if a user came from "public" STA interface. Cannot rely on Host header for security reasons.

I've added this to `httpd_main.c` function `httpd_accept_conn()` after `accept()` :
```
    char ip[20] = { 0 };
    struct sockaddr_in addr_to;
    socklen_t addr_to_len = sizeof(addr_to);
    getsockname(new_fd, (struct sockaddr *)&addr_to, &addr_to_len);
    inet_ntop(AF_INET, &addr_to.sin_addr, ip, sizeof(ip));
    printf("addr: %s\n", ip);
```
But this code always prints 0.0.0.0. Also, `struct sockaddr_in` filled in `accept()` has 0.0.0.0 sin_addr

## Environment

- Development Kit:      ESP32-DevKitC
- Kit version v4
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):  v3.3-beta1-64-gd8b625a32
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     Linux
- Power Supply:         external 5V

Alvin1Zhang, 2019-03-21:
@olegantonyan Sorry that get back to you so late, and thanks for reporting the issue.

anurag-kar, 2019-03-21:
@olegantonyan This has been answered on esp32.com forum: https://esp32.com/viewtopic.php?f=13&t=8680&p=36421#p36421

Alvin1Zhang, 2019-03-21:
Feel free to reopen, thanks.

olegantonyan, 2019-03-21:
@anurag-kar  not quite. I need not peer's IP address, but IP address of an interface where it came from (AP or STA)

anurag-kar, 2019-03-21:
@olegantonyan Sorry I was not very clear about out it. The solution is along the lines of what has been described in the forum. Anyways, the following piece of code works for me, which is a slight modification of what you have written.

```
   char ip[INET6_ADDRSTRLEN] = {0};
    struct sockaddr_in6 addr_to;
    socklen_t addr_to_len = sizeof(addr_to);
    getsockname(new_fd, (struct sockaddr *)&addr_to, &addr_to_len);
    inet_ntop(AF_INET, &addr_to.sin6_addr.un.u32_addr[3], ip, sizeof(ip));
    ESP_LOGI(TAG, "Server IP => %s", ip);
```

Please let us know if that works for you.

olegantonyan, 2019-03-21:
@anurag-kar perfect! This seems to work. Thank you

```
int new_fd = httpd_req_to_sockfd(req);
char ip[INET6_ADDRSTRLEN] = {0};
struct sockaddr_in6 addr_to;
socklen_t addr_to_len = sizeof(addr_to);
getsockname(new_fd, (struct sockaddr *)&addr_to, &addr_to_len);
inet_ntop(AF_INET, &addr_to.sin6_addr.un.u32_addr[3], ip, sizeof(ip));
ESP_LOGI(TAG, "Server IP => %s", ip);
```

closed 2019-03-21 by olegantonyan