espressif/esp-idf #2930: spi_slave example is giving validation errors on the receiver side

alafi, 2019-01-09:
Running the spi_slave example on the master branch [https://github.com/espressif/esp-idf/tree/master/examples/peripherals/spi_slave/receiver](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/spi_slave/receiver) gives the following error:

`rxdata not in DMA-capable memory or not WORD aligned`

I am not really sure but it seems like this commit broke the code: cfba157fdd1199bea1b9d9e3146d7efadd93db89

bryanjnorton, 2019-01-10:
Hello alafi
Try using the DMA_ATTR flag when you declare the buffers, also make them global. 
Example: 
    DMA_ATTR uint8_t sendbuf[128]="";
    DMA_ATTR uint8_t recvbuf[128]="";

-Bryan

alafi, 2019-01-12:
Thank you. That worked for me.

ginkgm, 2019-01-15:
@bryanjnorton @alafi thanks, some additional information, you can also declare the variable in the stack by    ``WORD_ALIGNED_ATTR``.

dhavalpatel0114, 2019-03-16:
@bryanjnorton  
hey, can you please tell me what is the reason behind adding DMA_ATTR tag.

ginkgm, 2019-03-18:
hello @dhavalpatel0114 ,
ESP32 DMA requires the receive buffer to start from a 32-bit boundary, see https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_slave.html#restrictions-and-known-issues.

The tag helps you to force the decorated global variable to be put from the boundary .

ginkgm, 2025-03-14:
Issue closed. If you have further questions, please open one new issue, thanks!

closed 2025-03-14 by ginkgm