espressif/esp-idf #2931: lwip - double free in tcp_close

M1cha, 2019-01-09:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    release/v3.2 @  ec79099
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [external 3.3V]

## Problem Description

The tcpip thread asserts while calling `memp_free` during `tcp_close`.
To me it looks like a double-free because the assert happens here:
```
MULTI_HEAP_ASSERT(!is_free(pb), pb); // block should not be free
```

### Steps to repropduce
Unfortunately this is really hard to reproduce and I couldn't build a minimal version which causes the bug yet.

### Code to reproduce this issue
Again, it's really hard to reproduce, but something like this is probably the cause:
```
shutdown(sock, SHUT_WR);
shutdown(sock, SHUT_RDWR);
close(sock);
```

## Debug Logs

```
lwip_shutdown(50, how=1)
lwip_selscan: fd=50 ready for reading
lwip_select: nready=1
lwip_recvfrom(50, 0x3ffbd3a4, 768, 0x0, ..)
lwip_recvfrom: top while sock->lastdata=0x0
tcp_recved: received 1 bytes, wnd 5744 (0).
lwip_recvfrom: netconn_recv err=-15, netbuf=0x0
TCP connection closed: LAST_ACK 49239 -> 62726.
tcp_pcb_purge
lwip_recvfrom(50): buf == NULL, error is "Connection closed."!
I (24834) wifi: station: XX:XX:XX:XX:XX:XX leave, AID = 1, bss_flags is 134243
I (24844) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
lwip_shutdown(50, how=2)
tcp_close: closing in State: CLOSED
CORRUPT HEAP: multi_heap.c:477 detected at 0x3ffda19c
abort() was called at PC 0x40090051 on core 0
0x40090051: multi_heap_assert at esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_free_impl at esp-idf/components/heap/multi_heap.c:477

Backtrace: 0x4009066c:0x3ffceda0 0x40090899:0x3ffcedc0 0x40090051:0x3ffcede0 0x400829b6:0x3ffcee00 0x40082dc1:0x3ffcee20 0x4000bec7:0x3ffcee40 0x4014d81d:0x3ffcee60 0x4014d8a7:0x3ffcee80 0x4014d8f3:0x3ffceea0 0x4014f6be:0x3ffceec0 0x4014f734:0x3ffceee0 0x4015c0ba:0x3ffcef00 0x4015cfad:0x3ffcef30 0x4014a015:0x3ffcef50 0x4008d255:0x3ffcef80
0x4009066c: invoke_abort at esp-idf/components/esp32/panic.c:707

0x40090899: abort at esp-idf/components/esp32/panic.c:707

0x40090051: multi_heap_assert at esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_free_impl at esp-idf/components/heap/multi_heap.c:477

0x400829b6: heap_caps_free at esp-idf/components/heap/heap_caps.c:130

0x40082dc1: _free_r at esp-idf/components/newlib/syscalls.c:42

0x4014d81d: mem_free at esp-idf/components/lwip/lwip/src/core/mem.c:151

0x4014d8a7: do_memp_free_pool at esp-idf/components/lwip/lwip/src/core/memp.c:231

0x4014d8f3: memp_free at esp-idf/components/lwip/lwip/src/core/memp.c:231

0x4014f6be: tcp_close_shutdown at esp-idf/components/lwip/lwip/src/core/tcp.c:1765

0x4014f734: tcp_close at esp-idf/components/lwip/lwip/src/core/tcp.c:1765

0x4015c0ba: lwip_netconn_do_close_internal at esp-idf/components/lwip/lwip/src/api/api_msg.c:1447 (discriminator 6)

0x4015cfad: lwip_netconn_do_close at esp-idf/components/lwip/lwip/src/api/api_msg.c:2000

0x4014a015: tcpip_thread at esp-idf/components/lwip/lwip/src/api/tcpip.c:483

0x4008d255: vPortTaskWrapper at esp-idf/components/freertos/port.c:403
```

liuzfesp, 2019-01-09:
HI @M1cha, does this issue exist in IDF without https://github.com/espressif/esp-idf/commit/ec7909956d3a8e0cac350e88b6b68880e69c0583?

M1cha, 2019-01-09:
@liuzfesp I've accidentally used the wrong commit-ID in my issue description. The one I tested is `d9c3a0881`. Do you think that `ec79099` could actually fix my issue?

liuzfesp, 2019-01-09:
@M1cha I'm not sure, https://github.com/espressif/esp-idf/commit/ec7909956d3a8e0cac350e88b6b68880e69c0583 is used to fix a thread-safe issue in LWIP mbox. Maybe you can try it. 

I will try to reproduce this issue myself and do more analysis.

M1cha, 2019-01-09:
I've just tested with ec79099 and this makes it break on `shutdown(50, SHUT_WR)` every single time, but the backtrace is always different. examples:

```
assertion "res == coreID || res == portMUX_FREE_VAL" failed: file "esp-idf/components/freertos/portmux_impl.inc.h", line 105, function: vPortCPUAcquireMutexIntsDisabledInternal
abort() was called at PC 0x400d439f on core 0
0x400d439f: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x4009066c:0x3ffcee00 0x40090899:0x3ffcee20 0x400d439f:0x3ffcee40 0x4008db5f:0x3ffcee70 0x4008cfac:0x3ffcee90 0x401598a5:0x3ffceed0 0x4015af29:0x3ffcef00 0x4015b91f:0x3ffcef30 0x40149d15:0x3ffcef50 0x4008d255:0x3ffcef80
0x4009066c: invoke_abort at esp-idf/components/esp32/panic.c:707

0x40090899: abort at esp-idf/components/esp32/panic.c:707

0x400d439f: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x4008db5f: vPortCPUAcquireMutexIntsDisabledInternal at esp-idf/components/freertos/tasks.c:4843
 (inlined by) vPortCPUAcquireMutexIntsDisabled at esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at esp-idf/components/freertos/tasks.c:4231

0x4008cfac: xQueueGenericReceive at esp-idf/components/freertos/queue.c:2037

0x401598a5: sys_arch_mbox_tryfetch at esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:444

0x4015af29: netconn_drain at esp-idf/components/lwip/lwip/src/api/api_msg.c:1447 (discriminator 6)

0x4015b91f: lwip_netconn_do_close at esp-idf/components/lwip/lwip/src/api/api_msg.c:1980

0x40149d15: tcpip_thread at esp-idf/components/lwip/lwip/src/api/tcpip.c:483

0x4008d255: vPortTaskWrapper at esp-idf/components/freertos/port.c:403
```

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401598a0  PS      : 0x00060830  A0      : 0x8015af2c  A1      : 0x3ffceed0  
0x401598a0: sys_arch_mbox_tryfetch at esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:444

A2      : 0x3ffda410  A3      : 0x3ffcef00  A4      : 0x3ffce178  A5      : 0x00000000  
A6      : 0x3ffce18c  A7      : 0x00000000  A8      : 0xd4544189  A9      : 0x3ffceeb0  
A10     : 0x00000000  A11     : 0x3ffcef00  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00060023  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xd4544189  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x401598a0:0x3ffceed0 0x4015af29:0x3ffcef00 0x4015b91f:0x3ffcef30 0x40149d15:0x3ffcef50 0x4008d255:0x3ffcef80
0x401598a0: sys_arch_mbox_tryfetch at esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:444

0x4015af29: netconn_drain at esp-idf/components/lwip/lwip/src/api/api_msg.c:1447 (discriminator 6)

0x4015b91f: lwip_netconn_do_close at esp-idf/components/lwip/lwip/src/api/api_msg.c:1980

0x40149d15: tcpip_thread at esp-idf/components/lwip/lwip/src/api/tcpip.c:483

0x4008d255: vPortTaskWrapper at esp-idf/components/freertos/port.c:403
```

liuzfesp, 2019-01-10:
Hi @M1cha, seems you can easily reproduce this issue, could you describe the reproduce steps in more detailed?

M1cha, 2019-01-10:
@liuzfesp https://github.com/M1cha/esp32_tcpbug
with this code I get the random crashes I described previously. (with ec79099)
Also, if I try to run the code without initializing wifi(only tcpip is initialized), sys_mbox_set_owner crashes while calling `listen`.

I still don't know if this is related to my original issue but I can definitely say, that ec79099 breaks a lot of things.

liuzfesp, 2019-01-10:
HI @M1cha OK, I"m now running you code to reproduce this issue. How long time does it need to reproduce the crash?

M1cha, 2019-01-10:
@liuzfesp for me it crashes immediately after the client closes the connection

liuzfesp, 2019-01-11:
@M1cha I have tested your code, however, everything looks OK for 12.5 hours test.  Both test method looks good for me:

1.  Connected station to the router then test, it's OK.
2. Only init TCPIP, it's OK.

Have you use 'git submodule update' to update your submodule after upgrade the IDF to https://github.com/espressif/esp-idf/commit/ec7909956d3a8e0cac350e88b6b68880e69c0583 ?

M1cha, 2019-01-11:
@liuzfesp you're totally right, I forgot to update the lwip submodule - sorry for the inconvenience.
I've just tested my project a few times with the new commit and wasn't able to reproduce the double-free yet.

I'll close this issue for now and reopen it in case it ever happens again.
Thank you for you help.

M1cha, 2019-01-14:
unfortunately the issue does still happen.
```
CORRUPT HEAP: multi_heap.c:477 detected at 0x3ffdfb0c
abort() was called at PC 0x4008ebcb on core 0
0x4008ebcb: multi_heap_assert at esp-idf/components/heap/multi_heap.c:699

Backtrace: 0x4008f333:0x3ffd3310 0x4008f62d:0x3ffd3330 0x4008ebcb:0x3ffd3350 0x4008efd5:0x3ffd3370 0x400826e9:0x3ffd3390 0x40082aa1:0x3ffd33b0 0x4000bec7:0x3ffd33d0 0x40146a30:0x3ffd33f0 0x40146aa7:0x3ffd3410 0x40146aef:0x3ffd3430 0x40148969:0x3ffd3450 0x40148a1f:0x3ffd3470 0x40143202:0x3ffd3490 0x401440c3:0x3ffd34c0 0x4014297a:0x3ffd34e0
0x4008f333: invoke_abort at esp-idf/components/esp32/panic.c:707

0x4008f62d: abort at esp-idf/components/esp32/panic.c:707

0x4008ebcb: multi_heap_assert at esp-idf/components/heap/multi_heap.c:699

0x4008efd5: multi_heap_free_impl at esp-idf/components/heap/multi_heap.c:699

0x400826e9: heap_caps_free at esp-idf/components/heap/heap_caps.c:130

0x40082aa1: _free_r at esp-idf/components/newlib/syscalls.c:42

0x40146a30: mem_free at esp-idf/components/lwip/lwip/src/core/mem.c:151

0x40146aa7: do_memp_free_pool at esp-idf/components/lwip/lwip/src/core/memp.c:231

0x40146aef: memp_free at esp-idf/components/lwip/lwip/src/core/memp.c:231

0x40148969: tcp_close_shutdown at esp-idf/components/lwip/lwip/src/core/tcp.c:1765

0x40148a1f: tcp_close at esp-idf/components/lwip/lwip/src/core/tcp.c:1765

0x40143202: lwip_netconn_do_close_internal at esp-idf/components/lwip/lwip/src/api/api_msg.c:1021

0x401440c3: lwip_netconn_do_close at esp-idf/components/lwip/lwip/src/api/api_msg.c:1999

0x4014297a: tcpip_thread at esp-idf/components/lwip/lwip/src/api/tcpip.c:483

```

and sometimes it doesn't abort but I see this message during `close`:
```
W (102527) lwip_arch: WARNING: mbox has 2 message, potential memory leaking
```

I know it's hard for you to work on this without an easy way to reproduce this but I hope it helps anyway.

liuzfesp, 2019-01-14:
@M1cha  the warning message is helpful, I will do more investigation about. But I think I need sometime before I can find the root cause of this issue.

M1cha, 2019-01-15:
@liuzfesp I can now easily reproduce this issue. I've just updated the repository at https://github.com/M1cha/esp32_tcpbug

The code now runs in AccessPoint mode and uses an externel (linux) test client.
This way I can cause many different bugs including deadlocks in the wifi task.

M1cha, 2019-01-17:
I've just bisected between 3.1(where the problem does not occur) and 3.2 and the commit that's causing all this is:
`5f3b532c8d lwip: fix crash caused by sys_mbox_free`

M1cha, 2019-01-23:
@liuzfesp I've found the actual problem:
- the esp-server closes the TX side.
- the client closes it's TX side which cause sthe esp to close RX and set the flag `TF_RXCLOSED` in `pcb->flags`
- the client completely closes the socket: `recv_flags & TF_CLOSED`. This causes `tcp_input_delayed_close` to delete the pcb without calling the function in `pcb->errf` which in turn means that `conn->pcb.tcp` now holds a pointer to freed memory
- the next call to shutdown or close tries to use and cleanup `conn->pcb.tcp` which results to "use after free" access.

espxiehang, 2019-02-18:
@M1cha hi M1cha,I"m now running you code to reproduce this issue.Failed to reproduce this issue.
I added some logs. Free is after close,and only once.
Please guide the method of reproduction.
Thanks.
### Debug Logs
I (7873) example: Socket created
I (7873) example: bind 58, 0,50d,2
I (7873) example: Socket binded
I (7883) example: Socket listening
I (7883) example: Socket accepted
I (7893) example: Received 100 bytes from 192.168.1.102:
I (7893) example: 
lwip_netconn_do_close 1946 tcp=0x3ffb9d74
lwip_netconn_do_close_internal 952 tcp=0x3ffb9d74
tcp_shutdown 435 set pcb=0x3ffb9d74 flag to RXCLOSED
I (7913) example: Connection closed
E (7913) example: Shutting down socket and restarting...
lwip_shutdown 2086 conn=0x3ffb9a7c tcp=0x3ffb9d74
lwip_netconn_do_close 1946 tcp=0x3ffb9d74
lwip_netconn_do_close_internal 952 tcp=0x3ffb9d74
tcp_shutdown 435 set pcb=0x3ffb9d74 flag to RXCLOSED
lwip_close 849 conn=0x3ffb9a7c tcp=0x3ffb9d74
lwip_netconn_do_close_internal 952 tcp=0x3ffb9d74
tcp_close 407 set pcb=0x3ffb9d74 flag to RXCLOSED
lwip_close 849 conn=0x3ffb99b8 tcp=0x3ffc598c
W (7953) lwip_arch: WARNING: mbox has 2 message, potential memory leaking

lwip_netconn_do_close_internal 952 tcp=0x3ffc598c
tcp_input_delayed_close 549 free pcb=0x3ffb9d74

M1cha, 2019-02-18:
Reproducing this bug is really fragile and just the fact that you enabled lwip debugging already changed the behavior. Also, while the bug exists in  all versions of esp-idf every released, it doesn't appear that easily in all of them.

I found out that this bug even exists in the latest mainline version of LWIP when running on a linux host instead of the esp. See my post on their mailing list: http://lists.nongnu.org/archive/html/lwip-devel/2019-01/msg00034.html
Unfortunately they didn't react yet.

my current fix looks like this:
```
diff --git a/src/core/tcp_in.c b/src/core/tcp_in.c
index 282c0d3a..64ad4b4c 100644
--- a/src/core/tcp_in.c
+++ b/src/core/tcp_in.c
@@ -538,12 +538,11 @@ tcp_input_delayed_close(struct tcp_pcb *pcb)
   if (recv_flags & TF_CLOSED) {
     /* The connection has been closed and we will deallocate the
         PCB. */
-    if (!(pcb->flags & TF_RXCLOSED)) {
-      /* Connection closed although the application has only shut down the
-          tx side: call the PCB's err callback and indicate the closure to
-          ensure the application doesn't continue using the PCB. */
-      TCP_EVENT_ERR(pcb->state, pcb->errf, pcb->callback_arg, ERR_CLSD);
-    }
+
+    /* call the PCB's err callback and indicate the closure to
+        ensure the application doesn't continue using the PCB. */
+    TCP_EVENT_ERR(pcb->state, pcb->errf, pcb->callback_arg, ERR_CLSD);
+
     tcp_pcb_remove(&tcp_active_pcbs, pcb);
     memp_free(MEMP_TCP_PCB, pcb);
     return 1;
```

While this works perfectly for me I'm not sure if this is the right solution.

espxiehang, 2019-02-22:
HI @M1cha, thanks for your deep investigation and following about this issue. 

As you said, the implementation of tcp_input_delayed_close() in latest LWIP (2.1.2) and ESP LWIP (2.0.3) is same. I will try more way to reproduce this issue and keep following this issue. However I can't merge your change into IDF unless I can reproduce this issue.

freakyxue, 2019-11-13:
@M1cha We are following your instructions to reproduce the problem

freakyxue, 2019-12-12:
hi @M1cha  
I analyzed what you said:

- the client closes it's TX side which causes the esp to close RX and set the flag TF_RXCLOSED in pcb->flags

I don't understand that,the client closes it's TX side,flag is not set to TF_RXCLOSED

liuzfesp, 2020-03-10:
Hi @M1cha, we spent more than one week trying to reproduce the double free issue in `tcp_close`, unfortunately we haven't reproduce it. Now the only thing we can do is to review the LWIP source code and try to find out the potential issue scenario. 

Our code review is based on your assumption and your demo code:

- The double free is because `tcp_input_delayed_close()`
- `tcp_input_delayed_close()` doesn't call`pcb->errf`
- pcb->flags has `TF_CLOSED` set and doesn't have TF_RXCLOSED set

According to LWIP code, the `TF_CLOSED` flags can only be set in two possible scenario:

1. TCP receives TCP ACK when it's in `LAST_ACK` state. `LAST_ACK` indicates that TCP has receives the TCP FIN packet from the peer of the connection, also it has sent out the TCP FIN to the peer and now is waiting for the ACK of the sending FIN. See https://github.com/espressif/esp-lwip/blob/da2740fa8d56b9b9e8a10602f38df1ea4dbd9b74/src/core/tcp_in.c#L960
2. Set in `tcp_trigger_input_pcb_close()`,  it's only called by `tcp_close_shutdown()`, see https://github.com/espressif/esp-lwip/blob/da2740fa8d56b9b9e8a10602f38df1ea4dbd9b74/src/core/tcp.c#L288, `tcp_close_shutdown()` can be called by `tcp_shutdown()` or `tcp_close()`. Since `TF_RXCLOSED` is not set, so it can't be called by `tcp_shutdown()`. As for`tcp_close()`, it will always set `TF_RXCLOSED`, so it can't be `tcp_close()`. So the scenario 2 is not the scenario to trigger this issue.

Now let consider scenario 1, in order to make the TCP to `LAST_ACK` state, the scenario should be:

- the TCP sockets receives FIN and gets into `CLOSE_WAIT` state first (this is a little different than the scenario describe by you)
- application calls shutdown(socket, SHUT_WR) to shutdown the socket, after this API is called, the socket sends the FIN and gets into `LAST_ACK`
- **the socket receives TCP ACK again because of some unknown reason and call `tcp_input_delayed_close()` to free TCP PCB**
- the application close the socket via `shutdown(socket, SHUT_RDWR)`, and leads to double TCP free issue

Even we know the potential scenario of this issue, we still can't reproduce this issue with normal socket application.  Since we can't verify your fix because we just can't replicate this issue, I would like to close this issue.

Anyway, free feel to reopen it if you encounter it again, I think we can do another round of investigation.

closed 2020-03-11 by Alvin1Zhang