espressif/esp-idf #2932: Core dump does not take `flash encryption`into account

vulptex, 2019-01-09:
## Environment

- IDF version:  e931fe9f578a7ce6b6aa28261750bb1344509e15 (all versions)

## Problem Description

Core dump does not take flash encryption into account. This is a huge security issue, in my point of view. When doing ssl connection handshakes or key exchanges someone can trigger watchdogs and force core dumps to flash (if enabled :+1: ). 

### Steps to repropduce

1 Enable flash encryption
2 Enable Coredump to flash
3 Enable watchdogs on threads

4 Block a thread until watchdogs triggers reboot
5 Use esptool to read core_dump partition 

6 Read plain text stack variables :-D

### Where is the issue

core_dump.c uses `raw` spi_flash_writes. 

I am not sure if we can find out if the partiton is encrypted or not and if its safe to use encryption at that stage.

igrr, 2019-01-10:
Hi @vulptex, do I understand correctly that you plan to enable core dump functionality in a production scenario, rather than in debugging?

vulptex, 2019-01-10:
Hi @igrr, yes correc. We want to build a crash dump service, which uploads the core dump partition to our backend. There we can build an online debugging tool.

Does that make sense or is there any other possibility? Even a user registered write callback might help, so we can write to our own partition or stream directly the writes to the backend.

igrr, 2019-01-10:
> stream directly the writes to the backend

if by "backend" you mean "backend server", then there's a good chance this will not work — if the device has crashed then it is likely that there is memory corruption, or RTOS structures are in an invalid state, or the device is out of memory — in other words, something as complex as sending data over network is likely to not work.

We will add a check whether the core dump partition is marked as "encrypted" in the partition table, and use `spi_flash_write_encrypted` instead of `spi_flash_write` if this is the case.

Custom core dump backends are something we will consider — we are currently collecting features for the next iteration of work on core dump module.

vulptex, 2019-01-10:
Exactly, backend == backend server.
Idea was to upload after next boot + add some information, not when crash/panic occurs.

We will disable core dump to flash  for now. Feel free to close this, or use this as follow up for your modification regarding spi_flash_write.

Alvin1Zhang, 2019-05-17:
@vulptex Thanks for reporting this issue. Would you help share if any updates for this issue? Thanks.

closed 2020-12-13 by espressif-bot