espressif/esp-idf #2933: Gatt_Client does not see the Gatt_server

Aalhabashna, 2019-01-09:
I have two ESP32 WROOM-32U boards. I have setup one as a GATT_Server and the other as GATT_Client. When I scan for devices using the nRF connect app, I can see the GATT_server. Also, when I advertise a service with the nRF connect app, it can be seen by the client. However, the GATT_Client cannot see the advertisements from the GATT_Server. I use the ESP-IDF gatt_client and gatt_server examples. Can you please advise what could be the problem?

Yulong-espressif, 2019-01-10:
It is strange， Can you show the log or code to me?

Yulong-espressif, 2019-01-10:
@Aalhabashna

Yulong-espressif, 2019-01-10:
@Aalhabashna, From the log of the client side you provided, there is no log after scan. Can you provide a complete log? Thanks.

Aalhabashna, 2019-01-10:
The last thing it shows on the client side is (I (654) GATTC_DEMO: scan start success). However, if I run the advertisement from nRF connect app the client will be able to see it. It will show this 

===================
I (27454) GATTC_DEMO: 63 2c f3 03 72 e3
I (27454) GATTC_DEMO: searched Adv Data Len 25, Scan Response Len 7
I (27454) GATTC_DEMO: searched Device Name Len 5
I (27454) GATTC_DEMO: nRF5x
I (27464) GATTC_DEMO:
===================

but it does not see the advertisements from the ESP32 server.

Aalhabashna, 2019-01-10:
a

Aalhabashna, 2019-01-10:
@Yulong-espressif 
The last thing it shows on the client side is (I (654) GATTC_DEMO: scan start success). However, if I run the advertisement from nRF connect app the client will be able to see it. It will show this

I (27454) GATTC_DEMO: 63 2c f3 03 72 e3
I (27454) GATTC_DEMO: searched Adv Data Len 25, Scan Response Len 7
I (27454) GATTC_DEMO: searched Device Name Len 5
I (27454) GATTC_DEMO: nRF5x
I (27464) GATTC_DEMO:

but it does not see the advertisements from the ESP32 server.

Yulong-espressif, 2019-01-11:
@Aalhabashna I haven't reproduce the issue you said on my platform. Can you provide your client-side code?

Aalhabashna, 2019-02-12:
It seems that the problem is that the antenna of both ESP32 nodes is too weak for them to see each other. When using external antennas, the nodes were able to see each other and establish a connection.

Weijian-Espressif, 2019-02-27:
@Aalhabashna,
 We have not reproduce the issue. What is your test distance? Can you test with two other ESP32?

Alvin1Zhang, 2019-03-14:
@Aalhabashna Hi, would you help provide any updates for the issue? Thanks.

Alvin1Zhang, 2019-04-18:
@Aalhabashna Hi, would you help provide any updates for the issue? Has your issue been fixed? Thanks.

Aalhabashna, 2019-04-18:
@Alvin1Zhang As I mentioned in a previous comment, when using external antennas, the nodes were able to see each other and establish a connection. I tested with another ESP32 and got the same results. The case can be closed. Thank you!

Alvin1Zhang, 2019-04-18:
@Aalhabashna Hi, thanks for the feedback.

closed 2019-04-18 by Alvin1Zhang