espressif/esp-idf #2934: SPIFFS PANIC'ED event called

filzek, 2019-01-09:
The SPIFFS doesnt work inside a secodary task, if SPIFFS are called from a task it deploy a guru panic'ed event. Tried everything to solve but was same result.

Can use the spiffs example and just move inside a task within a task and it will fire a exception.

igrr, 2019-01-10:
Hi @filzek, when you were opening this issue, you were presented with the issue template. Could you please supply information requested therein? In particular, IDF version, crash logs (preferrably decoded using IDF Monitor), code to reproduce the issue. Thanks.

filzek, 2019-01-16:
We found out the problem run on SPIFFS and in FAT, but in FAT if we increase the task heap it works, in SPIFFS never works same problem. We decided to change to FAT as it is working ok.

Default settings global.
Create a Initial Task that call other tasks from it.
After the 2nd task call the thrid task to mount the spiffs system hang with the SPIFFS PANIC event.

Same occurs with FAT, increasing the task heap solve the problem in FAT, but doesnt solve the problem in SPIFFS.

Alvin1Zhang, 2019-04-18:
@filzek Hi, could you please supply information requested in the issue template? In particular, IDF version, crash logs (preferrably decoded using IDF Monitor), code to reproduce the issue. Thanks.

closed 2019-05-22 by filzek