espressif/esp-idf #2936: Cmake: generating bootloader.bin fails

SkobecSlo, 2019-01-10:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32 
- IDF version: v3.3-beta1-183-g65d01336d
- Build System: CMake
- Compiler version : xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5)
- Operating System: Windows
- Power Supply: USB

## Problem Description
I tried to compile hello world example by following instructions on Get Started (Cmake preview) section of documentation. 
I could successfully open the menuconfig window, but then I had to use the idf.py build command, which failed. Mind you, I am using Git Bash command line, although I experienced the same behavior in Windows command prompt.

After I call idf.py build command process starts running normally, until it gets to the part where it has to generate bootloader.bin. This step fails to execute. Below is the output from Git Bash, I tried looking for solution on internet, but it seems there is none. 

Any ideas how can i solve it?
***
$ python2 C:/Users/MarkoSagadin/esp/esp-idf/tools/idf.py build
Python requirements from C:\Users\MarkoSagadin\esp\esp-idf\requirements.txt are satisfied.
Warn about uninitialized values.
-- Found Git: D:/Program Files/Git/mingw64/bin/git.exe (found version "2.18.0.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building empty aws_iot component due to configuration
-- Component names: soc log heap xtensa-debug-module app_trace freertos vfs newlib esp_ringbuf driver esp_event ethernet lwip tcpip_adapter mbedtls micro-ecc bootloader_support spi_flash partition_table app_update nvs_flash pthread smartconfig_ack wpa_supplicant esp32 cxx asio jsmn aws_iot bootloader bt coap console nghttp esp-tls esp_adc_cal tcp_transport esp_http_client esp_http_server esp_https_ota openssl esp_https_server esptool_py expat wear_levelling sdmmc fatfs freemodbus idf_test json libsodium mdns mqtt protobuf-c protocomm spiffs ulp unity wifi_provisioning main
-- Component paths: C:/Users/MarkoSagadin/esp/esp-idf/components/soc;C:/Users/MarkoSagadin/esp/esp-idf/components/log;C:/Users/MarkoSagadin/esp/esp-idf/components/heap;C:/Users/MarkoSagadin/esp/esp-idf/components/xtensa-debug-module;C:/Users/MarkoSagadin/esp/esp-idf/components/app_trace;C:/Users/MarkoSagadin/esp/esp-idf/components/freertos;C:/Users/MarkoSagadin/esp/esp-idf/components/vfs;C:/Users/MarkoSagadin/esp/esp-idf/components/newlib;C:/Users/MarkoSagadin/esp/esp-idf/components/esp_ringbuf;C:/Users/MarkoSagadin/esp/esp-idf/components/driver;C:/Users/MarkoSagadin/esp/esp-idf/components/esp_event;C:/Users/MarkoSagadin/esp/esp-idf/components/ethernet;C:/Users/MarkoSagadin/esp/esp-idf/components/lwip;C:/Users/MarkoSagadin/esp/esp-idf/components/tcpip_adapter;C:/Users/MarkoSagadin/esp/esp-idf/components/mbedtls;C:/Users/MarkoSagadin/esp/esp-idf/components/micro-ecc;C:/Users/MarkoSagadin/esp/esp-idf/components/bootloader_support;C:/Users/MarkoSagadin/esp/esp-idf/components/spi_flash;C:/Users/MarkoSagadin/esp/esp-idf/components/partition_table;C:/Users/MarkoSagadin/esp/esp-idf/components/app_update;C:/Users/MarkoSagadin/esp/esp-idf/components/nvs_flash;C:/Users/MarkoSagadin/esp/esp-idf/components/pthread;C:/Users/MarkoSagadin/esp/esp-idf/components/smartconfig_ack;C:/Users/MarkoSagadin/esp/esp-idf/components/wpa_supplicant;C:/Users/MarkoSagadin/esp/esp-idf/components/esp32;C:/Users/MarkoSagadin/esp/esp-idf/components/cxx;C:/Users/MarkoSagadin/esp/esp-idf/components/asio;C:/Users/MarkoSagadin/esp/esp-idf/components/jsmn;C:/Users/MarkoSagadin/esp/esp-idf/components/aws_iot;C:/Users/MarkoSagadin/esp/esp-idf/components/bootloader;C:/Users/MarkoSagadin/esp/esp-idf/components/bt;C:/Users/MarkoSagadin/esp/esp-idf/components/coap;C:/Users/MarkoSagadin/esp/esp-idf/components/console;C:/Users/MarkoSagadin/esp/esp-idf/components/nghttp;C:/Users/MarkoSagadin/esp/esp-idf/components/esp-tls;C:/Users/MarkoSagadin/esp/esp-idf/components/esp_adc_cal;C:/Users/MarkoSagadin/esp/esp-idf/components/tcp_transport;C:/Users/MarkoSagadin/esp/esp-idf/components/esp_http_client;C:/Users/MarkoSagadin/esp/esp-idf/components/esp_http_server;C:/Users/MarkoSagadin/esp/esp-idf/components/esp_https_ota;C:/Users/MarkoSagadin/esp/esp-idf/components/openssl;C:/Users/MarkoSagadin/esp/esp-idf/components/esp_https_server;C:/Users/MarkoSagadin/esp/esp-idf/components/esptool_py;C:/Users/MarkoSagadin/esp/esp-idf/components/expat;C:/Users/MarkoSagadin/esp/esp-idf/components/wear_levelling;C:/Users/MarkoSagadin/esp/esp-idf/components/sdmmc;C:/Users/MarkoSagadin/esp/esp-idf/components/fatfs;C:/Users/MarkoSagadin/esp/esp-idf/components/freemodbus;C:/Users/MarkoSagadin/esp/esp-idf/components/idf_test;C:/Users/MarkoSagadin/esp/esp-idf/components/json;C:/Users/MarkoSagadin/esp/esp-idf/components/libsodium;C:/Users/MarkoSagadin/esp/esp-idf/components/mdns;C:/Users/MarkoSagadin/esp/esp-idf/components/mqtt;C:/Users/MarkoSagadin/esp/esp-idf/components/protobuf-c;C:/Users/MarkoSagadin/esp/esp-idf/components/protocomm;C:/Users/MarkoSagadin/esp/esp-idf/components/spiffs;C:/Users/MarkoSagadin/esp/esp-idf/components/ulp;C:/Users/MarkoSagadin/esp/esp-idf/components/unity;C:/Users/MarkoSagadin/esp/esp-idf/components/wifi_provisioning;C:/Users/MarkoSagadin/esp/hello_world/main
-- IDF_VER: v3.3-beta1-183-g65d01336d
-- Project is not inside a git repository,                         will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Found PythonInterp: C:/Python27/python.exe (found version "2.7.15")
-- Found Perl: D:/Program Files/Git/usr/bin/perl.exe (found version "5.26.2")
-- Adding linker script C:/Users/MarkoSagadin/esp/hello_world/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/MarkoSagadin/esp/esp-idf/components/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/MarkoSagadin/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/MarkoSagadin/esp/esp-idf/components/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/MarkoSagadin/esp/esp-idf/components/esp32/ld/esp32.rom.spiram_incompatible_fns.ld
-- Building empty aws_iot component due to configuration
-- Component libraries:
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/MarkoSagadin/esp/hello_world/build
[1/800] Generating dummy_main_src.c
[2/800] Building C object esp-idf/coap/CMakeFiles/idf_component_coap.dir/libcoap/src/hashkey.c.obj
[3/800] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/gpio_periph.c.obj
[4/800] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/cpu_util.c.obj
[5/800] Building C object esp-idf/coap/CMakeFiles/idf_component_coap.dir/libcoap/src/block.c.obj
[6/800] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/rtc_clk.c.obj
[7/800] Building C object esp-idf/coap/CMakeFiles/idf_component_coap.dir/libcoap/src/mem.c.obj
[8/800] Building C object esp-idf/coap/CMakeFiles/idf_component_coap.dir/libcoap/src/coap_time.c.obj
[9/800] Building C object esp-idf/coap/CMakeFiles/idf_component_coap.dir/libcoap/src/encode.c.obj
[10/800] Building C object esp-idf/coap/CMakeFiles/idf_component_coap.dir/libcoap/src/debug.c.obj
[11/800] Building C object esp-idf/coap/CMakeFiles/idf_component_coap.dir/libcoap/src/subscribe.c.obj
[... 844 lines cut ...]
[47/49] Building C object CMakeFiles/bootloader.elf.dir/dummy_main_src.c.obj
[48/49] Linking C executable bootloader.elf
[49/49] Generating ../bootloader.bin
FAILED: bootloader.bin
[... cut ...]
[780/800] Linking C static library esp-idf\ulp\libulp.a
[781/800] Building C object esp-idf/unity/CMakeFiles/idf_component_unity.dir/unity/src/unity.c.obj
ninja: build stopped: subcommand failed.
Checking Python dependencies...
Running cmake in directory C:\Users\MarkoSagadin\esp\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized C:\Users\MarkoSagadin\esp\hello_world"...
Running ninja in directory C:\Users\MarkoSagadin\esp\hello_world\build
Executing "ninja all"...
ninja failed with exit code 1

marcogmaia, 2019-01-12:
This probably is something off with your sdkconfig.

projectgus, 2019-01-14:
@marcogmaia I'm not sure I agree, but if you have cases where "make menuconfig" is required (and idf.py menuconfig) is not working then these are bugs - please report them!

@SkobecSlo It seems like arguments are being passed incorrectly to esptool.py as this command (which is failing) looks valid:

> C:\Users\MarkoSagadin\esp\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 2MB -o C:/Users/MarkoSagadin/esp/hello_world/build/bootloader/bootloader.bin bootloader.elf

It seems like there at least one bug here (possibly the root cause), as the Python interpreter is not being named in this path - mostly when we run Python it should be as `<python interpreter name> <tool path.py> <args>` and for this command it's being stripped off.

Can you please try running this command in a Git Bash terminal and see if you also get the same error?

SkobecSlo, 2019-01-14:
I should also say that I have no problems building project from esp-idf tutorial for **make** build system.
If I download tool setup from this page: https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html and I also download tool chain for Cmake form this site https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/windows-setup.html I can run idf.py commands in minGW in command tool normally. This is the setup that i have been using so far, as a workaround for a problem described above. I hope that this information helps.

eflukx, 2019-02-20:
Yes somehow this seems have something to do with the Python interpreter invoked:

From Powershell this fails (with a 'too few arguments' message):
```
cmd.exe /C "cd /D C:\code\blm-esp32-firmware\build\bootloader && D:\Code\esp32\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 2MB -o bootloader.bin C:\code\blm-esp32-firmware\build\bootloader\bootloader.elf"
```

Whereas this works (see the explicit invocation of the `python`  executable
```
cmd.exe /C "cd /D C:\code\blm-esp32-firmware\build\bootloader && python D:\Code\esp32\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 2MB -o bootloader.bin C:\code\blm-esp32-firmware\build\bootloader\bootloader.elf"
```

Probably this only fails on some systems with particular versions of Python installed. Possibly this is a bug in esptool.py playing up.

Ps. The ambiguity of having multiple Python's installed on my Win10 machine gave me more headaches with the CMake build system. Stricter handling of the python path/environment is something to work on for esp32 CMake build system to work across any machine.

projectgus, 2019-03-26:
Hi @SkobecSlo @eflukx,

Sorry for the delay in getting back to you. This issue should be fixed as of commit d0b2d5e, there were some cases where the Python interpreter was not being correctly invoked. Can you please pull the latest master branch and let us know if there are any remaining problems?

Angus

eflukx, 2019-03-27:
@projectgus 

> Sorry for the delay in getting back to you. 

No worries. There was a workaround by setting up (only? can't fully remember) the correct/a working Python version. After some fiddling around on one of my (two) Windows machines (which gave me the issue) got ESP-IDF working like a charm.

As a general side note: overall, the CMake system gives much better user experience than the make based system (at least on Windows that is. For the make-based build system I always reverted to Linux as running that on Windows is/was a pain.. ;))

projectgus, 2019-03-27:
Yes, I think if setting the .py file association to the "correct" Python.exe worked for you then this commit should have fixed the root cause for good.

> As a general side note: overall, the CMake system gives much better user experience than the make based system (at least on Windows that is

Awesome! Improving the user experience was a major reason for creating the CMake-based system. :)

projectgus, 2019-03-27:
> For me there's no easy way to test if the patch worked (as my environment is fixed now).

OK. Will wait to see if @SkobecSlo has any feedback, but I'm pretty confident this is fixed now.

projectgus, 2019-04-30:
Closing as it seems like this is fixed. Thanks @eflukx for confirming.

closed 2019-04-30 by projectgus

wnoliveira, 2019-09-23:
I'm having a problem very similar to this. My "idf.py build" output is:

`C:\Projetos\Firmware\ESP32\WROOM\new old\inv600-A5\project>idf.py build
Executing action: all (aliases: build)
Running ninja in directory 'c:\projetos\firmware\esp32\wroom\new old\inv600-a5\project\build'
Executing "ninja all"...
[23/445] Performing build step for 'bootloader'
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-build bootloader/bootloader.elf bootloader/bootloader.bin bootloader/bootloader.map
cmd.exe /C "cd /D "C:\Projetos\Firmware\ESP32\WROOM\new old\inv600-A5\project\build\bootloader" && C:\Espressif\environment\.espressif\tools\cmake\3.13.4\bin\cmake.exe --build ."
[1/2] Linking C executable bootloader.elf
FAILED: bootloader.elf
cmd.exe /C "cd . && C:\Espressif\environment\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/bootloader.elf.dir/project_elf_src.c.obj  -o bootloader.elf  esp-idf/soc/libsoc.a esp-idf/xtensa/libxtensa.a esp-idf/log/liblog.a esp-idf/micro-ecc/libmicro-ecc.a esp-idf/spi_flash/libspi_flash.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/main/libmain.a -Wl,--cref -Wl,--Map=C:/Projetos/Firmware/ESP32/WROOM/new old/inv600-A5/project/build/bootloader/bootloader.map esp-idf/spi_flash/libspi_flash.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/spi_flash/libspi_flash.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/micro-ecc/libmicro-ecc.a esp-idf/soc/libsoc.a esp-idf/xtensa/libxtensa.a esp-idf/log/liblog.a esp-idf/soc/libsoc.a esp-idf/xtensa/libxtensa.a esp-idf/log/liblog.a -L C:/Espressif/idf/components/esp32/ld -T esp32.peripherals.ld C:/Espressif/idf/components/xtensa/esp32/libhal.a -Wl,--gc-sections -L C:/Espressif/idf/components/esp_rom/esp32/ld -T esp32.rom.ld -T esp32.rom.newlib-funcs.ld -T esp32.rom.libgcc.ld -L C:/Espressif/idf/components/bootloader/subproject/main -T esp32.bootloader.ld -T esp32.bootloader.rom.ld && cd ."
xtensa-esp32-elf-gcc.exe: error: old/inv600-A5/project/build/bootloader/bootloader.map: No such file or directory
ninja: build stopped: subcommand failed.
[36/445] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/src/hal/spi_flash_hal_iram.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

I am using the latest version of esp-idf and still getting the bootloader issue. Is this the same problem as described above?