espressif/esp-idf #2938: One and the same ESP32 board, same OS, same model of notebooks, but different result

keelung-yang, 2019-01-11:
I'm using the lastest ESP-IDF on  master.

I flush exactly the same app into one and the same ESP32 board(ESP32-PICO-KIT V4), and copy exactly the same pc application from my notebook to same guys in our office. 

**It works on 5 notebooks, and failed on 3 notebooks.**

All OS & most of Apps in our notebooks are same since they all installed by our IT team, I also checked them: Windows 7 with SP1, 64 bit, and the same version of CP210x USB driver.

I'm using PyModbus, which use PySerial, on UART0 as communication channel between PC &  ESP32 board. 

Thanks for infomation from @igrr in issue #2848, I know the reason is reboot: 
on the failed notebooks, the ESP32 board reboot once I OPEN or CLOSE COM port.
on other notebooks, the ESP32 board reboot only when I CLOSE COM port.

But, it's the same application running on same notebook: DELL E5580, one and the same ESP32 board.

I also checked serial port settings both in Windows Device Manager and Register, they are same exclued port name.

Some tips from PySerial Doc maybe helpful but need more info:
https://pyserial.readthedocs.io/en/latest/pyserial_api.html
[image]

keelung-yang, 2019-01-12:
I write a reset function to fix this temporarily, but the reason is still not digged out.
```python
def reset(port, wait=1):
    with serial.Serial(port) as ser:
        ser.dtr = False
        ser.rts = False
    time.sleep(wait)
```

Alvin1Zhang, 2019-03-15:
@keelung-yang Thanks for reporting this issue and sorry for getting back to you so late. We will look into this.

radimkarnis, 2023-08-08:
Hello @keelung-yang,
solving this issue 4,5 years after the initial report is impossible, sorry for not dealing with this sooner.

On the other hand, there have been many improvements to our tools over the last few years. IDF monitor and esptool.py should now be implementing much more robust reset sequences. Hopefully, the reported issues are not relevant anymore.

I will close this ticket, feel free to open a new one if you encounter any issues! Have a nice day.

closed 2023-08-08 by radimkarnis

keelung-yang, 2023-08-09:
Hi @radimkarnis ,
Yes. That's make sense. No need to dig this indeed.
Thanks for your reminder!