espressif/esp-idf #2939: Calling esp_vhci_host_send_packet on core 1 causes crash

jakkra, 2019-01-11:
## Environment

- Development Kit:      [own]
- Module or chip used:  [ESP32]
- IDF version : d9c3a088131d187eff28e7554577b541e13fa0bf
- Build System:         [Make]
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description
After upgrading from IDF 3.1 to 3.2 Beta 2 we have problems calling `esp_vhci_host_send_packet` on core 1. Making a context switch to core 0 before sending the packet with `esp_vhci_host_send_packet` does not result in a crash.

As the first stack trace had something with power management in it I also tried without BT Modem Sleep, but it also crashed.

According to https://github.com/espressif/esp-idf/issues/2813#issuecomment-449234169 `esp_vhci_host_send_packet ` should work when called from both cores as long as you don't mix which core it's called on.

### Expected Behavior
No crash
### Actual Behavior
Crash

### Code to reproduce this issue

## Debug Logs

```
WITH MODEM SLEEP

CORE 1
0x4008f728: vListInsert at C:/esp-idf/components/freertos/list.c:188 (discriminator 1)
0x4008db57: vTaskPlaceOnEventList at C:/esp-idf/components/freertos/tasks.c:3537
0x4008eaec: xQueueGenericReceive at C:/esp-idf/components/freertos/queue.c:2037
0x40125379: semphr_take_wrapper at C:/esp-idf/components/bt/bt.c:1321
0x4012fd39: API_vhci_host_send_packet at ??:?
0x40125516: esp_vhci_host_send_packet at C:/esp-idf/components/bt/bt.c:1321
0x4010e8b5: transport_send_packet at C:\cb_bt_bk/port/esp32/btstack_port_esp32.c:66
0x401102fd: hci_send_acl_packet_fragments at C:\cb_bt_bk/btstack/src/hci.c:4063
0x401103b1: hci_send_acl_packet_buffer at C:\cb_bt_bk/btstack/src/hci.c:4063
0x40114401: l2cap_send_prepared_connectionless at C:\cb_bt_bk/btstack/src/l2cap.c:1183
0x4011791a: att_write_request at C:\cb_bt_bk/btstack/src/ble/gatt_client.c:2079
0x40118829: gatt_client_write_value_of_characteristic_without_response at C:\cb_bt_bk/btstack/src/ble/gatt_client.c:2079
0x4011f731: writeCreditsCentral at C:\bluetooth/source/cb_xxx.c:1255
0x4010796a: handle_gatt_client_event at C:\cb_bt_bk/source/cb_gatt_client.c:257
0x40117099: emit_event_new at C:\cb_bt_bk/btstack/src/ble/gatt_client.c:2079
0x401172d1: emit_gatt_complete_event at C:\cb_bt_bk/btstack/src/ble/gatt_client.c:2079
0x4011806d: gatt_client_att_packet_handler at C:\cb_bt_bk/btstack/src/ble/gatt_client.c:2079
0x4011ef0e: att_packet_handler at C:\cb_bt_bk/btstack/src/ble/att_dispatch.c:74
0x40113ef1: l2cap_acl_le_handler at C:\cb_bt_bk/btstack/src/l2cap.c:1183
0x40114a54: l2cap_acl_handler at C:\cb_bt_bk/btstack/src/l2cap.c:1183
0x4010f721: hci_emit_acl_packet at C:\cb_bt_bk/btstack/src/hci.c:4063
0x40112869: acl_handler at C:\cb_bt_bk/btstack/src/hci.c:4063
0x4011294a: packet_handler at C:\cb_bt_bk/btstack/src/hci.c:4063
0x4010e9f2: transport_deliver_packets at C:\cb_bt_bk/port/esp32/btstack_port_esp32.c:66
0x4010ea49: transport_process at C:\cb_bt_bk/port/esp32/btstack_port_esp32.c:66
0x4010ee7e: cbBTSTACK_RUNLOOP_process at C:\cb_bt_bk/platform/rtsl/btstack_run_loop_rtsl.c:161
0x400d7957: OS_main at C:\source/cb_os.c:526
0x400d4d8b: spa_task at C:\main.c:839
0x4008e155: vPortTaskWrapper at C:/esp-idf/components/freertos/port.c:403
CORE 0
0x4008d114: uxPortCompareSet at C:/esp-idf/components/freertos/tasks.c:3537
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at C:/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at C:/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at C:/esp-idf/components/freertos/tasks.c:4231
0x4008d82d: xTaskIncrementTick at C:/esp-idf/components/freertos/tasks.c:3537
0x4008e22b: xPortSysTickHandler at C:/esp-idf/components/freertos/port.c:403
0x4008edd9: _frxt_timer_int at C:/esp-idf/components/freertos/portasm.S:303
0x4008263a: _xt_lowint1 at C:/esp-idf/components/freertos/xtensa_vectors.S:1105
0x401576c3: esp_pm_impl_waiti at C:/esp-idf/components/esp32/pm_esp32.c:487

WITHOUT MODEM SLEEP

CORE 1
0x4008f722: vListInsert at C:/esp-idf/components/freertos/list.c:188
0x4008db57: vTaskPlaceOnEventList at C:/esp-idf/components/freertos/tasks.c:3537
0x4008eaec: xQueueGenericReceive at C:/esp-idf/components/freertos/queue.c:2037
0x400d331e: esp_ipc_call_and_wait at C:/esp-idf/components/esp32/ipc.c:116
0x400d33be: esp_ipc_call at C:/esp-idf/components/esp32/ipc.c:123
0x4008438f: cause_sw_intr_to_core_wrapper at C:/esp-idf/components/bt/bt.c:1321
0x40088771: vhci_set_interrupt at vhci.c:?
0x4012fca5: API_vhci_host_send_packet at ??:?
0x40125506: esp_vhci_host_send_packet at C:/esp-idf/components/bt/bt.c:1321
0x4010e8a5: transport_send_packet at C:/cb_bt_bk/port/esp32/btstack_port_esp32.c:66
0x40110462: hci_host_num_completed_packets at C:/cb_bt_bk/btstack/src/hci.c:4063
0x40111077: hci_run at C:/cb_bt_bk/btstack/src/hci.c:4063
0x40112668: event_handler at C:/cb_bt_bk/btstack/src/hci.c:4063
0x4011292e: packet_handler at C:/cb_bt_bk/btstack/src/hci.c:4063
0x4010e891: transport_notify_packet_send at C:/cb_bt_bk/port/esp32/btstack_port_esp32.c:66
0x4010ea25: transport_process at C:/cb_bt_bk/port/esp32/btstack_port_esp32.c:66
0x4010ee6e: cbBTSTACK_RUNLOOP_process at C:/cb_bt_bk/platform/rtsl/btstack_run_loop_rtsl.c:161
0x400d7957: OS_main at C:/cb_rtsl/source/cb_os.c:526
0x400d4d8b: spa_task at C:\source/main.c:839
0x4008e155: vPortTaskWrapper at C:/esp-idf/components/freertos/port.c:403
CORE 0
0x4008e29f: uxPortCompareSet at C:/esp-idf/components/freertos/port.c:403
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at C:/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at C:/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vPortCPUAcquireMutex at C:/esp-idf/components/freertos/port.c:394
0x4008ce1a: vTaskSwitchContext at C:/esp-idf/components/freertos/tasks.c:3537
0x4008ee04: _frxt_dispatch at C:/esp-idf/components/freertos/portasm.S:406
0x4008edba: _frxt_int_exit at C:/esp-idf/components/freertos/portasm.S:206

```

## Other items if possible

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2750272/sdkconfig.zip)

TianHao-Espressif, 2019-01-16:
@jakkra, Your host is your own code, not bluedroid in ESP-IDF, right? There's maybe something incorrect about calling method. 

@wmy-espressif , please help take a look.

jakkra, 2019-01-16:
@TianHao-Espressif @wmy-espressif Yes we are not using Bluedroid. This issue came after we changed to esp-idf 3.2 beta from the esp-idf 3.1 branch. We have not made any changes other than swapping the esp-idf to newer version. This works fine with 3.1.

mywang-espressif, 2019-01-21:
@jakkra, there are some backported bugfixes on bluetooth modem sleep for branch release/v3.2 that came after esp-idf 3.2 beta. You can try the latest commit on branch release/v3.2 to see whether the issue still exists.

blueMoodBHD, 2019-01-25:
The root cause is that we should not call `esp_vhci_host_send_packet ` from within a critical section or when the scheduler is suspended. I have updated the document and will be merged into ESP-IDF, soon.

jakkra, 2019-02-04:
Great, thank you!

closed 2019-02-04 by jakkra