espressif/esp-idf #2941: Feature request: please make -O0 an option in optimisation levels.

detly, 2019-01-12:
## Environment

- Development Kit: Wemos Lolin D32 
- Module or chip used:  ESP32-WROOM-32
- IDF version: `v3.3-beta1-99-ge2fbee9f9`
- Build System: Cmake
- Compiler version: `1.22.0-80-g6c4433a`
- Operating System: Linux (Ubuntu 18.04)
- Power Supply: USB

## Problem Description

Using the higher-level `idf.py menuconfig` configuration, there are two optimisation levels available: release and debug. The debug level uses `-Og`, but this still results in a *lot* of `value has been optimized out` messages when using GDB (via the panic handler GDB stub, not JTAG).

It would be good to have yet another level of, say, "no optimisation" corresponding to `-O0` in the menuconfig. Not great for release builds of course, but good for particularly hard debugging sessions.

In the meantime, can anyone tell me how exactly I can force `-O0` globally in my project?

igrr, 2019-01-12:
@detly for now you can set EXTRA_CFLAGS += -O0 (or the equivalent `set` in CMake) and same for EXTRA_CXXFLAGS, in the project makefile.

igrr, 2020-03-23:
Implemented in bf682856890cd5732d1ca1ea8bd9422fc78a54aa

closed 2020-03-23 by igrr